adafruit/Adafruit_Blinka #185: Support for Beaglebone Green Wireless

crbn60, 2019-11-26:
I have a SeedStudio BeagleBone Green Wireless I would like to use CircuitPython with.

It is detected correctly but not supported:

```console
>>> detector.chip.id
'AM33XX'
>>> detector.board.id
'BEAGLEBONE_GREEN_WIRELESS'
>>> detector.board.any_beaglebone
True
>>> detector.board.BEAGLEBONE_GREEN_WIRELESS
True
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/board.py", line 107, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported BEAGLEBONE_GREEN_WIRELESS
```

What is required here? Make a copy of the [BeagleBone Black definition](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/beaglebone_black.py) and find anything that is not identical?

I believe the required information can be found on Seed's wiki:

http://wiki.seeedstudio.com/BeagleBone_Green_Wireless/#pin-map

I would like to attempt this myself if I can get a few pointers in the right direction?

makermelissa, 2019-11-27:
Since they both have the same processor according to https://beagleboard.org/green and https://beagleboard.org/black, you would first need to create a board file inside of Adafruit_Blinka/src/adafruit_blinka/board/ for the green. Then you would need to go through the root level modules and have it load the green where appropriate. It should be pretty straightforward, but feel free to ask if you have any specific questions.

makermelissa, 2019-11-27:
I just noticed I pointed to green instead of green wireless, but it still applies: https://beagleboard.org/green-wireless

crbn60, 2019-11-30:
With #187 I am able to make a LED blink!

But I'm not sure that is the correct approach to this, so please comment.

pdp7, 2019-12-01:
@crbn60 Since the Black and the Green Wireless have the same processor and board headers, I think it would be simpler just to have the BBGW use the same module as the Black:
```
elif board_id == ap_board.BEAGLEBONE_GREEN_WIRELESS:
    from adafruit_blinka.board.beaglebone_black import *
```

makermelissa, 2019-12-03:
Done!

closed 2019-12-03 by makermelissa