adafruit/Adafruit_Blinka #188: Support for Odroid Xu4

mhessler97, 2019-11-30:
Would it be possible to add support for the Odroid Xu4. I want to use the BNO055 with my Odroid Xu4 but since there is no support for it for the odroid yet I need to update the library for it which is currently using the deprecated Adafruit Python GPIO Library. So any assistance with this would be appreciated.

makermelissa, 2019-11-30:
Yes, that’s one that I have and was planning on adding in the near future.

hhk7734, 2020-03-25:
Hi, @mhessler97, @makermelissa
I will check the Odroid-XU4 gpiod pin map soon.
But, due to the hardware characteristics of Odroid XU4, GPIO-related operations may not work well.

If you have time to test, use Arduino to make sure the sensor is working properly.
Ref: https://github.com/hhk7734/oduino

hhk7734, 2020-05-25:
Blinka supports XU4 #294 :) :+1:

makermelissa, 2020-05-25:
Thank you for adding it @hhk7734

closed 2020-05-25 by makermelissa