adafruit/Adafruit_Blinka #189: Add support for i2c3 through i2c6 on raspberry pi6?

tbblake, 2019-12-02:
Poking around I managed to figure out that updating lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py with additional i2c bus pins allows me to use my additional i2c buses within adafruit display libraries.  My tweaked file doesn't include support for moving any of the additional i2c buses to their secondary pins, though adding/changing those should be trivial.  No clue if I did it *right* but it works on my rpi4.  I've got 4 of those little OLED displays going simultaneously.

patch file attached for my fix inside my ssd1306 virtualenv

[patchBlinka.txt](https://github.com/adafruit/Adafruit_Blinka/files/3911343/patchBlinka.txt)

makermelissa, 2019-12-03:
I have added i2c-3 already in #190, but just noticed you have additional ones defined here, so I changed the title to add these as well.

tbblake, 2019-12-03:
> I have added i2c-3 already in #190, but just noticed you have additional ones defined here, so I changed the title to add these as well.

Is it worth or even possible to add support for i2c-gpio enabled i2c buses, or the alternate pins that can be enabled for the enable-able i2c buses?

makermelissa, 2019-12-03:
It's possible to add the alternate pins if they are common enough configurations. If they are arbitrary, then it wouldn't be worth adding.

makermelissa, 2020-04-30:
Ok, I created a new library. It's available at https://github.com/adafruit/Adafruit_Python_Extended_Bus/releases

It allows you to create an I2C object by passing the Bus ID instead of some pins. Since it extends the busio I2C object, it should be compatible. It only works with Linux boards, but those seem to be the only one having this issue. Closing this for now. If you have any problems, please open an issue in the new repository.

closed 2020-04-30 by makermelissa