adafruit/Adafruit_Blinka #197: ASUS Tinker Board

Tasm-Devil, 2019-12-11:
Hey there,

can you please add support for the tinker board. It uses as SoC a Rockchip RK3288 

I use some of them in my course and it would be awesome to mess around with some sensors. 

**Thank you very much.**

iomari, 2020-01-03:
Yes, Tinkerboard support would be very welcome.

makermelissa, 2020-01-10:
Thanks for requesting this. No ETA at the moment, but I agree this would be a good one to add.

ladyada, 2020-05-07:
@BlitzCityDIY wanna do this one with @makermelissa 's help?

BlitzCityDIY, 2020-05-07:
funny story- i had actually forked recently with the intent to try porting following the guide so I would definitely be interested.

makermelissa, 2020-05-07:
Please let me know if you see anything that you see could be improved in the guide. Thanks!

jfs10, 2020-06-18:
Tinkerboard support would be great.

mattsmithuk, 2021-01-24:
Would also love Tinkerboard support!

tioguerra, 2021-02-25:
Please add Tinker Board support.

ladyada, 2021-05-05:
@BlitzCityDIY hey do ya need help with this?

BlitzCityDIY, 2021-05-23:
sorry somehow missed this ping. i hadn't really had time recently but will try to work on this now :)

ladyada, 2021-05-23:
all good - we figured ya knew the hardware best!

ElephantsFoot, 2023-12-04:
It would be great to have a Tinkerboard implementation! 😊

earlthesquirrel, 2024-03-19:
Any progress?  I have a Tinker board that I haven't found a good use for yet, but definitely could use for CircuitPython for some projects that I have in flight