adafruit/Adafruit_Blinka #202: New Binho requirement of hidapi causing problems in other repos

makermelissa, 2019-12-16:
When doing a release of PyBadger, Travis failed because hidapi would not install because the command `sudo apt-get install libudev-dev libusb-1.0` was not run on that repo. I'm unsure of the best solution, but here's a couple of possibilities:

* Put some kind of conditional in requirements.txt and setup.py so that hidapi is only included for BinHo
* Add `sudo apt-get install libudev-dev libusb-1.0` to the .travis.yml of all other repos as they fail

I think the first option would be best, but I'm not sure if it's possible. I figured I'd open this in case anybody else has ideas.

Also, as reference, here's the error with deploying PyBadger:
```
Building wheels for collected packages: Adafruit-Blinka, Adafruit-PlatformDetect, Adafruit-PureIO, pyftdi, hidapi, sysv-ipc, spidev, pyusb
  Building wheel for Adafruit-Blinka (setup.py): started
  Building wheel for Adafruit-Blinka (setup.py): finished with status 'done'
  Stored in directory: /home/travis/.cache/pip/wheels/da/18/80/a9156e455b46b88adfedc41c9d640dc12dde2da46bb9aa47c2
  Building wheel for Adafruit-PlatformDetect (setup.py): started
  Building wheel for Adafruit-PlatformDetect (setup.py): finished with status 'done'
  Stored in directory: /home/travis/.cache/pip/wheels/77/c0/06/edb6710d07f1552f7fb7d8cb7e991d2b952c6ce67de20131eb
  Building wheel for Adafruit-PureIO (setup.py): started
  Building wheel for Adafruit-PureIO (setup.py): finished with status 'done'
  Stored in directory: /home/travis/.cache/pip/wheels/7e/9f/e0/a4851d6f3a5fd21b05ff529b69206baabb8b8ec510dab94639
  Building wheel for pyftdi (setup.py): started
  Building wheel for pyftdi (setup.py): finished with status 'done'
  Stored in directory: /home/travis/.cache/pip/wheels/cc/eb/83/16ad5d23b6ac860ca098cdc535004b3d6b6276bdbe427ce321
  Building wheel for hidapi (setup.py): started
  Building wheel for hidapi (setup.py): finished with status 'error'
  Complete output from command /home/travis/virtualenv/python3.6.7/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-vhntteqz/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-2iuofsv0 --python-tag cp36:
  running bdist_wheel
  running build
  running build_ext
  cythoning hid.pyx to hid.c
  /tmp/pip-install-vhntteqz/hidapi/.eggs/Cython-0.29.14-py3.6-linux-x86_64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-vhntteqz/hidapi/hid.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  cythoning hidraw.pyx to hidraw.c
  /tmp/pip-install-vhntteqz/hidapi/.eggs/Cython-0.29.14-py3.6-linux-x86_64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-vhntteqz/hidapi/hidraw.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  building 'hid' extension
  creating build
  creating build/temp.linux-x86_64-3.6
  creating build/temp.linux-x86_64-3.6/hidapi
  creating build/temp.linux-x86_64-3.6/hidapi/libusb
  gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/opt/python/3.6.7/include/python3.6m -c hid.c -o build/temp.linux-x86_64-3.6/hid.o
  gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/opt/python/3.6.7/include/python3.6m -c hidapi/libusb/hid.c -o build/temp.linux-x86_64-3.6/hidapi/libusb/hid.o
  hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory
  compilation terminated.
  error: command 'gcc' failed with exit status 1
```

ladyada, 2019-12-16:
FYI, i believe hidapi is used for MCP2221not binho

makermelissa, 2019-12-16:
Ok, that makes more sense why it's showing up now.

makermelissa, 2019-12-16:
Looks like available conditionals are specified in: https://www.python.org/dev/peps/pep-0508/

makermelissa, 2019-12-16:
Based on this PR (https://github.com/pypa/pip/pull/3728), which appears to allow usage of Environment Variables, it might just be possible to make it conditional.

makermelissa, 2019-12-16:
Another possibility is removing it from requirements.txt and leaving it in setup.py. My understanding is that when done with a `pip install`, setup.py will be used and when done with `pip -r requirements.txt` the requirements.txt is used. I'm not sure this is ideal though.

makermelissa, 2019-12-17:
I think for the moment, I'm just going to go with option 2 for PyBadger and leave this open.

makermelissa, 2019-12-17:
I was wondering if this new module would require updating other guides. I went through the Blinka install process for Raspberry Pi using Raspbian Lite and had no issues.

makermelissa, 2019-12-17:
Apparently this is not fixed. It appears that when including Blinka in requirements, it will run the setup.py file. This is good news because I can just check for a Travis or GH actions specific variable and not add the requirement in.

makermelissa, 2019-12-18:
Issue is now fixed via removal, but I'd like to add hidapi back in conditionally with lots of testing beforehand, so I'll leave this open for now. 

The other possible course of action is to just have the user explicitly install both libusb and hidapi in the guide.

makermelissa, 2019-12-18:
@caternuson, if you're working on the guide for the MCP2221, you will need to have the users install hidapi manually for the time being. Its dependency on a package not installable via pip was causing issues with anything that had Blinka as a dependency.

caternuson, 2019-12-18:
Fun stuff :)

There will be a guide that covers manual install. Similar to what was done for FT232H.

Could we be doing something different with these "special case" things like MCP2221 and FT232H? One idea - remove the requirements completely and rely on the Learn guides to cover install steps.

makermelissa, 2019-12-18:
One thing I would like to do is have it somehow detect that it is in the CI environment and only have it added as a requirement if it is non-CI. I attempted to do that with environment variables earlier, but that didn't work.

If we want to remove the requirements and rely on the guide, I'm cool with that. In that case, all we would need to do is close this.

makermelissa, 2019-12-21:
Just an update that we now have a preinstaller script that is being deployed as part of converting each repo to Github Actions. Once that is all in place, we can re-add hidapi back in.

makermelissa, 2020-05-04:
We've decided to leave this as is with it not being a dependency and just having instructions in the guide, so closing.

closed 2020-05-04 by makermelissa

caternuson, 2020-05-05:
Cool. That approach has the benefit that potentially unnecessary libraries won't be installed. For example, why install hidapi if user does not even have an MCP2221.

PR #203 (2019-12-17): Removing hidapi from requirements to fix Travis
Since having hidapi in requirements.txt causes anything with a Blinka dependency to fail that doesn't install libusb (see #202), the easiest thing to do is leave it out of requirements.txt, but leave it in setup.py.

The method we have the user following (pip3 install adafruit-blinka) will still have it work because that uses setup.py whereas the repos use `pip3 install -r requirements.txt`.

Fixes #202.