adafruit/Adafruit_Blinka #206: MCP2221 analogio not being found

caternuson, 2019-12-18:
```console
$ export BLINKA_MCP2221=1
$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import digitalio
>>> import analogio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'analogio'
>>> 
```

Does it need to be added here?
https://github.com/adafruit/Adafruit_Blinka/blob/cac0d65a5abc39dd7b09360de284012a5e53dc31/setup.py#L47
and maybe other parts of the PyPi deploy?

makermelissa, 2019-12-18:
Quite possibly. I've never added a new module.

caternuson, 2019-12-18:
Me either. @kattni Got any pointers? `analogio.py` was added as part of the MCP2221 work. Looks like it's not being brought in when pip installing blinka.

kattni, 2019-12-19:
@caternuson I have no idea. Closest I've come to working with Blinka is testing that libs `pip` install when completing the Python section of guides.

caternuson, 2019-12-19:
Welp, there's #207 FWIW.

caternuson, 2019-12-19:
Yeah. It was that simple. Fixed by #207 and released as 3.3.4.

MCP2221 example:
```python
$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import analogio
>>> knob = analogio.AnalogIn(board.G1)
>>> knob.value
24768
>>> knob.value
43840
>>> knob.value
15360
>>> 
```

closed 2019-12-19 by caternuson