adafruit/Adafruit_Blinka #208: I2C writeto_then_readfrom on FT232H: AttributeError: 'bytearray' object has no attribute 'tobytes'

dglaude, 2019-12-27:
While trying to use an MLX90640 I2C thermal camera, connected to a FT232H, connected to a Mac mini, I got the following error: `AttributeError: 'bytearray' object has no attribute 'tobytes'`

The full error logs is at the end.

The test code I use is taken from here: https://pypi.org/project/adafruit-circuitpython-mlx90640/

I believe the problem is in src/adafruit_blinka/microcontroller/ft232h/i2c.py on line 33

But I have no clue what is the origin of the problem, nor if it is related to my specific setup...

I am sorry for the poor quality of this report. I hope this can be reproduce and help solve a bug.

> python3 testmlx.py 
> Traceback (most recent call last):
>   File "testmlx.py", line 8, in <module>
>     mlx = adafruit_mlx90640.MLX90640(i2c)
>   File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/adafruit_mlx90640.py", line 102, in __init__
>     self._I2CReadWords(0x2400, eeData)
>   File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/adafruit_mlx90640.py", line 757, in _I2CReadWords
>     i2c.write_then_readinto(addrbuf, inbuf, in_end=read_words*2) # in bytes
>   File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 136, in write_then_readinto
>     in_start=in_start, in_end=in_end)
>   File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/busio.py", line 80, in writeto_then_readfrom
>     in_start=in_start, in_end=in_end, stop=stop)
>   File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/adafruit_blinka/microcontroller/ft232h/i2c.py", line 35, in writeto_then_readfrom
>     relax=True).tobytes()
> AttributeError: 'bytearray' object has no attribute 'tobytes'

ladyada, 2019-12-27:
ok btw, the MLX will not work with FT232H...the FT232H doesnt buffer I2C write/reads so its very slow and the MLX requires you read the data within one frame

dglaude, 2019-12-27:
Thank you @ladyada
Let met guess... Adafruit MCP2221A Breakout is no better?

From my point of view, this can be closed.
But there might be a problem with writeto_then_readfrom in Blinka and this will be triggered by other driver for other hardware. So someone may want to double check this.

PS: For the MLX, I switched to PyGamer where demo code works as expected, no trying to displayio it.

ladyada, 2019-12-27:
MCP2221 works fine
https://learn.adafruit.com/jupyter-on-any-computer-with-circuitpython-libraries-and-mcp2221/thermal-camera

caternuson, 2020-01-08:
@dglaude Can you check what version of pyftdi you are using? The Blinka FT232H support was developed on a version prior to 0.40.0, which looks like it made some API breaking changes:
https://eblot.github.io/pyftdi/index.html#major-changes

dglaude, 2020-01-08:
@caternuson I must have been above: `Requirement already satisfied: pyftdi in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (0.42.2)`

From my point of view this issue can be closed since MLX90640 will not work (or not well) with FT232H.

If someone has similar issue with I2C communication with Blinka on OSX, this can be reopen, or a more specific one be open...

caternuson, 2020-01-08:
Thanks. I'm pretty sure that's the issue - pyftdi made an API breaking change at 0.40.0. The reads are now returning bytes and no longer need the `tobytes()` conversion. I'm going to re-open since this is a generic issue for I2C on the FT232H.

@ladyada Want to just update requirements.txt to limit it to an older pyftdi version that works? Or update Blinka to match pyftdi's latest API?

ladyada, 2020-01-08:
update plz!

caternuson, 2020-01-09:
The basic pyftdi API issue should be fixed in release version 3.3.9.

This does not fix MLX90640 use with FT232H. See above for info on that limitation.

closed 2020-01-09 by caternuson