adafruit/Adafruit_Blinka #210: 100% CPU use of libgpiod_pulsein on Raspberry Pi 4

michaellass, 2019-12-31:
Trying to access a DHT22 sensor from a Raspberry Pi 4 using `adafruit_dht` and therefore this library, I observe 100% CPU use of the process `libgpiod_pulsein`.

Simple reproducer:
```
#!/usr/bin/env python3

import adafruit_dht
import board
import time

pin=board.D2

dht_device = adafruit_dht.DHT22(pin)

time.sleep(3600)
```

This issue was already mentioned in https://github.com/adafruit/Adafruit_Blinka/issues/140#issuecomment-526713507 but I think it deserves its own bug report.

Attaching strace to the busy libgpiod_pulsein process shows the following messages in a high frequency:
```
ioctl(4, GPIOHANDLE_GET_LINE_VALUES_IOCTL, 0xbea2b1cc) = 0
gettimeofday({tv_sec=1577794452, tv_usec=413127}, NULL) = 0
msgrcv(32768, 0xbea2b37c, 4096, 1, IPC_NOWAIT) = -1 ENOMSG (No message of desired type)
```

Interestingly, the sensor readout itself works well. Also the deprecated https://github.com/adafruit/Adafruit_Python_DHT library does not show this problem.

I'm running the most recent version of raspbian. libgpiod2 is installed in version 1.2-3+rpi1.

andyhines, 2019-12-31:
I too am having this issue, using a pi4 and a DHT11 sensor. I just came here looking to see if there were any reported bugs after seeing a spike in the cpu temp/load and then `/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein` at the top of htop. I also switched to this version of reading the DHT after seeing that the old version was deprecated.

I can confirm that in my case as well, the sensor readings themselves are fine. It seems the library simply isn't going to sleep or stays active polling, perhaps waiting to serve requests for the temp/humidity. If i do find any clues i'll update!

makermelissa, 2019-12-31:
Have either of you noticed if this is specific to the Pi 4? Thanks

ladyada, 2019-12-31:
yes it busywaits for pulses thats the only way to bitbang reliably on the Pi/linux

andyhines, 2019-12-31:
I just set up and tested my same config but using a pi zero w, and can confirm that both the 4 and the zero w demonstrate the same issue. In the case of the zero w (single core) this is particularly problematic. I may revert to the deprecated code to avoid this.

0p3t0, 2020-01-01:
Hi, same issue here, i'm using a raspberry pi 3 A+, and I have same behaviour.

michaellass, 2020-01-01:
> yes it busywaits for pulses thats the only way to bitbang reliably on the Pi/linux

I guess that is OK if we busywait only when it's really required. If I understand the datasheet correctly, the sensor does not send data all the time but only once after being queried by the Pi. Communicating the current sensor readings should then take about 5ms. After that, there is no need to hog the CPU anymore (until the next call of `measure()`).

0p3t0, 2020-01-02:
If needed i've used a script as workaround:

#Kill libgpiod_pulsein

APP_ID=$(ps -C libgpiod_pulsein --no-header --format 'pid')
ps -C libgpiod_pulsein
if [ -n "${APP_ID}" ]; then
    echo "Stopping instance $APP_ID"
    pkill libgpiod_pulsein
fi

I've planned is every 5 minutes via cron :)

michaellass, 2020-01-17:
OK, I understand the issue now. A crude workaround is to throttle the main application loop of libgpiod_pulsein when it is paused, see https://github.com/michaellass/libgpiod_pulsein/commit/8c98ddcf29675686b89eb4d9e5656d5838337434.

However, a proper fix will be to separate the main application loop and the sensor communication. I will try to come up with a pull request later.

MaximKeegan, 2020-01-19:
Same issue on rpi4

ladyada, 2020-01-19:
@MaximKeegan can ya try https://github.com/adafruit/libgpiod_pulsein/pull/1 ?

MaximKeegan, 2020-01-20:
@ladyada `load average: 0.08, 0.08, 0.03` after applying patch

michaellass, 2020-01-20:
@MaximKeegan Thanks for testing and your feedback! I also noticed some checksum errors but couldn't really tell if the rate is higher than before. I will do some more testing on this.

ladyada, 2020-01-20:
thanks, which Pi are you using?

MaximKeegan, 2020-01-21:
I'm using rpi4 B 4gb. I also have Pi Zero Wireless, but have no time to check on this board(

michaellass, 2020-01-22:
@MaximKeegan Since you already went through the effort of testing my PR, maybe you want to test the most recent version. It should reduce the error rate significantly, at least on an rpi 4.

I hope to be able to test the PR on an rpi 2 soon. However, I have no access to any older hardware.

MaximKeegan, 2020-01-23:
CPU `load average: 0.08, 0.18, 0.52` and im not seeing Checksum errors. Good!))

michaellass, 2020-01-23:
If you could do that it would be of much help I think. We should definitely test the change on that class of devices before merging the PR.

MaximKeegan, 2020-01-26:
I have found some strange behavior on rpi4. 
`Jan 26 08:20:10 raspberrypi python3[437]: temperature=21.3 humidity=19.9
Jan 26 08:20:15 raspberrypi python3[437]: temperature=21.3 humidity=3283.9
Jan 26 08:20:20 raspberrypi python3[437]: temperature=21.3 humidity=19.9`
Without any CRC errors or something. And some time a lot of CRC errors (for a minute or two). But I think it's unrelated...

michaellass, 2020-01-26:
If I understand the communication protocol correctly, two bits flipped in that one transmission:

```
21.3 / 19.9:
00000000 11000111 00000000 11010101 10011100

21.3 / 3283.9:
10000000 01000111 00000000 11010101 10011100
```

That last 8 bits is the parity data. It is unfortunately identical for both of these, so the error wasn't detectable.

Regarding the increased error rates for a minute or so: Could it be that the system is under higher load during that time? That would however also influence the current version of the code.

ladyada, 2020-01-26:
DHT's just aren't reliable when read by linux-based machines. we really wish folks would stop using them, they aren't even the cheapest humidity sensors :)

kilianbr, 2020-02-11:
I'm sorry, if my question is a bit offtopic. I've got the exact same problems described here. Do I get those latest fixes of CPU usage just by installing the newest version of the Adadruit_DHT library? 
Feel free to delete my comment, if it's the wrong place..

Saeven, 2020-04-26:
I think I've been hunting the same problem, I couldn't read from a DHT because libgpiod_pulsein is hanging for a very long time.

Simple code that causes the issue:

```
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT22(board.D4)
print("{:.1f},{}".format(dhtDevice.temperature,dhtDevice.humidity))
```

I get **one** reading, any subsequent readings fail.  ps aux | grep pulse reveals this:

    ..../bcm238x/pulseio/libgpiod_pulsein --pulses 81 --queue 7906 -i gpiochip0 4  

top shows it at 95%+ percent.

jamespoore, 2020-05-07:
I got my sensor today and installed **adafruit-circuitpython-dht** from pip and I also have this problem. 100% CPU usage on my Pi4.

Is the pull request progressing?

TIA

makermelissa, 2020-05-07:
I don't think it was making any progress. I mean from what I can tell, it looks good, it just needs to be packaged up, so thanks for poking. @jamespoore. :)

closed 2020-05-07 by makermelissa

lane-maxwell, 2020-07-17:
I know this was closed, but I'm still experiencing libgpiod_pulsein spiking and staying at 100%.  I've built and installed the library from the libgpiod repo after I noticed it, thinking that I had an old version, but still the same.  It works for a while, then spikes to 100% and restarting the python script is the only thing that resolves it (can't read from the sensor once this happens).  I've gone back to the legacy Adafruit_DHT.read_retry and it's working fine.

This is on a Pi 4.

michaellass, 2020-07-20:
@lane-maxwell I have not encountered this within several of months running the new code 24/7. However, there might be some race condition in the new code that leads to this behavior. Anyway, this would be a new bug as the one described here was definitely fixed. So probably it's worth a separate bug report.

makermelissa, 2020-07-20:
@lane-maxwell, make sure you have the latest version of Blinka. You can force it to get the latest of each package with this command: `pip3 install --upgrade --force-reinstall adafruit-blinka`

lane-maxwell, 2020-08-06:
I've done the above, and can confirm that I have the latest versions of all the packages.  I've been doing a bit of testing and here's my find.  First, the application I'm referencing is a Flask app, exposing a temperature endpoint, but only actually reading from the sensor when a web request is received.  If I start the app but do not hit the temp endpoint, then it spikes to 100% consumption and never goes down.  Once I hit the endpoint that performs a read from the sensor, then it drops down and behaves normally.

If I background this script then libgpiod_pulsein will stay at or above 100%
`
#!/usr/bin/python3

import adafruit_dht
import board
import time

dhtDevice = adafruit_dht.DHT11(board.D7)

while True:
    pass
    time.sleep(30)
`

`
PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
5144 pi        20   0   10324    408    336 S 100.0   0.0   0:35.93 libgpiod_pulsei
`

michaellass, 2020-08-06:
@lane-maxwell Thanks for the detailed description. I think I understand the issue now. Let's continue the discussion at https://github.com/adafruit/Adafruit_CircuitPython_DHT/pull/47.

darton, 2020-09-09:
The problem with 100% CPU use of libgpiod_pulsein on Raspberry Pi with DHT22  still exists. 

Raspbian GNU/Linux 10 (buster)
Raspberry Pi 3 Model B Plus Rev 1.3
Python 3.7.3
Adafruit-Blinka-5.3.4 
Adafruit-PlatformDetect-2.16 
Adafruit-PureIO-1.1.5 RPi.GPIO-0.7.0 
adafruit-circuitpython-dht-3.5.1 
pyftdi-0.51.2 pyserial-3.4 pyusb-1.0.2 
rpi-ws281x-4.2.4 sysv-ipc-1.0.1

MarcelCoding, 2021-11-07:
I too still have the issue on a rpi 3b+

PR #287 (2020-05-07): Updated libgpiod to reduce CPU Usage
Fixes #210. Also I updated a message in PulseIn so issues like #259 are more obvious about how to fix for the user.