adafruit/Adafruit_Blinka #213: Binho Nova I2C failing with ImportError: cannot import name 'i2cPorts'

makermelissa, 2020-01-07:
As mentioned in here:
https://forums.adafruit.com/viewtopic.php?f=60&t=160585

Looking at the code I think I know the issue:
It looks like the issue is occurring at https://github.com/adafruit/Adafruit_Blinka/blob/master/src/busio.py#L36
and that looks like it loads i2cports from https://github.com/adafruit/Adafruit_Blinka/blob/master/src/microcontroller/pin.py#L38
which calls 
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/nova/i2c.py

and it doesn't appear i2cports is defined there.

@fig1010, would you like to take a look? Thanks.

fig1010, 2020-01-07:
Looking into this now...

fig1010, 2020-01-07:
@makermelissa Looks like there was a merge error when the following commit was introduced to bring in MICROCHIP_MCP2221 in the busio.py code:

Nova i2c also needs to have the following lines originally, but the MCP2221 code got in between.
self._i2c = I2C()
return

--------------------------------------------------------------------------
        elif detector.board.binho_nova:
            from adafruit_blinka.microcontroller.nova.i2c import I2C
    +   elif detector.board.microchip_mcp2221:
            from adafruit_blinka.microcontroller.mcp2221.i2c import I2C
            self._i2c = I2C()
            return

https://github.com/adafruit/Adafruit_Blinka/commit/c72b2913f8b11bc650b5a17f087a058cf7948fb9#diff-eee69fa6bdad79a8bbe1378b7e82cb94

makermelissa, 2020-01-07:
Ah excellent. Thanks for figuring this out.

caternuson, 2020-01-08:
@fig1010 Sorry. That was probably my bad. I may have done that when re-basing for the MCP2221 PR. The two code blocks were very similar and I forgot to retain the one for the nova case. I don't have the Nova HW to test with, can you verify the fix is just that one line?

fig1010, 2020-01-08:
@caternuson The change would just be the following.  I've tested with the change below and it works fine.

         elif detector.board.binho_nova:
             from adafruit_blinka.microcontroller.nova.i2c import I2C
+            self._i2c = I2C()
+            return
         elif detector.board.microchip_mcp2221:
             from adafruit_blinka.microcontroller.mcp2221.i2c import I2C
             self._i2c = I2C(frequency=frequency)

I also noticed that there was a commit to add frequency to I2C init.  I can add that to Binho Nova and do a pull request once tested separately.

fig1010, 2020-01-08:
@makermelissa @caternuson I can submit a pull request now that fixes both the merge error and also adds frequency support in our i2c init.

fig1010, 2020-01-08:
https://github.com/adafruit/Adafruit_Blinka/pull/216

closed 2020-01-09 by ladyada