adafruit/Adafruit_Blinka #221: Add Onion Omega boards on blinka

yeyeto2788, 2020-01-09:
Hey guys! 

I want to add Onion Omega boards on blinka and I'm a bit lost on it since I cannot really find that much info as I did for the Pine64.

I have found [this]( https://github.com/OnionIoT/onion-gpio-sysfs) and it seems everything is using the sysfs interface.

Can anyone point me on how to do it?

Regards.

makermelissa, 2020-01-09:
We've used libgpiod with lots of success. Doing a quick search, it looks like libgpiod is available for openwrt. If you can get the working, the rest is much easier.

yeyeto2788, 2020-01-09:
I have installed it and this is the output:

```console
root@Omega-198C:/# opkg install gpiod-tools
Installing gpiod-tools (1.0.1-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/gpiod-tools_1.0.1-1_mips_24kc.ipk
Configuring gpiod-tools.
root@Omega-198C:/# ^C
root@Omega-198C:/# gpiodetect
gpiochip0 [ath79-gpio] (32 lines)
gpiochip1 [ath9k-phy0] (16 lines)
root@Omega-198C:/# cd /sys/class/gpio
root@Omega-198C:/sys/class/gpio# for i in   gpiochip* ; do echo `cat $i/label`: 
`cat $i/base` ; done
ath79-gpio: 0
ath9k-phy0: 496
root@Omega-198C:/sys/class/gpio# cat   /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-31, parent: platform/ath79-gpio, ath79-gpio:
 gpio-8   (                    |USB power           ) out hi    
 gpio-11  (                    |reset               ) in  lo    
 gpio-27  (                    |onion:amber:system  ) out lo    

gpiochip1: GPIOs 496-511, ath9k-phy0:
 gpio-497 (                    |ath9k-phy0          ) in  lo    
root@Omega-198C:/sys/class/gpio# 
```

What would be the next step?

BTW, thanks for the help.

makermelissa, 2020-01-09:
Here's a page on using libgpiod to find which pins are which:
https://www.beyondlogic.org/an-introduction-to-chardev-gpio-and-libgpiod-on-the-raspberry-pi/

yeyeto2788, 2020-01-09:
I also found this:
```console
root@Omega-198C:/sys/class/gpio# gpioinfo
gpiochip0 - 32 lines:
        line   0:      unnamed       unused  output  active-high 
        line   1:      unnamed       unused  output  active-high 
        line   2:      unnamed       unused  output  active-high 
        line   3:      unnamed       unused  output  active-high 
        line   4:      unnamed       unused  output  active-high 
        line   5:      unnamed       unused  output  active-high 
        line   6:      unnamed       unused  output  active-high 
        line   7:      unnamed       unused  output  active-high 
        line   8:      unnamed  "USB power"  output  active-high [used]
        line   9:      unnamed       unused   input  active-high 
        line  10:      unnamed       unused  output  active-high 
        line  11:      unnamed      "reset"   input  active-high [used]
        line  12:      unnamed       unused   input  active-high 
        line  13:      unnamed       unused   input  active-high 
        line  14:      unnamed       unused   input  active-high 
        line  15:      unnamed       unused   input  active-high 
        line  16:      unnamed       unused   input  active-high 
        line  17:      unnamed       unused   input  active-high 
        line  18:      unnamed       unused   input  active-high 
        line  19:      unnamed       unused   input  active-high 
        line  20:      unnamed       unused   input  active-high 
        line  21:      unnamed       unused   input  active-high 
        line  22:      unnamed       unused   input  active-high 
        line  23:      unnamed       unused   input  active-high 
        line  24:      unnamed       unused   input  active-high 
        line  25:      unnamed       unused   input  active-high 
        line  26:      unnamed       unused   input  active-high 
        line  27:      unnamed "onion:amber:system" output active-low [used]
        line  28:      unnamed       unused   input  active-high 
        line  29:      unnamed       unused   input  active-high 
        line  30:      unnamed       unused   input  active-high 
        line  31:      unnamed       unused   input  active-high
gpiochip1 - 16 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed "ath9k-phy0"   input  active-high [used]
        line   2:      unnamed       unused   input  active-high 
        line   3:      unnamed       unused   input  active-high 
        line   4:      unnamed       unused   input  active-high 
        line   5:      unnamed       unused   input  active-high 
        line   6:      unnamed       unused   input  active-high 
        line   7:      unnamed       unused   input  active-high 
        line   8:      unnamed       unused   input  active-high 
        line   9:      unnamed       unused   input  active-high 
        line  10:      unnamed       unused   input  active-high 
        line  11:      unnamed       unused   input  active-high 
        line  12:      unnamed       unused   input  active-high 
        line  13:      unnamed       unused   input  active-high 
        line  14:      unnamed       unused   input  active-high 
        line  15:      unnamed       unused   input  active-high
```

makermelissa, 2020-01-09:
Excellent. Since there's 2 gpio chips, you probably would use a tuple in the Pins file to define them.

makermelissa, 2020-01-09:
You may need to look at datasheets too.

yeyeto2788, 2020-01-09:
Yeap, cuz as I could read on the link you sent me the name of the pins are not present. So I might need to look how to get/convert the pin name.

By any chance there is an example of a board with 2 gpio chips?

makermelissa, 2020-01-09:
Yeah, I know there's a couple. Maybe look at the snapdragon apq8016 for instance.

yeyeto2788, 2020-01-09:
Great, I'll then do it tomorrow. I'll look for the datasheet and the example you are referring.

makermelissa, 2020-01-09:
Sounds good.

yeyeto2788, 2020-01-09:
Couldn't wait for it.

With the documents below I could do a first attempt (https://github.com/yeyeto2788/Adafruit_Blinka/commit/0247ac4c6e81fec209f215e58fb0ddb3304a97b1).

**Datasheet**: https://www.openhacks.com/uploadsproductos/ar9331_datasheet.pdf
**Onion Omega pinout**: http://i.imgur.com/5wN5KKe.png
**Schematic**: https://github.com/OnionIoT/Onion-Hardware/blob/master/Schematics/Omega.pdf

When you get a chance can you please check it out? I'm not just sure it is that simple.

I'll see any notification tomorrow and do all pending actions on all repos.

yeyeto2788, 2020-01-12:
Hey @makermelissa,

Based on some info I gather from the internet I found out that the way `sysfs` and `libgpiod` calculate the GPIO is the same. For that I created https://github.com/yeyeto2788/GPIO_Calculator which I hope it helps to know somehow what pin is assigned based on its character device.

The issue is that with the Onion Omega I haven't found a better way to know what pin is assigned to a character device.

Regarding Onion Omega2 the chip is different and I haven't had the change to look it up.

Regards.

makermelissa, 2020-01-12:
Excellent! I bet that will be useful.

yeyeto2788, 2020-01-12:
I'll be great if you, @ladyada and more people helping with blinka give it a try. 

Maybe we can make it a bigger tool (which I didn't find anywhere more than the formula used on `sysfs` and `libgpiod` to know what character device is assigned based on the pin letter and pin number) to help on adding more boards into blinka.

I'm open to add features, receive PRs, comments an so on.

yeyeto2788, 2020-01-13:
Hey @makermelissa,

After some attemps I try using blinka with the modification I've done on my branch but I'm still getting this error:

```console
root@OpenWrt:/overlay/workspace/Adafruit_Blinka# python3
Python 3.6.10 (default, Jan 07 2020, 20:42:48) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev0+gdf3fdc7.d20200113-py3.6.egg/board.py", line 119, in <module>
NotImplementedError: Board not supported ONION_OMEGA
>>> 
```

I had to clone my branch and then do a `python3 ./setup.py install` in order to get blinka installed. the weird part is that as far I remember I added (https://github.com/yeyeto2788/Adafruit_Blinka/commit/47a2c5869698c36035eb52ba9134abf0672410ac) all changes you did for Pine64 but in this case for the Onion Omega :cry:

makermelissa, 2020-01-13:
Make sure you src/board.py file is the same as in the commit you posted. The error sounds like it's not handling the onion board. For installing a custom Blinka, I've always just done `pip3 install --upgrade adafruit-blinka` and then used FTP to overwrite any files I wanted to test. If you do `pip3 show adafruit-blinka`, tit will show you the location it's installed. You can then copy files on top.

yeyeto2788, 2020-01-13:
I finally got it to work, dumb me (I was on master :face_with_head_bandage: :gun:).

After installation successful I face this:
```console
root@OpenWrt:~# python3
Python 3.6.10 (default, Jan 07 2020, 20:42:48) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev20+g47a2c58-py3.6.egg/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 2, in <module>
ModuleNotFoundError: No module named 'gpiod'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev20+g47a2c58-py3.6.egg/board.py", line 119, in <module>
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev20+g47a2c58-py3.6.egg/adafruit_blinka/board/onion_omega.py", line 3, in <module>
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev20+g47a2c58-py3.6.egg/adafruit_blinka/microcontroller/atheros/ar9331/pin.py", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/Adafruit_Blinka-3.3.7.dev20+g47a2c58-py3.6.egg/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 4, in <module>
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh
```

So I tried to install libgpiod as suggested on the error, but building python bindings and I got some errors:
```console
root@OpenWrt:/overlay/workspace/libgpiod# ./autogen.sh --enable-tools=yes --pref
ix=/usr/lib/ --enable-bindings-python CFLAGS="-I/$include_path" && make && make 
install && ldconfig
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:61: error: possibly undefined macro: AC_PROG_LIBTOOL
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:87: error: possibly undefined macro: AC_CHECK_HEADERS
configure.ac:168: error: possibly undefined macro: AC_LIBTOOL_CXX
configure.ac:170: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
configure.ac:175: error: possibly undefined macro: AC_LANG_PUSH
configure.ac:177: error: possibly undefined macro: AC_LANG_POP
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

But guess what? It seems like the board got recognized and not error regarding board definition, right?

What else can I try?

makermelissa, 2020-01-13:
I found libgpiod and gpiod-tools. However, it seems the python binding is missing. This can be checked by starting the python REPL and typing `import gpiod`. Maybe find something using pip?

makermelissa, 2020-01-13:
A quick search yields: https://packages.debian.org/search?keywords=gpiod.
Ideally, the package we want is `python3-libgpiod`. There's one that says mipsel. Maybe figure out a way to add a new source to opkg and install it that way? Maybe there's something more openwrt specific.

makermelissa, 2020-01-14:
I get what you were asking. Try installing `libtool-bin` and `automake`.

yeyeto2788, 2020-01-14:
Hey @makermelissa, 

Thanks for the help, I'll try your recommendations once I get home.

If you ran out of space I recommend you to have an SD card or USB drive connected to the Onion as I did and also follow the steps written [here](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration) or [here](https://docs.onion.io/omega2-docs/usb-storage.html) which expands you filesystem and it is helpful for testing things out.

Also I added `swap` (https://docs.onion.io/omega2-docs/extending-omega-memory.html). Take into account that I previously formatted the USB drive using `fsdisk` (you can see [this](http://h-wrt.com/en/doc/flash) tutorial).

makermelissa, 2020-01-14:
Expanding it was my plan. Thanks for the links, it saves me a search.

yeyeto2788, 2020-01-14:
Hey @makermelissa ,

I have tried the following but no success :cry:

```console
root@OpenWrt:/# python3
Python 3.6.10 (default, Jan 07 2020, 20:42:48) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'gpiod'
>>> exit()
root@OpenWrt:/# pip3 install python3-libgpiod
Collecting python3-libgpiod
  Could not find a version that satisfies the requirement python3-libgpiod (from versions: )
No matching distribution found for python3-libgpiod
You are using pip version 18.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

So, I tried thiese steps from the installation script (https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh) which should be the same steps for installing it in any linux distro, right?

But still failing :cry:
```console
root@OpenWrt:/overlay/workspace/libgpiod# export PYTHON_VERSION=3
root@OpenWrt:/overlay/workspace/libgpiod# ./autogen.sh --enable-tools=yes --pref
ix=/usr/lib/ --enable-bindings-python CFLAGS="-I/$include_path" && make && sudo 
make install && sudo ldconfig
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'autostuff'.
libtoolize: copying file 'autostuff/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
configure.ac:170: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

I tried installing autoconf (already installed on system) and tried again with compilation:
```console
root@OpenWrt:/overlay/workspace/libgpiod# opkg install autoconf
Package autoconf (2.69-2) installed in root is up to date.
root@OpenWrt:/overlay/workspace/libgpiod# ./autogen.sh --enable-tools=yes --pref
ix=/usr/lib/ --enable-bindings-python CFLAGS="-I/$include_path" && make && sudo 
make install && sudo ldconfig
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'autostuff'.
libtoolize: copying file 'autostuff/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
configure.ac:170: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

If you think there is something else we can do, do not hesitate on telling me cuz it's driving me nuts.

I hope we can get things sorted and working soon. If you want after making it work, I can write a quick guide on how to do it and then (if required) publish it as learning guide.

makermelissa, 2020-01-14:
Perhaps next try a search for the error `Unexpanded AX_ macro found. Please install GNU autoconf-archive.` and see what you can dig up. 

I agree that this board, or at least openwrt, is being particularly troublesome.

Yes, we can currently get libgpiod working from the command line, but we need the python bindings to be able to control it from Python and thus Blinka. It's just nobody has created a package for openwrt for that, which makes things difficult.

makermelissa, 2020-01-14:
Another option to explore is DebWrt, which might have a lot more packages

yeyeto2788, 2020-01-14:
That might be an option also but the problem is that I don't see new versions of the firmware

In the other hand, what version of `libgpiod` did you try to build? I just noticed that there where changes on `libgpiod` so I gather the changes (v1.5-rc2) but still failing.

I'll try to look more info on the net tomorrow.

makermelissa, 2020-01-16:
I think I git cloned off libgpiod master (like in the Raspberry Pi Installer Script) and compiled. I won't be able to really dig into this until I finish up another project I'm working on though.

yeyeto2788, 2020-01-16:
No problem, I'm still looking a solution for it.

makermelissa, 2020-04-07:
I'm going to take another crack at this. I've rewritten the SPI driver so it doesn't require spidev and perhaps I can get GPIO working either with another OS or another GPIO package.

makermelissa, 2020-04-07:
I have reloaded the latest OnionOS in order to start from the beginning (and determined that OnionOS still won't work), then loaded OpenWRT 19.07 which appears to be a newer version. This version include the latest bindings, so this is looking promising.

makermelissa, 2020-04-07:
Next challenge, it doesn't look like the gpiod package (which has the tools such as gpiodetect for figuring out GPIO maps to libgpiod) is not included. If I can get that installed on mine, it at least won't be a necessary component for users.

makermelissa, 2020-04-07:
Never mind. It's called `gpiod-tools` so were' good.

yeyeto2788, 2020-04-08:
That's awesome, I have tried a few minutes ago on my onion omega (fresh install of 19.07.2) and it seems like it's not finding the package `python3-gpiod`:

```console
root@OpenWrt:/# opkg list-installed | grep gpiod
gpiod-tools - 1.3-1
libgpiod - 1.3-1
root@OpenWrt:/# opkg list-installed | grep python3
python3-base - 3.7.7-1
python3-light - 3.7.7-1
```

Can you please tell me how you did it? I thought it was going to be automatically installed when I did the `opkg install gpiod-tools` but it didn't.

I see the [package defined](https://github.com/openwrt/packages/blob/96c244ea67c43f66365f038b2887c8f730deec44/libs/libgpiod/Makefile#L66) on the openWrt /packages repository but no in the [packages released](https://openwrt.org/packages/table/start?dataflt%5BName_pkg-dependencies*%7E%5D=gpiod) in openWrt

I also wanted to mention that I remember doing a commit by mistake with the definition of the original onion omega board on https://github.com/yeyeto2788/Adafruit_Blinka/commit/6c2159457a94d067b3ccf3d4da84113c5e01c67c, so we might need to add yours on it.

makermelissa, 2020-04-08:
Sure. First I did a fresh install using the instructions at https://docs.onion.io/omega2-docs/manual-firmware-installation.html to put OnionOS back on the board. Then using sysupgrade, I loaded the image at http://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt76x8/openwrt-19.07.2-ramips-mt76x8-omega2p-squashfs-sysupgrade.bin. After doing initial setup and then an `opkg update`, I was able to install the packages with `opkg install python3-gpiod python3-pip gpiod-tools`.

yeyeto2788, 2020-04-08:
That might be the case because from what I have seen the release have done almost at the same time:
mine: http://downloads.openwrt.org/releases/19.07.2/targets/ar71xx/generic/
yours: http://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt76x8/

If you do a quick `ctrl+f` and search for `omega` the date is the same. The weird thing is I don't know how you got installed the `python3-gpiod` package.

I'll try flashing the whole firmware onto my board to see if it works.

makermelissa, 2020-04-08:
Sounds good. You might also try the snapshot version instead of the release version and see if that has it.

makermelissa, 2020-04-10:
As an update, I have been having lots of issues with the SPI and I2C drivers. I have switched over to attempting this on OnionOS, which while it does not have libgpiod, I wrote a class to use periphery that is working. I2C tested good on that and SPI is looking more promising. The other benefit to targeting OnionOS is that it comes on the board making setup massively easier for the user. The only downside so far that I've seen is nano doesn't appear to be available, so I need to use vi to edit files which makes it a bit tougher for the average user.

yeyeto2788, 2020-04-13:
Hey @makermelissa, I guess there is no way to port the Omega first generation to Blinka I found that the latest releases of the Onion OS for my board is from 2018 (http://repo.onion.io/omega/images/) and not updated at all. I'll try to find the Omega2 I have and see if I can find a USB-TTL adapter and follow the steps in order to get it working.

makermelissa, 2020-04-13:
Yeah, the onion has been particularly challenging and I think that's mostly due to the mips processor. I got i2c and uart working great. Spi is writing, but not reading correctly and I'm currently working on that.

yeyeto2788, 2020-04-14:
I did a quick search and found out that CS0 is connected to the flash storage might that be the issue? https://docs.onion.io/omega2-docs/communicating-with-spi-devices.html

ladyada, 2020-04-14:
@yeyeto2788 good eye! @makermelissa you're using spi0.1?

makermelissa, 2020-04-14:
Yes @ladyada, spidev0.1 is the only one available. I was able to get partial communication on it using a separate gpio as the cs line, but I could give it another try using the hardware cs line instead.

ladyada, 2020-04-14:
nah i dont think that should matter...

makermelissa, 2020-04-17:
PR submitted as #273. SPI input not working, but output does. We can use the Adafruit_CircuitPython_BitbangIO library for reading sensors and hardware output works for displays, which should be sufficient.

yeyeto2788, 2020-04-18:
That might be an option, as long as you can install Blinka on it. The good thing is that you got it working on the omega2 right? I even ask for help on the onion community but not luck.

makermelissa, 2020-04-18:
Yes, the Omega2+ for sure. It should work on other variations of the Omega2 as well in theory. I just don't have those to test on.

makermelissa, 2020-04-22:
@yeyeto2788, are you planning on adding the original Omega board as well or can we close this?

makermelissa, 2020-04-24:
I'm going to close this for now.

closed 2020-04-24 by makermelissa

yeyeto2788, 2020-04-25:
Hey @makermelissa,

Sorry for the late reply, for the moment we can keep it close until I get somehow the new version of the OnionOS running.

If there is anything else I can help with just let me know.