adafruit/Adafruit_Blinka #227: Specify ftdi:ft232h in ft232h Pin class

anekimken, 2020-01-18:
I'm using the FT232H to talk to a sensor, and I also have a USB-serial converter plugged in to talk to some lab equipment. The USB-serial converter also uses and FTDI chip (2232), and blinka is not able to tell them apart. `import  board` leads to
```
pyftdi.usbtools.UsbToolsError: 2 USB devices match URL 'ftdi:///1'
````

In microcontroller/ft232h/pin.py, communication is initiated by calling `i2c.configure("ftdi:///1")`. A simple change to `i2c.configure("ftdi://ftdi:ft232h/1")` fixes the problem. As best I can tell, this makes sure pyftdi loads the first ft232h it finds. In the current implementation, it finds `"ftdi://ftdi:ft232h/1"` and `ftdi://ftdi:2232:FT3HTN9H/1`, the latter of which is my USB-Serial converter. Making this switch limits the search to ft232h, although it looks like it will still just load the first one it finds.

anekimken, 2020-01-18:
Small update: there's a similar `.configure('ftdi:///1')' command here: https://github.com/adafruit/Adafruit_Blinka/blob/bf49679028e57296e105c3e50b2cd25abc8dc19b/src/adafruit_blinka/microcontroller/ft232h/i2c.py#L9

and here:
https://github.com/adafruit/Adafruit_Blinka/blob/bf49679028e57296e105c3e50b2cd25abc8dc19b/src/adafruit_blinka/microcontroller/ft232h/spi.py#L14

Looks like the same `"ftdi://ftdi:ft232h/1"` fixes it.

caternuson, 2020-01-18:
We only tested with a single FT232H. I *thought* that URL would just grab the first one found, which is why it was used. But it sounds like it's the first FTDI thing, not just the first FT232H.

Thanks for figuring out a better URL. And it sounds like you've tested it with your setup. Would you be interested in submitting a PR with that change? Basically just update the URL everywhere to the one that grabs the first FT232H found.

closed 2020-01-22 by ladyada

caternuson, 2020-01-22:
Thanks again for pointing this out and making a PR @anekimken

anekimken, 2020-01-22:
No problem. I'll let you know if I discover anything else in the library.