adafruit/Adafruit_Blinka #229: long delay between SPI transactions

jerryneedell, 2020-01-19:
I have been working with the RFM9x boards via SPI on a Raspberry Pi Zero - W and I noticed something odd. I just wanted to check if this makes sense.

When I am polling a register via SPI, there seems to be a very long (~6ms) delay between each read from the device. This is much shorter running the same code on an MCU (I'm using the STM32F405 for this test)

Looking at the code, I see that Blinka (being a good neighbor) opens and closes the SPI device for every read/write. Is this the reason for the delay? Is 6ms expected?

I have another library that directly controls the SPI bus and it does not show the same delay. It only opens the device once at the start. Not sure how well this plays in general...

Here is a logic analyzer screen shot that shows the SPI Chip select on the RPi (panel 4) and the SPI CS on the STM32F405 (bottom panel)  when both devices are running the same code and polling for the same flag. You can see that the actual SPU transactions are similar in length (CS low) but there is this long delay (~6ms)  between reads on the RPi
[image]

Does this make sense -- is it just something to live with on the RPi?

ladyada, 2020-01-19:
you def have to 'play nice' with the kernel and request the SPI device, which is more complex than on a microcontroller where you can grab it whenever :) 
you could try tracing where the delay comes from by twiddling a gpio pin and watching where the delay comes from -  im pretty sure it's kernel-internal! :(

jerryneedell, 2020-01-19:
OK -- I'll keep poking at it and see what I can learn.  I just wanted to check and see if what I was seeing was expected behavior or if I was misinterpreting something.

romilly, 2020-01-20:
Thanks for the question and explanation.
ladyada, You just answered an internal debate about whether to use a
feather or a Pi for an intelligent breadboard, which needs frequent fast
but short SPI transactions.
Feather it is!

ladyada, 2020-01-20:
for reaction time, an M4 microcontroller will beat a microcomputer
for throughput, a computer will beat a microcontroller just about every time.

makermelissa, 2020-04-30:
It seems like there isn't really anything else we can do here. Also, SPI was redone using ioctl in PureIO since this issue was created, but I'd imagine the speed is around the same, so closing.

closed 2020-04-30 by makermelissa