adafruit/Adafruit_Blinka #238: Cannot import board

kmac62, 2020-01-27:
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/board.py", line 33, in <module>
    import adafruit_platformdetect.constants.boards as ap_board
ImportError: No module named 'adafruit_platformdetect.constants'
>>> 

Using Python 3.5.3 and used pip3 to install packages. I can import digitalio but I get the same issue when importing busio.

yeyeto2788, 2020-01-27:
Hey @kmac62 I would suggest to update the Adafruit_Python_PlatformDetect first. 
`pip install -U  Adafruit-PlatformDetect --force-reinstall` there was a major change on it.

Regards.

makermelissa, 2020-01-27:
Hi @kmac62, which board and OS are you trying to use this with?

kmac62, 2020-01-27:
> Hey @kmac62 I would suggest to update the Adafruit_Python_PlatformDetect first.
> `pip install -U Adafruit-PlatformDetect --force-reinstall` there was a major change on it.
> 
> Regards.

Thank you, this worked!

closed 2020-01-27 by kmac62