adafruit/Adafruit_Blinka #239: connections issues with mcp2221

machinehistories, 2020-01-27:
Here is the test on the mcp2221 which looks ok
PS C:\Users\jason\python_dev\hello> set BLINKA_MCP2221=1
PS C:\Users\jason\python_dev\hello> python3
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> dir(board)
['G0', 'G1', 'G2', 'G3', 'I2C', 'SCL', 'SDA', 'SPI', '__builtins__', 
'__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
>>> import hid
>>> hid.enumerate()
[{'path': b'\\\\?\\hid#dell093b&col03#5&2363dd55&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1739, 'product_id': 30383, 'serial_number': '9999', 'release_number': 5, 'manufacturer_string': 'Microsoft', 'product_string': 'HIDI2C Device', 'usage_page': 13, 'usage': 14, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_0d62&pid_0a1c&col01#6&c66af87&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 
'vendor_id': 3426, 'product_id': 2588, 'serial_number': '', 'release_number': 13062, 'manufacturer_string': 'DELL Technologies', 'product_string': 'Keyboard', 'usage_page': 65417, 'usage': 16, 'interface_number': -1}, {'path': b'\\\\?\\hid#dell093b&col04#5&2363dd55&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1739, 'product_id': 30383, 'serial_number': '9999', 'release_number': 5, 'manufacturer_string': 'Microsoft', 'product_string': 'HIDI2C Device', 'usage_page': 65280, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#{00001812-0000-1000-8000-00805f9b34fb}_dev_vid&02046d_pid&b015_rev&0013_ef79df93e768&col01#9&286d22c2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd', 'vendor_id': 1133, 'product_id': 45077, 'serial_number': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\u31303', 'release_number'
: 19, 'manufacturer_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\
u31303', 'product_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\u3
1303', 'usage_page': 1, 'usage': 6, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col02#3&36a7043c&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 1, 'usage': 12, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_0d62&pid_0a1c&col02#6&c66af87&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 3426, 'product_id': 2588, 'serial_number': '', 'release_number': 13062, 'manufacturer_string': 'DELL Technologies', 'product_string': 'Keyboard', 'usage_page': 
65417, 'usage': 204, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col03#3&36a7043c&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col04#3&36a7043c&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': 
b'\\\\?\\hid#intc816&col05#3&36a7043c&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col06#3&36a7043c&0&0005#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': 
-1}, {'path': b'\\\\?\\hid#intc816&col07#3&36a7043c&0&0006#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': 
'', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col08#3&36a7043c&0&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col09#3&36a7043c&0&0008#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_0d62&pid_0a1c&col04#6&c66af87&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 3426, 'product_id': 2588, 'serial_number': '', 'release_number': 13062, 'manufacturer_string': 'DELL Technologies', 'product_string': 'Keyboard', 'usage_page': 12, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#intc816&col01#3&36a7043c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd', 'vendor_id': 32903, 'product_id': 2590, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 1, 'usage': 6, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_187c&pid_0550#6&1c2bc3d3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 6268, 'product_id': 1360, 'serial_number': '00.01', 'release_number': 512, 'manufacturer_string': 'Alienware', 'product_string': 'AW-ELC', 'usage_page': 65280, 'usage': 1, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_8087&pid_0ac2#6&2e0188e2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 6268, 'product_id': 1360, 'serial_number': '', 'release_number': 512, 'manufacturer_string': '', 'product_string': '', 'usage_page': 0, 'usage': 0, 'interface_number': -1}, {'path': b'\\\\?\\hid#{00001812-0000-1000-8000-00805f9b34fb}_dev_vid&02046d_pid&b015_rev&0013_ef79df93e768&col02#9&286d22c2&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1133, 'product_id': 45077, 'serial_number': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\u31303', 'release_number
': 19, 'manufacturer_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ
\u31303', 'product_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\u
31303', 'usage_page': 1, 'usage': 2, 'interface_number': -1}, {'path': b'\\\\?\\hid#{00001812-0000-1000-8000-00805f9b34fb}_dev_vid&02046d_pid&b015_rev&0013_ef79df93e768&col03#9&286d22c2&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1133, 'product_id': 45077, 'serial_number': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹㕦搷絡ぜ\u31303', 'rele
ase_number': 19, 'manufacturer_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰
〹㕦搷絡ぜ\u31303', 'product_string': '㝻㔴ㅡ愷ⴰ㐷㍤ㄭ搱ⴰ㙢敦〭愰挰〹
㕦搷絡ぜ\u31303', 'usage_page': 65347, 'usage': 514, 'interface_number': -1}, {'path': b'\\\\?\\hid#dell093b&col01#5&2363dd55&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1739, 'product_id': 
30383, 'serial_number': '9999', 'release_number': 5, 'manufacturer_string': 'Microsoft', 'product_string': 'HIDI2C Device', 'usage_page': 
1, 'usage': 2, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_0d62&pid_0a1c&col03#6&c66af87&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd', 'vendor_id': 3426, 'product_id': 2588, 'serial_number': '', 'release_number': 13062, 'manufacturer_string': 'DELL Technologies', 'product_string': 'Keyboard', 'usage_page': 1, 'usage': 6, 'interface_number': -1}, {'path': b'\\\\?\\hid#dell093b&col02#5&2363dd55&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1739, 'product_id': 30383, 'serial_number': '9999', 'release_number': 5, 'manufacturer_string': 'Microsoft', 'product_string': 'HIDI2C Device', 'usage_page': 13, 'usage': 5, 'interface_number': -1}, {'path': b'\\\\?\\hid#vid_04d8&pid_00dd&mi_02#7&2239931d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 1240, 'product_id': 221, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'Microchip Technology Inc.', 'product_string': 'MCP2221 USB-I2C/UART Combo', 'usage_page': 65280, 'usage': 1, 'interface_number': 2}]
>>> import hid
>>> device = hid.device()
>>> device.open(0x04D8, 0x00DD)
>>> import os
>>> os.environ["BLINKA_MCP2221"]
'1'

Here are the errors when running the suggested commands on the mcp2221. I get similar errors for all of the examples be it for i2c, digital, or analog. I get errors in Mu as well but I thought vscode might be more informative so I grabbed the session from there.

PS C:\Users\jason\python_dev\hello> python3     
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
[... cut ...]
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 66, in gp_set_mode
    current = self._hid_xfer(b'\x61')
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 56, in _hid_xfer
    return self._hid.read(64)
  File "hid.pyx", line 122, in hid.device.read
OSError: read error
>>>

caternuson, 2020-01-27:
Can you post a screen shot of what the MCP2221 HID device is showing up as in Device Manager?

machinehistories, 2020-01-28:
[image]

caternuson, 2020-01-28:
Thanks. Sorry - I should have provided some additional info. There will also be an entry under Human Interface Devices. Can you also expand that branch and screen cap that as well.

What kind of USB port/cable/hubs/etc. are being used to connect the MCP2221?

machinehistories, 2020-01-29:
[image]
It is plugged directly into a usbport on a laptop USB 3.1 Gen 1 Type-A

caternuson, 2020-02-02:
@machinehistories Do you have a USB 2.0 hub or a USB 2.0 port available on the PC?

machinehistories, 2020-02-04:
I still get this issue BLINKA_MCP2221 environment variable set, but no MCP2221 device found even with a usb 2.0 hub however if I use the library shown in the code below it works so I don't think it is the board or my computer. 
```from PyMCP2221A import PyMCP2221A
import time
mcp2221 = PyMCP2221A.PyMCP2221A()

mcp2221.Reset()
time.sleep(1)
mcp2221 = PyMCP2221A.PyMCP2221A()
print('-'*20)
print('MCP2221(A) ADC Test')
print('-'*20)
print(" CTRL+C keys to exit.")

mcp2221.ADC_1_Init()
mcp2221.ADC_2_Init()
mcp2221.ADC_3_Init()
while 1:
    mcp2221.ADC_DataRead()
    print("----------------")
    print ('AD1: 0x{:02x}'.format(mcp2221.ADC_1_data))
    print ('AD2: 0x{:02x}'.format(mcp2221.ADC_2_data))
    print ('AD3: 0x{:02x}'.format(mcp2221.ADC_3_data))
    x = int(mcp2221.ADC_3_data)
    print(x)
    time.sleep(2)```

Here is the out put when I try to run the blinka test file supplied with the blinka library tutorial:
File "basic_test_mcp2221.py", line 3, in <module>
    import digitalio
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\digitalio.py", line 46, in <module>
    if detector.board.microchip_mcp2221:
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\adafruit_platformdetect\board.py", line 309, in microchip_mcp2221
    return self.id == boards.MICROCHIP_MCP2221
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\adafruit_platformdetect\board.py", line 27, in id
    chip_id = self.detector.chip.id
  File "C:\Users\jason\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\adafruit_platformdetect\chip.py", line 38, in id
    raise RuntimeError('BLINKA_MCP2221 environment variable ' + \
RuntimeError: BLINKA_MCP2221 environment variable set, but no MCP2221 device 
found

caternuson, 2020-02-04:
@machinehistories Thanks for that additional info. Can you also try the `BLINKA_MCP2221_HID_DELAY` feature that was added in the 3.8.0 release:
https://github.com/adafruit/Adafruit_Blinka/releases/tag/3.8.0

See here for more info:
https://forums.adafruit.com/viewtopic.php?f=60&t=160414&start=30#p795484

caternuson, 2020-02-04:
For me, that other library also hits the same general issue on the Windows 10 test machine I'm using:
```python
Python 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 23:11:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyMCP2221A import PyMCP2221A
>>> mcp2221 = PyMCP2221A.PyMCP2221A()
>>> mcp2221.Reset()
Reset
>>> mcp2221.ADC_1_Init()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\cater\blinka\lib\site-packages\PyMCP2221A\PyMCP2221A.py", line 293, in ADC_1_Init
    rbuf = self.mcp2221a.read(65)
  File "hid.pyx", line 122, in hid.device.read
OSError: read error
>>>
```

machinehistories, 2020-02-05:
Thank you so much. All the tests are working great. I do want to mention something though,  Adafruit claims the stemma cables are compatible with grove cables. They really aren't unless you considerable shave down the plastic on the stemma connector. Also the single prong on the grove goes up while the double prong on the stemma connector has to face down in order to line up with the proper pins. Again the stemma connector simply doesn't fit in a grove connector unless it is modified. I really think this should be mentioned or stated. Anyways you are good to close this thread - hats off to you.

ladyada, 2020-02-05:
you def can fit the JST cable connector into the proprietary modern Grove, they go upside down but you should not have to cut or modify it at all!

caternuson, 2020-02-05:
Please try the 3.9.0 release of Blinka. That seems to have been a better fix than what was attempted in 3.8.0. More info:
https://forums.adafruit.com/viewtopic.php?f=60&t=160414&start=30#p795894

machinehistories, 2020-02-06:
I must being doing something wrong. Maybe you can look at this video and see what the issue is. The stemma cable obviously works well with stemma port and a stemma port can effectively receive a grove cable but a grove port can’t receive a stemma cable.  Excuse the video quality. Also I have so much respect for your company, many thanks for all the great work. 
https://youtu.be/nUooq3ebBlU

ladyada, 2020-02-06:
doh, we got the grove adapter cable and stemma cable mixed. yes the cable isnt grove compatible without cutting. you *can* use a grove cable in the stemma port but not the other way around. if you cant use the cable - email support@adafruit for a refund. 
if you have more questions please post in the forums

closing this issue as it's gone off topic :)

closed 2020-02-06 by ladyada