adafruit/Adafruit_Blinka #241: Recent support added for RPi I2C Bus 3 breaks use of Adafruit LCD display device on Bus 1

jfathman, 2020-01-28:
I have both I2C Bus 1 and I2C Bus 3 enabled on the RPi 3 B+.

The Adafruit I2C LCD display for the RPi is connected to Bus 1 and appears at device address 0x20.

I have other unrelated I2C devices on Bus 1 and Bus 3 (but not at address 0x20 on Bus 3).

I recently brought up a new SD card for the RPi, and installed the current Adafruit software to support the Adafruit LCD, which includes adafruit-circuitpython-charlcd and other packages documented at Adafruit and many subordinate packages.

With the latest software, the LCD no longer operates.  With debugging, I determined the root cause is the Adafruit python code is now trying to open the LCD display on Bus 3 by default, whereas with previous versions of the Adafruit software it correctly opened the LCD display on Bus 1 by default.

I see there have been more than a hundred recent changes to Adafruit_Blinka, and in looking over the list, I am suspicious of this one (but haven't proved it):

https://github.com/adafruit/Adafruit_Blinka/commit/f3a89113466b63c41e62e6bc75eeb40c535c6f75

Is that the culprit?  If so, will that change stand, even though it breaks the Adafruit LCD module?

If the change must stand for other reasons, is there a way (or can it be added) so the application level LCD display code can specify Bus 1 as an override to the new Bus 3, that is breaking our code?

Thank you very much for the Adafruit free software, and your consideration how best to solve this problem for Adafruit LCD display module users.

ladyada, 2020-01-29:
hi @jfathman   what pins are Bus 3 on?

jfathman, 2020-01-29:
Hi @ladyada, thanks for responding so quickly --

This is the config for our I2C Bus 3 (for other devices, our Adafruit LCD is at 0x20 on Bus 1):

    # configure additional bit-banged I2C bus
    dtoverlay=i2c-gpio,i2c_gpio_sda=16,i2c_gpio_scl=26

Our current workaround is to edit smbus.py to force the bus back to Bus 1:

    /usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py

    def open(self, bus):
        """Open the smbus interface on the specified bus."""
        # Close the device if it's already open.
        if self._device is not None:
            self.close()
        # Try to open the file for the specified bus.  Must turn off buffering
        # or else Python 3 fails (see: https://bugs.python.org/issue20074)
        # workaround to force bus 3 back to bus 1:
        bus = 1
        self._device = open('/dev/i2c-{0}'.format(bus), 'r+b', buffering=0)
        # TODO: Catch IOError and throw a better error message that describes
        # what's wrong (i.e. I2C may not be enabled or the bus doesn't exist).

We would be happy to modify our top level application code to pass down a Bus Number parameter if you need that flexibility in lower level code.

ladyada, 2020-01-29:
mel, what was this supposed tomfix? maybe we can use sda3 and scl3 for bus 3?

makermelissa, 2020-01-29:
It wasn't working at all on bus 3 before. I just changed it trying out busses and if it errors out, it goes to the next one. Perhaps we can change the order it tries it in.

jfathman, 2020-01-29:
If it is possible to make the bus selection explicit at the application level in the long run, that might be better, since auto-detection would always be at the mercy of the variable mix of disparate I2C devices across multiple buses.  You might even imagine a use case where there are Adafruit LCD displays on both Bus 1 device 0x20 and Bus 3 device 0x20, with two separate app instances, each specifying the separate bus.

Seems unlikely perhaps, but in our case, if the Adafruit LCD Display had supported app level I2C bus selection, we might have used it to move the LCD to a non-default bus.  Our objective was to put fast I2C peripherals like the LCD on a faster bus for better performance, and slow I2C peripherals on a slower bus to meet timing constraints.

makermelissa, 2020-01-29:
I like the idea. Let's see how easy it would be to implement something like that without breaking other things. :)

juan-beardo, 2020-04-04:
I ran into a similar issue this week. Had been running on an a previous version of Raspbian (2019-09-26-raspbian-buster) which had been upgraded and updated. The default Python version was 3.5. Had added two (2) additional I2c busses, I2C-3 and I2C-4, based on instructions provided here: https://www.instructables.com/id/Raspberry-PI-Multiple-I2c-Devices/. The /boot/config.txt was edited with

	# Activate additional I2c busses 4,3
	dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=2,i2c_gpio_sda=17,i2c_gpio_scl=27
	dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=2,i2c_gpio_sda=23,i2c_gpio_scl=24

Also had edited /home/pi/.local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/bcm283x/pin.py.  Changed around line 142. Added in additional I2C Ports as configured:

	# This Specific System Config i2c ports!
	i2cPorts = (
		(4, 27, 17), (3, 24, 23), (1, SCL, SDA), (0, D1, D0),
	)
This edit was based on previous research and adapting guidance from https://github.com/adafruit/Adafruit_Blinka/issues/79 and https://github.com/adafruit/Adafruit_Blinka/issues/189, which were referenced from https://forums.adafruit.com/viewtopic.php?f=60&t=160748&p=790992&hilit=raspberry+pi#p790806.

Updated adafruit-blinka (sudo pip3 install --upgrade adafruit-blinka), had to update setuptools (sudo pip3 install --upgrade setuptools) to update respective blinka libraries. Ultimately things broke, specifically use of I2C RGB LCD on I2C-1 failed. Attempted uninstall and reinstall of respective blinka libraries but no luck. Could no longer access I2C RGB LCD as Python3 could not find the device although it was there.

Had spare micro SD card, so did fresh Raspbian OS, Adafruit Blinka, and Circuit Python libraries install. Edited config.txt to add I2C-4 and I2C-3. Default Python now 3.7. Edited /usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pin.py to reflect current I2C bus configuration (note different location now - not /home/pi/.local).

Current system configuration: Raspberry Pi 3B+ with clean install 2020-02-13-raspbian-buster.  I2C-1 (default) hosts DS3231 (addr UU), BME280 (77), RGB LCD PiPlate with RGB Negative 20x4 LCD display (20). I2C-3, I2C-4 each host a single SHT-30 Mesh-protected T/RH sensor (addr 44). All three I2C busses are active and all sensors recognized with sudo i2cdetect -y n, where n=respective I2C bus. Previous Python3 code with adafruit-blinka libraries now working. 

Suspect multiple incompatibilities between OS updates and adafruit-blinka updates based on installation locations. 

Also, Editing /usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pin.py to reflect current I2C bus configuration seems to work fine for my setup.

makermelissa, 2020-04-04:
I think the fix for this is to change https://github.com/adafruit/Adafruit_Blinka/blob/master/src/busio.py#L46 from `pass` to `continue`.

jsharper, 2020-04-12:
I'm using a TCA9548A multiplexer with i2c-mux (https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=141517) on a rpizerow.  It creates 8 virtual i2c buses (eg. /dev/i2c-3 thru /dev/i2c-10).  I'm seeking a way to select the bus by bus number for not just bus 3 but any arbitrary bus (I have an oled display on each of several muxed buses).  The API seems to be pin-based, but that doesn't align with virtual muxed buses.  Could a workaround be a bunch of virtual pin definitions eg. SDA3/SCL3 thru SDA10/SDL10 that can be used to request a particular bus? I think that's what @ladyada may have been suggesting above.

tannewt, 2020-04-24:
> If it is possible to make the bus selection explicit at the application level in the long run, that might be better, since auto-detection would always be at the mercy of the variable mix of [...]

I don't want the CircuitPython API (and therefore Blinka) to have explicit instance awareness because it makes the resulting code less portable. If underlying bus speed is a concern, the implementation should change it's allocation policy based on the `frequency` given to `__init__`.

jsharper, 2020-04-24:
@tannewt I definitely appreciate the simplicity and corresponding portability that your design philosophy drives towards.  I'm curious how you would ideally want CP to handle the use case of multiple i2c buses?  It seems like what you are proposing (no explicit bus instance awareness) is just fundamentally opposed to the idea of there being more than one bus in use, but I'm probably reading too much into what you are saying.  Any clarification would be appreciated.

makermelissa, 2020-04-24:
If it's any help, the way that CircuitPython deals with multiple I2C busses on microcontrollers is allowing the user to specify the SDA and SCL pins and then it grabs the bus from there (See https://circuitpython.readthedocs.io/en/latest/shared-bindings/busio/I2C.html). Blinka tries to emulate this but because of the flexibility provided by Raspberry Pi, there are definitely instances where it is not properly auto-detected.

The way the Blinka auto-detection code is supposed to work is it looks at the list of I2C ports defined and finds the first matching bus number and attempts to initialize it. If initialization fails, it goes to the next matching one and keeps trying until it either successfully initializes the I2C bus or runs out of ports to try.

I think the issue that we're running into is it that there's really a huge number of possible combinations that users can use and we only have a small number of those combinations in the board definition. Currently the only proper way to fix this is to match the board definition I2C ports to what the user has defined, but the issue with that is as soon as Blinka is upgraded in the future, those are overwritten.

Perhaps there's a better system way of detecting what pins the I2C ports are actually using such as looking at /boot/config.txt or something.

jsharper, 2020-04-24:
Not only a huge number of combinations, but also the existence of i2c buses that do not (directly) exist on physical pins at all.

tannewt, 2020-04-25:
@jsharper As Melissa says, multiple I2C busses are handled by instantiating multiple busio.I2C() calls with different pins.

The TCA9548A is a bit trickier. In CircuitPython, the driver would be instantiating with the upstream bus and I'd expect it to make N I2C-like objects available. They can then be passed into whatever driver(s) are on that bus.

If a Linux driver is used it gets a bit trickier. We probably need a way to produce pins based on the device tree and use that to reverse engineer the "internal" connections. I've just started looking into device tree so I'm not sure the best way to map it to the CircuitPython world.

makermelissa, 2020-04-30:
Ok, I created a new library to address the issue. It's available at https://github.com/adafruit/Adafruit_Python_Extended_Bus/releases

It allows you to create an I2C object by passing the Bus ID instead of some pins. Since it extends the busio I2C object, it should be compatible. It only works with Linux boards, but those seem to be the only one having this issue. Closing this for now. If you have any problems, please open an issue in the new repository.

closed 2020-04-30 by makermelissa

jsharper, 2020-05-02:
@makermelissa Thank you!  It works well!

szabolcszijarto, 2021-04-14:
Very useful, works for me too 👍