adafruit/Adafruit_Blinka #242: Neopixels on NVIDIA Jetson

makermelissa, 2020-01-29:
NeoPixel support is crashing on the NVidia Jetson TX2.

See more details here:
https://forums.adafruit.com/viewtopic.php?f=60&t=161551

yeyeto2788, 2020-01-30:
Hey @makermelissa,

I did a quick look at Adafruit-Circuitpython-NeoPixel lib and the error comes from https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L34 which is in calling the `neopixel` module of the board which in this case is only applied for Raspberry (https://github.com/adafruit/Adafruit_Blinka/blob/master/src/neopixel_write.py#L14-L19)

It's there a way to make a common and agnostic `neopixel` module?

Regards.

makermelissa, 2020-01-30:
Maybe try this library: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel_SPI

yeyeto2788, 2020-01-31:
Hey @makermelissa, I have taken a look at the library you mentioned and I noticed the following:

 * Blinka is a requirement for this library and having it as a requirement in blinka is not the best approach.
* This could be integrated into blinka as a agnostic module using the `SPI` of the boards, but it will be like re-writting or copy and pasting code.
 * We could extract the part of neopixel (`neopixel_write.py` and the imported modules on it) since it will only work on raspberry pi devices and let the users know about this deprecation and if they still want to be able to drive neopixels it's just a matter of doing a `pip install Adafruit_CircuitPython_NeoPixel_SPI`

Let me know your thoughts :yum:

FunkyMax, 2020-01-31:
Hey @makermelissa !
Thanks for sharing the neopixel_spi library with me. This should work like a charm on my development board (at least no importing errors anymore) but soon I'd have to move to a smaller carrier board that unfortunately does not offer any SPI interface.
I'll go for a small MC in between my Linux machine and the neopixels for now. However, once the neopixel library receives an update and supports more than just the RasPi, I think lots of folks out there would be delighted :)
Thanks for your support!

ladyada, 2020-01-31:
NVIDIA would have to supply and support the neopixel addition. the chances of that happening are slim - but you should ask them. we cannot do it because we don't have any insight into any internals, DMA or whatever. you could use DotStar LEDs, would look the same and work well.

FunkyMax, 2020-02-01:
Okay, I'll have a look on the DotStar LEDs then. Thanks for the advice!
One suggestion: I think it might be a good idea to rewrite the **Python & CircuitPython** paragraph on top of this [page](https://learn.adafruit.com/adafruit-neopixel-uberguide/python-circuitpython) since it is kind of misleading information since currently the only supported computer is the Raspberry Pi (as @yeyeto2788 mentioned earlier). What do you think?

ladyada, 2020-02-01:
sure, added

ladyada, 2020-10-17:
can use SPI now 
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/neopixels-with-spi

closed 2020-10-17 by ladyada