adafruit/Adafruit_Blinka #245: Orange_PI_2_Plus pin.value issue.

yeyeto2788, 2020-02-05:
Hey guys! 

I'm finally able to create devices using Blinka, but I just got stock on the beginning with the following error.

```console
root@orangepiplus2e:~# python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> pin = board.PD14
>>> pin.value()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 78, in value
    return self._line.get_value()
AttributeError: 'NoneType' object has no attribute 'get_value'
```

Platform detection:
```console
root@orangepiplus2e:~# python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> board.detector.chip.id
'SUN8I'
>>> board.detector.board.id
'ORANGE_PI_PLUS_2E'
>>> 
```

Is there an specific version of `libgpiod` needed in order to use it? 

Regards.

P.D: How can I help on writing a guide for using Blinka on Orange Pi 2 Plus or at least update the actual one?

ladyada, 2020-02-05:
hi you aren't using blinka in the right way for GPIO - check out this guide
https://learn.adafruit.com/circuitpython-essentials/circuitpython-digital-in-out

yeyeto2788, 2020-02-05:
Ups!!! my bad. I though I could use it as in Micropython. 

I just wanted to test the DHT22 sensor as in https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup but first test the functionality of Blinka.

Also I just encountered this using `digitalio.DigitalInOut`:
```console
In [9]: dhtDevice = adafruit_dht.DHT22(pin)

In [10]: dhtDevice.temperature
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-10-bb6e5b54f869> in <module>()
----> 1 dhtDevice.temperature

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in temperature(self)
    225             data returned from the device (try again)
    226         """
--> 227         self.measure()
    228         return self._temperature
    229 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in measure(self)
    179                 pulses = self._get_pulses_pulseio()
    180             else:
--> 181                 pulses = self._get_pulses_bitbang()
    182             #print(len(pulses), "pulses:", [x for x in pulses])
    183 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in _get_pulses_bitbang(self)
    135         """
    136         pulses = array.array('H')
--> 137         with DigitalInOut(self._pin) as dhtpin:
    138             # we will bitbang if no pulsein capability
    139             transitions = []

/usr/local/lib/python3.7/dist-packages/digitalio.py in __init__(self, pin)
     82 
     83     def __init__(self, pin):
---> 84         self._pin = Pin(pin.id)
     85         self.direction = Direction.INPUT
     86 

AttributeError: 'DigitalInOut' object has no attribute 'id'
```

and using the `board.PXX` I get this:
```console
In [1]: import board

In [2]: import adafruit_dht

In [3]: dhtDevice = adafruit_dht.DHT22(board.PC4)

In [4]: dhtDevice.temperature
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
<ipython-input-4-bb6e5b54f869> in <module>()
----> 1 dhtDevice.temperature

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in temperature(self)
    225             data returned from the device (try again)
    226         """
--> 227         self.measure()
    228         return self._temperature
    229 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in measure(self)
    179                 pulses = self._get_pulses_pulseio()
    180             else:
--> 181                 pulses = self._get_pulses_bitbang()
    182             #print(len(pulses), "pulses:", [x for x in pulses])
    183 

/usr/local/lib/python3.7/dist-packages/adafruit_dht.py in _get_pulses_bitbang(self)
    147             dhtval = True   # start with dht pin true because its pulled up
    148             dhtpin.direction = Direction.INPUT
--> 149             dhtpin.pull = Pull.UP
    150             while time.monotonic() - timestamp < 0.25:
    151                 if dhtval != dhtpin.value:

/usr/local/lib/python3.7/dist-packages/digitalio.py in pull(self, pul)
    137             self.__pull = pul
    138             if pul is Pull.UP:
--> 139                 self._pin.init(mode=Pin.IN, pull=Pin.PULL_UP)
    140             elif pul is Pull.DOWN:
    141                 if hasattr(Pin, "PULL_DOWN"):

/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py in init(self, mode, pull)
     45                 if pull != None:
     46                     if pull == self.PULL_UP:
---> 47                         raise NotImplementedError("Internal pullups not supported in libgpiod, use physical resistor instead!")
     48                     elif pull == self.PULL_DOWN:
     49                         raise NotImplementedError("Internal pulldowns not supported in libgpiod, use physical resistor instead!")

NotImplementedError: Internal pullups not supported in libgpiod, use physical resistor instead!
```

Should I open an issue on the dht library and close this one? Or is there a way to tell dht library that I have the resistor on the pin already?

ladyada, 2020-02-05:
yes go ahead - but DHT probably wont work by the way, unless its a very very fast chip.

yeyeto2788, 2020-02-05:
I just take a look at https://github.com/adafruit/Adafruit_Blinka/blob/bd7b612b3f59a62c75a3c884bd2c9cbf8cb9fbae/src/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py#L42-51 there is not way to set a physical resistor.

yeyeto2788, 2020-02-05:
Great! Thanks for the help @ladyada, I'll open the issue on the library itself.

closed 2020-02-05 by ladyada