adafruit/Adafruit_Blinka #249: Python3.4 not compatible with pyftdi>=0.40.0

amrsoll, 2020-02-14:
The [pyftdi] package requires python3.5 and up. This is a major bummer for the Jessie platform that is stuck at python3.4 out-of-the-box.

Please consider to update setup.py with `'Programming Language :: Python :: 3.5'` instead of `Python :: 3.4` so as to not bring confusion.

[pyftdi]: https://pypi.org/project/pyftdi/0.40.5/

makermelissa, 2020-02-14:
Excellent idea

makermelissa, 2020-02-14:
I'm pretty sure the pyftdi was added as a requirement for using the Binho Nova and since it is meant to run on multiple platforms, we should leave this, which means we should go the route of requiring Python 3.5, which I believe is the version that CircuitPython is based off of anyways.

PR #250 (2020-02-14): Update Python required version to 3.5
Fixes #249.
closed 2020-02-14 by makermelissa