adafruit/Adafruit_Blinka #251: Unable to select SPI device

ajstamps, 2020-02-15:
in Adafruit_Blinka/src/adafruit_blinka/microcontroller/generic_linux/spi.py line 90, we are unable to adjust device as it is always set to 0. This created an issue with my setup which has two SPI devices. Is it possible to add the option of choosing device number with the default being 0? I've created this issue in case anyone runs into this issue in the future. 

Device: Raspberry Pi 3
Hats: PiCAN 2 + GPS + Gyro + Accelerometer, 12-Ch 10/12-Bit ADC HAT for Raspberry Pi v3.0
Using SPI
Kept hitting "Could not open SPI device - check if SPI is enabled in kernel!"

ladyada, 2020-02-15:
you can select which SPI port ya want here
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/bcm283x/pin.py#L134

ajstamps, 2020-02-15:
I was attempting to use 
[https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx](https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx)

Specifically the example: mcp3xxx_mcp3008_single_ended_simpletest.py

This kept giving me the error I mentioned above. The full error is:

> Could not open SPI device - check if SPI is enabled in kernel!
> Traceback (most recent call last):
>   File "mcp3xxx_mcp3008_single_ended_simpletest.py", line 21, in <module>
>     print('Raw ADC Value: ', chan.value)
>   File "/usr/local/lib/python3.7/dist-packages/adafruit_mcp3xxx/analog_in.py", line 63, in value
>     return self._mcp.read(self._pin_setting, is_differential=self.is_differential) << 6
>   File "/usr/local/lib/python3.7/dist-packages/adafruit_mcp3xxx/mcp3xxx.py", line 97, in read
>     spi.write_readinto(self._out_buf, self._in_buf)
>   File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 209, in write_readinto
>     return self._spi.write_readinto(buffer_out, buffer_in, out_start, out_end, in_start, in_end)
>   File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 90, in write_readinto
>     self._spi.open(self._port, 0)
> FileNotFoundError: [Errno 2] No such file or directory

Which never uses that library it seems. Any tips on changing it? The example code has not been changed.

ladyada, 2020-02-15:
what's in 'ls /dev/i2c* /dev/spi*'

ajstamps, 2020-02-15:
`/dev/i2c-1  /dev/spidev0.1
`

/boot/config.txt is:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2c1=on
dtparam=spi=on
enable_uart=1
dtoverlay-spi1-3cs
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=mcp3008:spi1-0-present
dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
```

ladyada, 2020-02-15:
you cant share blinka with other devices on CS pin 0

dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=mcp3008:spi1-0-present

needs to be removed

ajstamps, 2020-02-15:
Currently one device is set to CE0 (mcp2515) and the other is set to CE1 (MCP3008). The top one is specified by the installation guide from the manufacturer of the CAN hat which does not use blinka. Removing the bottom one doesn't seem to affect whether the Pi can use the MCP3008.

ladyada, 2020-02-15:
you must remove both

ajstamps, 2020-02-16:
But won't removing

>dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25

cause the CAN hat to fail? I'm not currently able to test it, is the Pi able to detect this information without specifying it?

ladyada, 2020-02-16:
correct, it isnt compatible with blinka

ajstamps, 2020-02-16:
I should've been more specific about the two boards. I know that the CAN hat doesn't use blinka, it uses python-can. One part of the hats installation guide says that you need to add that line in /boot/config.txt. This seemed to reserve an SPI port for the CAN controller that blinka was trying to connect to, which made blinka think that SPI wasn't enabled thus the error. When it in fact can, it just needed to switch the device argument from 0 to 1 in spi.py. 

I just tested leaving the device in spi.py at 0 and 1 while leaving in the dtoverlay for CAN. It will fail at 0 (default) but succeeds to read the analog inputs at 1. I think adding the option to select which device as a default argument when creating the SPI bus can mitigate this issue. 

Anyways, this is a fringe case. I'm not sure how many people would use a SPI device that blinka can use with a device that blinka can't use and run into this, but figured I'd open something to help them if they run into it.

ladyada, 2020-02-16:
correct, you can't share the same SPI with blinka and the kernel! we dont even recommend it when there's another CS pin in use, use SPI 3 instead!

caternuson, 2020-03-06:
@ajstamps Can we close this? Seems like the general issue has at least been understood.

Dmytro-Bez, 2020-03-20:
I also have a problem setting up SPI.I would like precise instructions when setting up SPI. Please help in solving this problem.

tannewt, 2020-03-20:
@Dmytro-Bez Please post to the support forum here: https://forums.adafruit.com/viewforum.php?f=60

makermelissa, 2020-04-24:
First of all, if you have something like: `/dev/spidev0.1`, that means is uses SPI bus 0 and the Chip Select 1. Since we usually use GPIO's for CircuitPython devices, the Chip select isn't so important.

Secondly, when updated SPI recently,  I quietly added a feature to allow passing a tuple in the boards definition so you to specify both bus and device, so this shouldn't be an issue anymore.

closed 2020-04-24 by makermelissa

Dmytro-Bez, 2020-04-24:
It's been a month - everything works ...