adafruit/Adafruit_Blinka #252: adafruit ft232h windows 10 setup, "Board not supported None"

jsaksris, 2020-02-17:
I have followed this tutorial 
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows
but at the end import board gives me this error 
[image]

import usb seems fine
[image]

I'm working on Powershell with admin privilege.

Thank you for your help

ladyada, 2020-02-17:
hi make sure you have done the environment variable setting

jsaksris, 2020-02-17:
Works fine on cmd... should leave this open in case someone want to share how to set environment in powershell?

ladyada, 2020-02-17:
please tell us how to do it and we'll add to the tutorial :) we dont use powershell :D

makermelissa, 2020-02-17:
Oh yeah, I ran into this a month ago and just ended up using command. :) I'm definitely interested if you find a way to get it working.

caternuson, 2020-02-17:
At powershell prompt, try: `$env:BLINKA_FT232H=1`

makermelissa, 2020-02-24:
@caternuson, that worked. Want to add to your guide?

caternuson, 2020-02-24:
@makermelissa sure thing, done:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows#set-environment-variable-3-30

closed 2020-02-24 by makermelissa