adafruit/Adafruit_Blinka #254: "Magic Commands" in Jupyter Notebook not supported to set environment variable

ShawnHymel, 2020-02-20:
I'm trying to work with Blinka in Jupyter Notebook, and it seems to work only if I set the required environment variable in the command prompt before running the kernel. Since I'm running Anaconda on Windows, this would look like:
```
set BLINKA_NOVA=1
jupyter notebook
```
If I do this, I can `import board` with no problem.

If I try to run the magic command or shell command in a Jupyter Notebook cell, it does not seem to work:
`!set BLINKA_NOVA=1`
or
`%env BLINK_NOVA=1`
I can see that the variable has been set by calling `!set` or `%env`. However, when I run `import board` I get the `Board not supported None` error message. I'm not sure if this is a bug in Blinka or Jupyter Notebook, but any help would be appreciated!

tannewt, 2020-02-20:
What device are you running Jupyter on?

ShawnHymel, 2020-02-20:
Windows 10 (x64) machine with Chrome as the interface.

tannewt, 2020-02-21:
@willingc or @makermelissa might have an idea.

willingc, 2020-02-21:
@ShawnHymel If you try this from a notebook cell before importing the board:

```
%%bash
export BLINK_NOVA=1
```

Does that work? I'm guessing it is not a bug on either the blinka or the notebook, but how environments work and Anaconda.

ShawnHymel, 2020-02-21:
@willingc I get the following error when I run that cell: `Couldn't find program: 'bash'`

I'm on Windows, so I'm guessing I'm getting that error because I'm not running bash. The best alternative I could find is:

```
%%cmd
set BLINKA_NOVA=1
set
```
This seems to set the environment variable for that cell only. If I run

```
%%cmd
set
```
In another cell, the BLINKA_NOVA variable seems to have disappeared (and, as a result, I can't run `import board`).

makermelissa, 2020-02-21:
Perhaps try `%set_env`? Here's something I found useful: https://ipython.readthedocs.io/en/stable/interactive/magics.html

willingc, 2020-02-21:
Or alternatively `%env BLINKA_NOVA=1` @makermelissa gave you the best doc page.

makermelissa, 2020-02-21:
@ShawnHymel, Another option that might work is to search for `env` in your run bar on Windows and go to Edit the system environment variable and set a system variable through there. Then it should be system wide. You may need to reboot after doing that.

ShawnHymel, 2020-02-21:
!set, %set_env, and %env seem to successfully set the variable (I can read the list in a different cell). However, I still get `Board not supported None` when I try to `import board`, which tells me something's not reading those particular environment variables.

I can set the environment variable globally, including calling `set BLINKA_NOVA=1` in the Anaconda prompt before launching Jupyter Notebook. I was hoping to find a way to do it from within Jupyter Notebook (as, I believe it can be done in Mac/Linux). This might be a Jupyter Notebook problem and not a Blinka problem :-/

willingc, 2020-02-22:
I would recommend following @makermelissa's suggestion to set the environment variable globally. The reason that it can be done in Mac/Linux is that it uses a different type of shell than Windows. Here's an article from Jake Vanderplas' book that explains a bit more: https://jakevdp.github.io/PythonDataScienceHandbook/01.05-ipython-and-shell-commands.html

It's not a bug but a limitation of the Windows operating system. If you are using Windows Subsytem Linux, I believe that the command will work (though I haven't tried it since I don't have a Windows system handy). Good luck @ShawnHymel. FYI you can ask questions about   Jupyter in the Jupyter Discourse site too (http://discourse.jupyter.org).

ShawnHymel, 2020-02-22:
Ah, good to know it's a Windows thing. I'll set it globally then. Thanks!

closed 2020-02-22 by ShawnHymel