adafruit/Adafruit_Blinka #257: Using ServoKit on additional i2c buses

A3throd, 2020-02-25:
I cannot seem to make ServoKit working by creating i2c-3 with GPIO4 and GPIO5 neither by trying i2c-4 with GPIO6 and GPIO7 on the Raspberry Pi 4  B. The only difference is by creating it on Bus 3 it seems to prioritize that instead of Bus 1 (the designated i2c bus). Is there any way I could modify the scripts so that it works with the CircuitPython libraries?

A3throd, 2020-02-25:
I intend to use a FeatherWing motor control board on i2c-1 and it would make things easier I reckon if the servo controller could be gotten up and running on an additional i2c bus.

ladyada, 2020-02-25:
you can share the i2c bus - just make sure the two boards have unique I2C addresses. duplicate busses are only needed if you have an addr conflict

caternuson, 2020-03-06:
There are HAT versions of the motor:
https://www.adafruit.com/product/2348
and servo control boards:
https://www.adafruit.com/product/2327
which are easier to use with a Pi than a FeatherWing.

And there are two KIT libraries if you are using motor:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit
or servo:
https://github.com/adafruit/Adafruit_CircuitPython_ServoKit

They both allow for specifying a specific I2C bus:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit/blob/master/adafruit_motorkit.py#L66
https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/blob/master/adafruit_servokit.py#L79
but that doesn't seem necessary here. Easiest is to just use the default I2C bus and make sure there are no address conflicts.

Closing. This doesn't look like a repo issue.

closed 2020-03-06 by caternuson