adafruit/Adafruit_Blinka #258: Is remote GPIO supported?

hyansuper, 2020-02-29:
For me, adafruit-blinka is a great supplement to gpiozero to control i2c/spi devices, but I really need [remote GPIO functionality like that of gpiozero](https://gpiozero.readthedocs.io/en/stable/recipes_remote_gpio.html) so I can control multiple boards in one script. Do you have plan to support remote GPIO for i2c/spi devices?
thanks!

pdp7, 2020-02-29:
@hyansuper that is very interesting.  I was not aware of that functionality.  I'll have to take a look at how it could be implemented for this library.

pdp7, 2020-02-29:
@makermelissa any thoughts about remote GPIO for blinka?

ladyada, 2020-02-29:
no plans for remote GPIO - this library is for native control of hardware only :)

closed 2020-02-29 by ladyada

hyansuper, 2020-03-01:
@pdp7 For your information, it seems that remote GPIO is a low level functionality of Raspberry pi, which can be enabled via command: sudo raspi-config -> interfacing options -> remote GPIO. 
If blinka depends on [pigpio](http://abyz.me.uk/rpi/pigpio/pigpiod.html) like gpiozero, maybe it can be implemented. but I don't know about other adafruit custom boards
https://gpiozero.readthedocs.io/en/stable/remote_gpio.html