adafruit/Adafruit_Blinka #259: RuntimeError with PulseIn on Raspberry pi

hyansuper, 2020-03-03:
This happens on my  Raspberrypi zero(Buster) and Raspberrypi 3(stretch):
```
Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board, pulseio
>>> p = pulseio.PulseIn(board.D14)
/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: error while loading shared libraries: libgpiod.so.2: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 64, in __init__
    message = self._wait_receive_msg()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 80, in _wait_receive_msg
    raise RuntimeError("Timed out waiting for PulseIn message")
RuntimeError: Timed out waiting for PulseIn message
```
I first notice this error when I was using ultrasonic sensor, see https://github.com/adafruit/Adafruit_CircuitPython_HCSR04/pull/16

yeyeto2788, 2020-03-03:
@hyansuper did you try with other pin? I just noticed this on your error log `libgpiod_pulsein: error while loading shared libraries: libgpiod.so.2: cannot open shared object file: No such file or directory`

hyansuper, 2020-03-04:
I don't have UART enabled. I also tried many other pins, same error. Do I have to manually install libgpiod? I tried to install libgpiod with [this script](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh) but failed. below is the log running the script
```
Installing build requirements - this may take a few minutes!

Hit:1 http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian buster InRelease
Hit:2 http://mirrors.tuna.tsinghua.edu.cn/raspberrypi buster InRelease
Reading package lists... Done
N: Ignoring file 'raspi.list.b' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
Reading package lists... Done
Building dependency tree
Reading state information... Done
autoconf is already the newest version (2.69-11).
autoconf-archive is already the newest version (20180313-1).
automake is already the newest version (1:1.16.1-4).
build-essential is already the newest version (12.6).
git is already the newest version (1:2.20.1-2+deb10u1).
libtool is already the newest version (2.4.6-9).
pkg-config is already the newest version (0.29-6).
python3 is already the newest version (3.7.3-1).
python3-dev is already the newest version (3.7.3-1).
python3-setuptools is already the newest version (40.8.0-1).
swig3.0 is already the newest version (3.0.12-2).
wget is already the newest version (1.20.1-1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
N: Ignoring file 'raspi.list.b' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20200212-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
N: Ignoring file 'raspi.list.b' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
Cloning libgpiod repository to /tmp/libgpiod.kLk2

Cloning into '.'...
remote: Enumerating objects: 1130, done.
remote: Counting objects: 100% (1130/1130), done.
remote: Compressing objects: 100% (1040/1040), done.
remote: Total 5081 (delta 735), reused 114 (delta 87)
Receiving objects: 100% (5081/5081), 850.97 KiB | 4.00 KiB/s, done.
Resolving deltas: 100% (3459/3459), done.
Building libgpiod

autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: creating directory autostuff
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'autostuff'.
libtoolize: copying file 'autostuff/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:61: installing 'autostuff/ar-lib'
configure.ac:61: installing 'autostuff/compile'
configure.ac:64: installing 'autostuff/config.guess'
configure.ac:64: installing 'autostuff/config.sub'
configure.ac:42: installing 'autostuff/install-sh'
configure.ac:42: installing 'autostuff/missing'
bindings/cxx/Makefile.am: installing 'autostuff/depcomp'
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
[... 105 lines cut ...]
checking for ppoll... yes
checking for realpath... yes
checking getopt.h usability... yes
checking getopt.h presence... yes
checking for getopt.h... yes
checking dirent.h usability... yes
checking dirent.h presence... yes
checking for dirent.h... yes
checking sys/poll.h usability... yes
checking sys/poll.h presence... yes
checking for sys/poll.h... yes
checking sys/sysmacros.h usability... yes
checking sys/sysmacros.h presence... yes
checking for sys/sysmacros.h... yes
checking linux/gpio.h usability... yes
checking linux/gpio.h presence... yes
checking for linux/gpio.h... yes
checking linux/version.h usability... yes
checking linux/version.h presence... yes
checking for linux/version.h... yes
configure: error: "libgpiod needs linux headers version >= v5.5.0"
cp: cannot stat 'bindings/python/.libs/gpiod.so': No such file or directory
cp: cannot stat 'bindings/python/.libs/gpiod.la': No such file or directory
cp: cannot stat 'bindings/python/.libs/gpiod.a': No such file or directory
```

makermelissa, 2020-04-29:
Hi, have you been able to solve this? If not please try running `sudo apt install libgpiod2`.

hyansuper, 2020-05-07:
Solved after I run `sudo apt install libgpiod2`
Thank you!

closed 2020-05-07 by hyansuper

Firesphere, 2021-04-28:
If you are encountering this problem on an older version of Raspberry Pi OS, download libgpiod2 from the Debian website, the one for Buster, and install that one. LibGPIOD2 is not available on versions before buster, so you'll have to do it the hacky way

CesMak, 2022-03-12:
I encounter this problem only if I run my python program from a systemd service. Any ideas why?

Firesphere, 2022-03-12:
The systemd service runs as possibly the wrong user, who does not have the libraries required installed. What's your systemd service look like?

CesMak, 2022-03-13:
[Unit]
Description=dht logging service
After=multi-user.target

[Service]
Type=idle
ExecStart=/usr/bin/python3 /home/pi/dht_logger.py
Environment=PYTHONUNBUFFERED=1

[Install]
WantedBy=default.target

Firesphere, 2022-03-13:
Try adding `User={User that works when running the command manually]` to the Service part

CesMak, 2022-03-13:
Wow thanks for that fast answer it works! Perfect.
Do you know if I also need my service file here:
sudo cp /etc/systemd/system/dht_logger.service **/lib/systemd/system/dht_logger.service**
I read some hints on that but I am not sure for what that would be required?

Firesphere, 2022-03-13:
No, you should not need to do that, if you have enabled the timer (If you use a timer) or the service, via `systemctl enable servicename`, a symlink to the required location (in your case it's added to default.target), which is enough.

tschetschpi2, 2022-03-25:
I was running into this too as root. The comment about systemd service file needing a username added made me try it as a non-root user and it works now. Thank you!

RandomTypek, 2024-01-29:
Any idea how to fix this on arch? Running arch linux arm on my rpi, installing libgpiod doesn't work and libgpiod2 doesn't exist. I even tried converting the libgpiod2 deb package with debtap, didn't work either

makermelissa, 2024-01-29:
> Any idea how to fix this on arch? Running arch linux arm on my rpi, installing libgpiod doesn't work and libgpiod2 doesn't exist. I even tried converting the libgpiod2 deb package with debtap, didn't work either

You could try compiling from source: https://github.com/adafruit/libgpiod_pulsein

If you need libgpiod, that can also be compiled from source. See https://learn.adafruit.com/adding-a-single-board-computer-to-blinka/software-setup#installing-libgpiod-from-source-3058818.