adafruit/Adafruit_Blinka #260: Using multiple FT232H devices on the same PC

MoTheItGuy, 2020-03-04:
Hi there,
So I need some help.  We are trying to use 2 (or more) Adafruit FT232H boards (I bought 10 so far for use in a project) each to control several 5m strips of Neopixels as each FT232H only supports 1 SPI output.

The problem is that the blinka library only seems to support 1 board.

When I connect 2, I get this message.
pyftdi.usbtools.UsbToolsError: 2 USB devices match URL 'ftdi:///1'

I have already used the utility to name the 2 boards differently, but can't seem to specify that anywhere.

I imagine I'm not the only person that's ever wanted to use more than 1 of these awesome little boards.

Any help is greatly appreciated.

Thanks!
Mo

caternuson, 2020-03-06:
Correct, the current implementation only supports a single FT232H. And you can't really specify which one, it just grabs the first one found. Are you using the latest version of the library? That search URL looks like it's from an older version. It won't fix the multi-board issue, but you should at least be able to use one of the FT232Hs.

I'd suggest changing the name of this issue to "Add support for multiple FT232H".

mpratt14, 2020-07-07:
@MoTheItGuy Just so you understand, this is much more than a change to the software. The actual problem is that the two devices have the exact same USB vendor and product ID, so the computer sees the two boards as the same device. 

The actual solution is to program the EEPROM on the board to have a different product ID ([using the FT_PROG tool from FTDI](https://www.ftdichip.com/Support/Utilities.htm#FT_PROG)), and then handle identifying it here in the library...

See page 47 of the datasheet:
https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232H.pdf

rknobel, 2020-07-28:
Any progress for this issue? Need help with running two or more FT232H boards on the same Linux PC.

mpratt14, 2020-07-28:
2 things have to be handled (but I think the first one is ok already in this library, but not in pyftdi)

USB IDs:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/chip.py#L72

https://github.com/eblot/pyftdi/search?q=0x6014&unscoped_q=0x6014
(for each file here ^: adding your new product ID to the existing somehow)

the hardware url:
https://github.com/eblot/pyftdi/blob/master/pyftdi/doc/urlscheme.rst

because the default url or automatically generated url will point to only one of the boards, you have to form the urls manually with vendor and product IDs instead of strings like "232h"

identifying both boards in Adafruit Blinka requires:
1. reprogram the EEPROM of one of the FT232H to a Product ID that does not match any other device in pyftdi
2. edit the pyftdi library so that the NEW unique product ID refers to the FT232H along with the original
3. when you call/instantiate each board, there must be a unique url with the vendor and product IDs used in the .configure() function

It's possible that you can get away with programming the product ID to 0x6010 which is the FT2232H...but someone would have to test that

Either way, understand that most of the changes would occur in the pyftdi library, not here.

caternuson, 2020-07-28:
Here's a basic example of enabling FT232H and then showing the available pins:
```python
$ export BLINKA_FT232H=1
$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> dir(board)
['C0', 'C1', 'C2', 'C3', 'C4', 'C5', 'C6', 'C7', 'D4', 'D5', 'D6', 'D7', 'I2C', 'MISO', 'MOSI', 'SCK', 'SCL', 'SCLK', 'SDA', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
>>>  
```

What would the expected behavior of the above be for the case of multiple FT232H's?

mpratt14, 2020-07-28:
@caternuson I would say no change. Pin mapping is the same for both of them since it's the same device. The change comes with instantiation and configuration, AKA creating an object for each board. I would do it like this for example (if using SPI). This is something the user has to handle for the most part. The only thing this library can do is have a set range of product IDs that can be automatically detected and refer to FT232H configuration/url. It all relies on one of the FT232H being reprogrammed with a different ID, and the pyftdi library being able to identify that ID.

(just a draft, I cannot test this)

```
import board
import busio

baudrate = 80000000

BOARD1 = BOARD2 = busio.SPI(board.SCLK, board.MOSI, board.MISO)

BOARD1.configure("ftdi://ftdi:0x6014/1", baudrate)
BOARD2.configure("ftdi://ftdi:0x????/1", baudrate)

while not BOARD1.try_lock(): pass
while not BOARD2.try_lock(): pass

BOARD1.write([0xff])
BOARD2.write([0xff])
```

mpratt14, 2020-07-28:
Actually pyftdi library might not require changes after all

https://eblot.github.io/pyftdi/installation.html#custom-usb-vendor-and-product-ids

mpratt14, 2020-07-29:
Also, it looks like forming URLs based on serial number is not possible unless you program them. I get all FFFF on my board

took code from mockusb.py
https://github.com/eblot/pyftdi/blob/master/pyftdi/tests/mockusb.py

```
>>> from pyftdi.usbtools import UsbTools
>>> UsbTools.find_all([(0x403, 0x6014)])[0]
(UsbDeviceDescriptor(vid=1027, pid=24596, bus=0, address=255, sn='\uffff\uffff\uffff\uffff\uffff\uffff\uffff\uffff', index=None, description='\uffff\uffff\uffff\uffff\uffff\uffff'), 1)
>>> devdesc, _ = UsbTools.find_all([(0x403, 0x6014)])[0]
>>> dev = UsbTools.get_device(devdesc)
>>> serialn = UsbTools.get_string(dev, dev.iSerialNumber)
>>> print(serialn)
￿￿￿￿￿￿￿￿
>>> serialn.encode('utf-16be')
b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
```

mpratt14, 2020-07-29:
@MoTheItGuy @rknobel it may be possible to use different URLs based on the bus and address. Plug in both of your FT232H and do this and compare the two outputs, see if anything is different between them.

```
from pyftdi.usbtools import UsbTools
UsbTools.find_all([(0x403, 0x6014)])[0]
UsbTools.find_all([(0x403, 0x6014)])[1]
```

rknobel, 2020-07-29:
@mpratt14 Here is the outout:
(UsbDeviceDescriptor(vid=1027, pid=24596, bus=1, address=6, sn='\uffff\uffff\uffff\uffff\uffff\uffff\uffff\uffff', index=None, description='\uffff\uffff\uffff\uffff\uffff\uffff'), 1)
(UsbDeviceDescriptor(vid=1027, pid=24596, bus=1, address=7, sn='\uffff\uffff\uffff\uffff\uffff\uffff\uffff\uffff', index=None, description='\uffff\uffff\uffff\uffff\uffff\uffff'), 1)

rknobel, 2020-07-29:
So, based on this [link](https://github.com/eblot/pyftdi/blob/master/pyftdi/doc/urlscheme.rst), I can open different connection based on the address.
I am still confused about how to override internal calls in board.py file.

Here is the snippet of my code:
 `import os
import time
os.environ["BLINKA_FT232H"]="1"
import board
import digitalio
#check that the adafruit env variable is se
MANDATORY_ENV_VARS = ["BLINKA_FT232H"]
for var in MANDATORY_ENV_VARS:
    if var not in os.environ:
        raise EnvironmentError("Failed beca
        print( "Exit and restart after sett
    else :
        print( "Environment BLINKA_FT232H s
# setup some initial ports from board
led = [ digitalio.DigitalInOut(board.C0),
        digitalio.DigitalInOut(board.C1),
        digitalio.DigitalInOut(board.C2),
        digitalio.DigitalInOut(board.C3),
        digitalio.DigitalInOut(board.C4),
        digitalio.DigitalInOut(board.C5),
        digitalio.DigitalInOut(board.C6),
        digitalio.DigitalInOut(board.C7) ]`

Where do I put a URL for the module I want to open a connection? There is no call to open connection.
Thanks for all the help.

mpratt14, 2020-07-29:
Yeah, pyftdi and Blinka syntax are different. They have not provided a way to set the URL or open the port manually.

Now we are seeing the actual required change that needs to take place. The problem is the Pin class always sets the url the same every time it is imported. According to the pyftdi docs, url must be configured before port opens...however Adafruit Blinka opens the port as soon as you import digitalio...
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/ft232h/pin.py#L24

It takes redoing that classes init in order to reconfigure the url.

I just realized you will need a different script for each board, because the url configuration is in the "board" object...
The library doesn't support instantiating different boards in the same script, even if there is no conflicts...

Note that bus and address in the url must be hexadecimal, for me the address was 255 so url had ff

BTW the section you wrote to make sure BLINKA_FT232H is set is a waste. The variable is required for `import board`, nothing else.

```
import os
import time
os.environ["BLINKA_FT232H"]="1"
import board
import digitalio

from pyftdi.i2c import I2cController

i2c = I2cController()

i2c.configure("ftdi://ftdi:232h:0:ff/1")

board.pin.Pin.ft232h_gpio = i2c.get_gpio()

#setup some initial ports from board
led = [ digitalio.DigitalInOut(board.C0),
digitalio.DigitalInOut(board.C1),
digitalio.DigitalInOut(board.C2),
digitalio.DigitalInOut(board.C3),
digitalio.DigitalInOut(board.C4),
digitalio.DigitalInOut(board.C5),
digitalio.DigitalInOut(board.C6),
digitalio.DigitalInOut(board.C7) ]

...
...
...

```

script for one board would have
`i2c.configure("ftdi://ftdi:232h:1:6/1")`

script for the other board would have
`i2c.configure("ftdi://ftdi:232h:0:7/1")`

rknobel, 2020-07-30:
@mpratt14 I was not sure if I can use your code as-is, or I need to change init in digitalio? 
I tried to run your code as-is:
With 1 device connected, I get these errors:
`Traceback (most recent call last):
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/usbtools.py", line 314, in parse_url
    desc, _ = candidates[idx]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/rknobel/dev/hwvv/pyhwvv/driver/USB_Serial_GPIO/test_code.py", line 14, in <module>
    i2c.configure("ftdi://ftdi:232h:1:6/1")
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/i2c.py", line 509, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 593, in open_mpsse_from_url
    devdesc, interface = self.get_identifiers(url)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 374, in get_identifiers
    cls.DEFAULT_VENDOR)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/usbtools.py", line 318, in parse_url
    urlstr)
pyftdi.usbtools.UsbToolsError: No USB device matches URL ftdi://ftdi:232h:1:6/1`

If I have two devices connected I get this:
`Traceback (most recent call last):
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/usbtools.py", line 314, in parse_url
    desc, _ = candidates[idx]
IndexError: list index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/rknobel/dev/hwvv/pyhwvv/driver/USB_Serial_GPIO/test_code.py", line 14, in <module>
    i2c.configure("ftdi://ftdi:232h:1:6/1")
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/i2c.py", line 509, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 593, in open_mpsse_from_url
    devdesc, interface = self.get_identifiers(url)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 374, in get_identifiers
    cls.DEFAULT_VENDOR)
  File "/home/rknobel/.local/lib/python3.6/site-packages/pyftdi/usbtools.py", line 318, in parse_url
    urlstr)
pyftdi.usbtools.UsbToolsError: No USB device matches URL ftdi://ftdi:232h:1:6/1`

At the end, all I need to do from FT232 board is to control seven Didital IOs on C side. Do I have to use Blinka? I should be able to just control pin logic using pyFTDI library.

caternuson, 2020-07-30:
> At the end, all I need to do from FT232 board is to control seven Didital IOs on C side. Do I have to use Blinka? I should be able to just control pin logic using pyFTDI library.

You generally only need Blinka if you want to use the FT232H with Adafruit CircuitPython Libraries. An example FT232H setup would be something like:
```
[PC USB]===[FT232H]---[I2C sensor / SPI sensor / GPIO]
```
and you are using the Adafruit library that goes along with the I2C/SPI sensor. All such Python libraries are being written to the CircuitPython hardware standard. Blinka is the "shim" that provides that standard hardware interface for various platforms.

If your use case is not using any sensor or hardware that is relying on an Adafruit CircuitPython library, then you don't need Blinka. You could just use pyftdi directly. Or any other means of directly interfacing with the FT232H.

caternuson, 2021-05-05:
Closing this. We are going to add support for FT2232H, but not multiple FT232H's (nor multiple FT2232H's). There's also support for specifying the MPSEE URL being added. So you could have multiple FT232H's attached, and specify one to use, but can only use one at any given time.

If you need more pins, the FT2232H will be the option.

closed 2021-05-05 by caternuson

cls5times, 2024-05-29:
> Closing this. We are going to add support for FT2232H, but not multiple FT232H's (nor multiple FT2232H's). There's also support for specifying the MPSEE URL being added. So you could have multiple [...]

Hello,

Has this been implemented? (I.e., having multiple FT232H's plugged-in but only using one at a time via MPSEE URLs?)
Is there an example on how to do so?

Thanks!