adafruit/Adafruit_Blinka #262: RuntimeError: I2C slave address was NACK'd

fgallaire, 2020-03-07:
Trying to use the Adafruit ATECC608 Breakout with the Adafruit MCP2221A Breakout on Linux.

siddacious, 2020-03-07:
@brentru maybe you'll have some insight here?
The atecc is quirky about when it acknowledges its presence, so I'm not surprised that there is an issue with getting it to respond

fgallaire, 2020-03-07:
hello @siddacious, I had no issue with the atecc breakout when used from the Feather STM32F405

siddacious, 2020-03-07:
Thanks for the info. The i2c implementation for CircuitPython is totally different than the one in Blinka so it doesn't surprise me that you're seeing different results

fgallaire, 2020-03-07:
OK, how can we fix it ?

jerryneedell, 2020-03-07:
just FYI -- I tried running the atecc_simpltest on an RaspberyPi4 with Blinka and it works fine.

jerryneedell, 2020-03-07:
BUT -- I get the same failure the I try with an mcp2221a on MAcOS
```
jerryneedell:~/projects/blinka$ python3 atecc_simpletest.py 
Traceback (most recent call last):
  File "atecc_simpletest.py", line 10, in <module>
    atecc = ATECC(i2c)
  File "/usr/local/lib/python3.6/site-packages/adafruit_atecc/adafruit_atecc.py", line 132, in __init__
    self._i2c_device = I2CDevice(self._i2c_bus, address)
  File "/usr/local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 68, in __init__
    i2c.writeto(device_address, b'')
  File "/usr/local/lib/python3.6/site-packages/busio.py", line 87, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/mcp2221/i2c.py", line 13, in writeto
    self._mcp2221.i2c_writeto(address, buffer, start=start, end=end)
  File "/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 274, in i2c_writeto
    self._i2c_write(0x90, address, buffer, start, end)
  File "/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 205, in _i2c_write
    raise RuntimeError("I2C slave address was NACK'd")
RuntimeError: I2C slave address was NACK'd
jerryneedell:~/projects/blinka$ 
```

fgallaire, 2020-03-07:
@jerryneedell thanks for your tests, as it works on RaspberyPi4 there's hope to fix that bug !

fgallaire, 2020-03-09:
@caternuson seems you are the only one with expertise on the MCP2221 I2C code, what do you think about this issue ?

ladyada, 2020-03-09:
@fgallaire the ATECC is  a particularly weird I2C chip due to its sleep mode, we have no ETA when we may look at why it isnt working with MCP2221

caternuson, 2020-03-09:
Self assigned for tracking. Will try and take a look at some point.

wildestpixel, 2020-03-21:
Had similar NACK issue with BME280 on MCP2221A

Gave the implicit address of the i2c device as follows : 

`bme280 = adafruit_bme280.Adafruit_BME280_I2C(board.I2C(), 0x76)`

And the code works

Scanned for the address as follows : 

```
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> scan = i2c.scan()
>>> for addr in scan:
...     print("Adress = {0:3d}, Hex = {1}".format(addr, hex(addr)))
...     
...     
...
```
Make sure you get your indentation on the print function

This won't necessarily solve your issue, but fixed mine if its worth consideration

bergold, 2020-08-31:
I have a similar issue with the `FT232H` and the `BNO055` on macOS.
I'm getting a `pyftdi.i2c.I2cNackError: NACK from slave` error when trying to instantiate the `adafruit_bno055.BNO055_I2C` class.

fgallaire, 2021-01-13:
This issue is still pending, my project is stuck by it since a 10 months !

caternuson, 2021-01-14:
Did some investigating today. No solution, so just dumping findings.

Suspecting something with the ATECC's sleep/wake behavior seems most likely at this point. The CircuitPython driver currently is using an I2C scan to wake it up. I *think* that's a bit of hack, since the actual wake sequence is...strange. See wake command trace here:
http://ww1.microchip.com/downloads/en/Appnotes/Atmel-8847-CryptoAuth-SWI-I2C-Seamless-Debugging-Saleae-Logic-Analyzer-ApplicationNote.pdf
The description and trace don't really match. Is that long SCL low time needed? Or is it only SDA that matters?

The use of scan ends up working on native boards as well as RPi. Here's the I2C traffic on an Itsy M4 with the line:
```python
atecc = ATECC(i2c)
```
[image]
That's four loops of an I2C scan before the ATECC finally ACK's and is considered woken up. Then another scan before finally initing the sensor. Zooming in towards ends:
[image]
and showing ~2.8ms between the ACK in the scan and the start of I2C comms to init the sensor.

With the MCP2221, the I2C scan is much slower (there's a hardwired 1ms sleep in the driver). The ATECC does ACK on the scan:
[image]
but it's like ~250ms later until the end of the scan.

Is that too long and it went back to sleep? Is the specific I2C traffic pattern generated by the MCP2221 not good enough to serve as a wake up?

caternuson, 2021-01-14:
@fgallaire blocking many ATECC chips? Or other chips?

@wildestpixel @bergold The posts above mentioning other sensors and interface boards should be in separate issues.

fgallaire, 2021-01-15:
@caternuson I was talking about @wildestpixel @bergold chips, it's the same bug for all of us, only one issue is the best way to go.

caternuson, 2021-01-15:
@fgallaire Please try the 1.2.3 release of the ATECC library:
https://github.com/adafruit/Adafruit_CircuitPython_ATECC/releases/tag/1.2.3

fgallaire, 2021-01-16:
@caternuson works well for me, what a great job !
Thanks a lot

fgallaire, 2021-01-16:
The bug may however reappear randomly. More feedback later.

fgallaire, 2021-01-16:
After running the second example:
https://github.com/adafruit/Adafruit_CircuitPython_ATECC/blob/master/examples/atecc_csr.py
I'm now stuck with this bug again, even after reconnecting the board or rebooting the computer.

caternuson, 2021-01-16:
Can you post the stack trace you are getting. That example locks the chip, so may be running into something else.

fgallaire, 2021-01-16:
@caternuson
```
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    atecc = ATECC(i2c)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_atecc/adafruit_atecc.py", line 119, in __init__
    if (self.version() >> 8) not in (_ATECC_508_VER, _ATECC_608_VER):
  File "/usr/local/lib/python3.8/dist-packages/adafruit_atecc/adafruit_atecc.py", line 191, in version
    self.idle()
  File "/usr/local/lib/python3.8/dist-packages/adafruit_atecc/adafruit_atecc.py", line 145, in idle
    i2c.write(self._i2cbuf, end=1)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_bus_device/i2c_device.py", line 102, in write
    self.i2c.writeto(self.device_address, buf, start=start, end=end)
  File "/home/fgallaire/.local/lib/python3.8/site-packages/busio.py", line 115, in writeto
    return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
  File "/home/fgallaire/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/mcp2221/i2c.py", line 19, in writeto
    self._mcp2221.i2c_writeto(address, buffer, start=start, end=end)
  File "/home/fgallaire/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 315, in i2c_writeto
    self._i2c_write(0x90, address, buffer, start, end)
  File "/home/fgallaire/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 233, in _i2c_write
    raise RuntimeError("I2C slave address was NACK'd")
RuntimeError: I2C slave address was NACK'd

```

caternuson, 2021-01-16:
Thanks. Yah, it's failing right off again. Can you not even get what was working before to work again? Sounds like you saw some success initially.

fgallaire, 2021-01-16:
Yes, success running atecc_simpletest.py multiple times with one time the bug back but after reconnecting all was ok again.
Since running atecc_csr.py it's completely dead.

caternuson, 2021-01-16:
@brentru Any ideas? Sounds like maybe the board is "locked" now or something?

fgallaire, 2021-01-18:
https://learn.adafruit.com/adafruit-atecc608-breakout/python-circuitpython
"This will permanently lock your chip. You will only need to set this once."

@brentru what does mean "lock" exactly ?
I really would like to help to fix the MCP2221A/ATECC issue, but you the (only) one with the technical skills so I need your help tu understand

lindes, 2021-02-17:
I was hitting this same symptom, too -- not with an ATECC, but with an MCP2221A, trying to use it with any/all of the PMSA003I Air Quality Breakout (Adafruit PID 4632), and the 128x64 (938, QT edition) and 128x32 (4440) monochrome screens.  All 3 were having the same problem, whether I used any 1, or all 3 connected at once, or whatever.

Something [above](https://github.com/adafruit/Adafruit_Blinka/issues/262#issuecomment-760502545), though, made me find a way around the problem -- which I'm hoping will be helpful to others in ultimately tracking this down (or working around it, in the mean time): 

> The CircuitPython driver currently is using an I2C scan to wake it up. I _think_ that's a bit of hack, since the actual wake sequence is...strange...

Well, I was also doing a `scan`, just as something that I figured might be a good way to decide which chunks of code to run -- wanting to only attempt to talk to boards that were actually plugged in.  Here's a narrowed-down repro with just the PMSA:

```python
import board
import busio
import adafruit_pm25.i2c
i2c = busio.I2C(board.SCL, board.SDA)
i2c.scan() # <=--- culprit line!
partcounter = adafruit_pm25.i2c.PM25_I2C(i2c)
print(partcounter.read())
```

With the scan in there, I get the following:

```
Traceback (most recent call last):
  File "./repro", line 8, in <module>
    partcounter = adafruit_pm25.i2c.PM25_I2C(i2c)
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_pm25/i2c.py", line 53, in __init__
    self.i2c_device = I2CDevice(i2c_bus, address)
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 154, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/busio.py", line 116, in writeto
    return self._i2c.writeto(address, buffer, stop=stop)
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/mcp2221/i2c.py", line 19, in writeto
    self._mcp2221.i2c_writeto(address, buffer, start=start, end=end)
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 324, in i2c_writeto
    self._i2c_write(0x90, address, buffer, start, end)
  File "/home/lindes/src/test-local/env/lib/python3.7/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 241, in _i2c_write
    raise RuntimeError("I2C slave address was NACK'd")
RuntimeError: I2C slave address was NACK'd
```

(Note: my line numbers in mcp2221.py won't match yours -- I was adding debugging code, trying to figure out what was going on.)

If I comment out that `i2c.scan()` line, though, it works!  (Giving `{'pm10 standard': ` ... etc.)

So, the bug is probably in, or at least related to, the `scan()` code?  Perhaps something isn't properly being cleaned up?

A bit more info, in case it's useful...

In `_i2c_write`, just before the [`raise RuntimeError`](https://github.com/adafruit/Adafruit_Blinka/blob/540ca76e59dd2935909fe98e35585a909522166a/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L233) that's doing this, I added a call to `self._status_dump()`... and I noticed something interesting: this code path is being hit for every single ID in the `i2c_scan`'s [`range`](https://github.com/adafruit/Adafruit_Blinka/blob/540ca76e59dd2935909fe98e35585a909522166a/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L341) (0-121, decimal, inclusive)!  Even the id values corresponding to attached (and recognized by scan) devices.  They're just being ignored by the `try`/`except` code on [Line 345](https://github.com/adafruit/Adafruit_Blinka/blob/540ca76e59dd2935909fe98e35585a909522166a/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L345).  Whether that's where the problem is or not, I'm not sure, but of course in the failing version of my code above, that same error gets hit (and not caught, unless I catch it) when trying to create the `PM25_I2C` interface.

So, my final bit of information in this update is to say that when I have the code for screens in, too, and wrap that in try/except blocks that report the errors but continue past them, here's what I get from the last four `_status_dump` calls (the last one in the scan, and the 3 attempts to create interface objects) (plus some comments added by hand to show what's coming from what):

```
# last item in scan:
# starting with some debugging output I added at the top of _i2c_write:
DEBUG: _i2c_write: cmd=144 (0x90), addr=121, buff=b'\x00', start=0, end=1
# and then the _status_dump, right above the raise RuntimeError:
     0  1  2  3  4  5  6  7  8  9
0 : 10 00 00 00 00 00 00 00 25 01
1 : 00 00 00 00 75 0e f2 00 10 28
2 : 40 60 01 01 00 00 f1 7b f0 00
3 : 00 00 30 00 3b 00 11 2f 13 64
4 : 03 48 00 26 90 10 41 36 31 32
5 : 2a 01 e3 00 b1 00 81 02 69 01
6 : 00 00 00 00
# some additional output from my more complete test code (which I'll include
# below, just in case it's useful):
--- done with scan ---
i2c.scan results: [18, 60, 61] ['0x12', '0x3c', '0x3d']
[... 50 lines cut ...]
scan_results = i2c.scan()
print("--- done with scan ---")

print("i2c.scan results:", scan_results, list(map(hex, scan_results)))
for id in scan_results:
[... cut ...]
    big_oled = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c, addr=0x3d)
except RuntimeError as e:
    print("Big OLED failed! e={}".format(e))
```

So...... I hope that's helpful!  If nothing else, if others are facing this problem, and you're doing an `i2c.scan()`, then see if you can try it without that, and maybe things will work for you.  It definitely has me going further than I was!

(And indeed, I've now fairly successfully gotten all devices working.  I just can't scan to see which ones are there, without breaking things.)

CatzRunning, 2021-05-01:
[BUG] "RuntimeError: I2C slave address was NACK'd"
I found the bug in code

file : /home/pi/.local/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py

function : self.__probe_for_device()  

`self.i2c.writeto(self.device_address, b"")`
change to
`self.i2c.writeto(self.device_address, b'\x00')`

fgallaire, 2021-05-03:
@brentru @caternuson what about this fix ?

ladyada, 2021-05-03:
its not a good idea to write data to random devices, and the person wasnt talking about the ATECC608. you should probably assume the ATECC will never work with blinka/MCP2221 - they are just not compatible due to the way ATECC broke I2C standards!

fgallaire, 2021-05-03:
@ladyada but people are reporting the same bug from many other devices, so it seems to me that it looks like a general bug in the I2C code ?

mcharmon, 2021-05-04:
I am seeing a similar issue when using an stm32f405 (with upython) as a slave device and a linux pc with mcp2221 as the master.  It works for a while, but eventually will crash with a runtime error.  This is an issue because I am using multiple stm32 slaves and I cant have one NACK crashing my entire program.  Do you have any advice for helping this issue.  I think it may be related to this thread which is why Im posting here.  Thanks

ladyada, 2021-05-04:
if folks wanna try catz' fix and it works let us know - we dont have any other fix! note the mcp2221 i2c core is not very well documented so could be there's a bug but we don't know what it is :)

caternuson, 2021-05-10:
@makermelissa I think at this point it's unclear how universal of a fix this is. Waiting for feedback from people per @ladyada 's suggestion above.

brandrisken, 2022-09-27:
I run into raise RuntimeError("I2C slave address was NACK'd" when doing the brakout on https://learn.adafruit.com/16-channel-pwm-servo-driver/python-circuitpython having the PCA9685 on the I2C bus of a MCP2221A. The error message is very identical to those reported earlier. The crash happens when on the line
`pca = PCA9685(i2c_bus)`
and the output is

`
  File ~/anaconda3/lib/python3.9/site-packages/adafruit_pca9685.py:141 in __init__
    self.i2c_device = i2c_device.I2CDevice(i2c_bus, address)

  File ~/anaconda3/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py:63 in __init__
    self.__probe_for_device()

  File ~/anaconda3/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py:176 in __probe_for_device
    self.i2c.writeto(self.device_address, b"")

  File ~/anaconda3/lib/python3.9/site-packages/busio.py:175 in writeto
    return self._i2c.writeto(address, buffer, stop=stop)

  File ~/anaconda3/lib/python3.9/site-packages/adafruit_blinka/microcontroller/mcp2221/i2c.py:22 in writeto
    self._mcp2221.i2c_writeto(address, buffer, start=start, end=end)

  File ~/anaconda3/lib/python3.9/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py:333 in i2c_writeto
    self._i2c_write(0x90, address, buffer, start, end)

  File ~/anaconda3/lib/python3.9/site-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py:251 in _i2c_write
    raise RuntimeError("I2C slave address was NACK'd")

RuntimeError: I2C slave address was NACK'd`

If I try `i2c_bus.scan()` i get an empty list []. I work on linux platform. Does this indicate some wiring errors, or this a similar problem as reported earlier?

ladyada, 2022-09-27:
you have wiring errors - can post to the adafruit forum if using adafruit products

closed 2022-09-27 by ladyada

brandrisken, 2022-09-27:
Thank you for the very quick responce. I will thorougly ckeck the wiring,
although the things I can easily check looks OK. I am a beginner through.
If I don't solve it I will post i Adafruits, the boards are Adafruit
products.
Kind regards,
Anders