adafruit/Adafruit_Blinka #263: bibangio - No module named 'machine'

zacreid, 2020-03-07:
Hi,
when I try and run the following code
`import bitbangio
import board

i2c = bitbangio.I2C(3,2)
`
I get an error when bitbangio.py trys and imports machine

Is there anyway to install this module or is there something wrong with my install? I've also tried force reinstalling adafruit-blinka

ladyada, 2020-03-07:
bitbangio is not supported on raspberry pi or other linux computers at this time

yeyeto2788, 2020-04-06:
Hey @ladyada 

I have borrowed some code from https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/SPI.py

Is it possible to do something like the code below? I know the class methods do not correspond to the ones on the https://github.com/adafruit/Adafruit_Blinka/blob/2bb7cbf33c9970aa242b6446f24c49713bd2c113/src/bitbangio.py#L61

I have change all calls to `Adafruit_GPIO` to `busio.DigitalInOut` and set the values corresponding to the ones on the original code. I also notice that we should be inheriting from the `Lockable` class but before doing more things I just wanted to confirm if something like this will work.

```python
from .digitalio import DigitalInOut

MSBFIRST = 0
LSBFIRST = 1

class BitBang(object):
    """Software-based implementation of the SPI protocol over GPIO pins."""
    
    
    def init(self, baudrate, polarity, phase, bits, firstbit=MSBFIRST, sck, mosi, miso, ce):
        """Initialize bit bang (or software) based SPI.  Must provide a BaseGPIO
        class, the SPI clock, and optionally MOSI, MISO, and SS (slave select)
        pin numbers. If MOSI is set to None then writes will be disabled and fail
        with an error, likewise for MISO reads will be disabled.  If SS is set to
        None then SS will not be asserted high/low by the library when
        transfering data.
        """
        self._sclk = sck
        self._mosi = mosi
        self._miso = miso
        self._ss = ce
        # Set pins as outputs/inputs.
        self._sclk.switch_to_output()
        
        if self._mosi is not None:
            self._mosi.switch_to_output()
        if self._miso is not None:
            self._miso.switch_to_output()
        if self._ss is not None:
            self._ss.switch_to_output()
            # Assert SS high to start with device communication off.
            self._ss.value(True)
        # Assume mode 0.
        self.set_mode(0)
        # Assume most significant bit first order.
        self.set_bit_order(MSBFIRST)

    def set_clock_hz(self, hz):
        """Set the speed of the SPI clock.  This is unsupported with the bit
        bang SPI class and will be ignored.
        """
        pass

    def set_mode(self, mode):
        """Set SPI mode which controls clock polarity and phase.  Should be a
        numeric value 0, 1, 2, or 3.  See wikipedia page for details on meaning:
        http://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus
        """
        if mode < 0 or mode > 3:
            raise ValueError('Mode must be a value 0, 1, 2, or 3.')
        if mode & 0x02:
            # Clock is normally high in mode 2 and 3.
            self._clock_base = True
        else:
            # Clock is normally low in mode 0 and 1.
            self._clock_base = False
        if mode & 0x01:
            # Read on trailing edge in mode 1 and 3.
            self._read_leading = False
        else:
            # Read on leading edge in mode 0 and 2.
[... 110 lines cut ...]
                self._sclk.value(not self._clock_base)
                # Handle read on leading edge of clock.
                if self._read_leading:
                    if self._miso.value():
                        # Set bit to 1 at appropriate location.
                        result[i] |= self._read_shift(self._mask, j)
                    else:
                        # Set bit to 0 at appropriate location.
                        result[i] &= ~self._read_shift(self._mask, j)
                # Return clock to base.
                self._sclk.value(self._clock_base)
                # Handle read on trailing edge of clock.
                if not self._read_leading:
                    if self._miso.value():
                        # Set bit to 1 at appropriate location.
                        result[i] |= self._read_shift(self._mask, j)
                    else:
                        # Set bit to 0 at appropriate location.
                        result[i] &= ~self._read_shift(self._mask, j)
        if deassert_ss and self._ss is not None:
            self._ss.value(True)
        return result
```

Regards.

ladyada, 2020-04-13:
yah @makermelissa will make a software bitbang library https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO

makermelissa, 2020-04-13:
Thanks @yeyeto2788. I'll probably be using this code as a starting point.

yeyeto2788, 2020-04-14:
Great, I started doing some refactor on the `bitbangio` within blinka on https://github.com/yeyeto2788/Adafruit_Blinka/tree/bitbangio_spi/src/bitbangio which can help you better than this code. If you need any help do not hesitate on contacting me.

Regards.

makermelissa, 2020-04-14:
Excellent. I wanted to use the same bitbangio interface available to circuitpython so that users could easily use the Adafruit_CircuitPython_BitbangIO library with minimal code changes and this looks great.

makermelissa, 2020-04-14:
Ok, I'm very close to getting bitbang spi finished. write and readinto are both working and I just need to add a few more checks to write_readinto and pylint.
[image]
[image]
Sending out 0xD0 and expecting a response of 0x60 (96).

yeyeto2788, 2020-04-15:
Wujuuuuu! That's great news. I can confirm you have done an amazing job on this.

I tried it with the [PCD8455](https://github.com/adafruit/Adafruit_CircuitPython_PCD8544) and guess what..........

[image]

IT WORKS!!!

ladyada, 2020-04-15:
thats awesome. thank you for testing! :)

makermelissa, 2020-04-15:
Thanks for testing @yeyeto2788. I still have a little more testing do do myself.

ladyada, 2020-04-21:
done here https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO

closed 2020-04-21 by ladyada