adafruit/Adafruit_Blinka #264: Adafruit library download

jayjabour, 2020-03-19:
I am trying to install Adafruit motorkit library for their Raspberry Pi Motor Controller hat.  I have Python 3.7.0 installed, but the download process will not install with the reason given 'Python 3.5 or later is required'.  Python 3.5 is on the Pi, but I am using new version 3.7.  Any help appreciated.

makermelissa, 2020-03-19:
Try explicitly typing python3 or pip3. It's possible that you also have python 2 (both can exist simultaneously) and that's what it is running with. What is the command you're typing to run it?

jayjabour, 2020-03-19:
> <reply@reply.github.com>
> <author@noreply.github.com>

Melissa

Thanks for the reply.  The command I used was;

        sudo pip3 install adafruit-circuitpython-motorkit

I am a Linux novice, a lot of experience with C++ and Java on PC and Mac,
but I have avoided Linux.  Thanks for any help you can give.

Jay Jabour

<https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6013772
JXPJANCNFSM4LPSNWFA> .
"https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6013772
21", "url": 
"https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6013772
21", "name": "View Issue" }, "description": "View this Issue on GitHub",
"publisher": { "@type": "Organization", "name": "GitHub", "url":
"https://github.com" } } ]

makermelissa, 2020-04-06:
Could you explain more what you meant by:
> Python 3.5 is on the Pi, but I am using new version 3.7.

Meaning you are able to run either?
Also, are you running Raspbian or a different OS? We officially support Raspbian, but I've heard people have been able to get it working on Ubuntu and Diet Pi.

jayjabour, 2020-04-07:
Melissa

The Raspberry Pi came with Python 3.5.  When I tried to install the Motor
driver support it gave me a message that the library does not work with
Python version 3.5.  I then updated the Python version to 3.7.  Again, tried
to install the Motor driver library and I still got the message that the
library does not support versions before 3.5.  I can now run either 3.5 and
3.7.

I am using Raspberry Pi Raspbian version of Linux.

William Jabour

<reply@reply.github.com>
Date:  Monday, April 6, 2020 at 11:47 AM

<https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6098758
H2SXANCNFSM4LPSNWFA> .
"https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6098758
76", "url": 
"https://github.com/adafruit/Adafruit_Blinka/issues/264#issuecomment-6098758
76", "name": "View Issue" }, "description": "View this Issue on GitHub",
"https://github.com" } } ]

makermelissa, 2020-04-14:
Hi, I saw another user with a similar problem. Please try this:
`pip3 install adafruit-circuitpython-motorkit`
(without the sudo) and see if that works.

makermelissa, 2020-04-24:
I'm pretty sure the above was the correct answer as it fixed it for the user with a similar problem. Closing due to lack of response.

closed 2020-04-24 by makermelissa