adafruit/Adafruit_Blinka #269: RuntimeError: ws2811_init failed with code -2 (Out of memory)

dustyyeti, 2020-04-09:
I'm driving a Neopixel strip successfully with a Raspi 1B (512MB RAM) Raspbian Stretch without GUI using the python control scripts. 

Things work fine but long uptime the Pi (between a few hours and a few days) it fails
In the log files I find this:
`RuntimeError: ws2811_init failed with code -2 (Out of memory)`

I'm using a 5 minute cycle crontab script job to turn the lights on and off, so over the course of a day the strip is directly manipulated 400+ times.

On another board someone asked if I am calling `ws2811_fini` at then end of the script, which look like a thing to do, but there's no call to initialize it, or any reference in the libraries to it at all.

Can anyone assiste?

ladyada, 2020-04-12:
the underlying ws28xx may have a memory leak...you should try to avoid creating new neopixel objects or running the script repeatedly. e.g. try running the script once and sending it signals based on behavior you want!

makermelissa, 2020-04-29:
Looks like you found the underlying issue at https://github.com/jgarff/rpi_ws281x/issues/390 as I see you posted there. Closing here.

closed 2020-04-29 by makermelissa