adafruit/Adafruit_Blinka #271: MCP2221 import board error in macos

GoJian, 2020-04-14:
Hi -

I seek help in resolving this problem. I want to develop sensors using the adafruit MCP2221 board. My platform is a MacBook pro running latest Catalina (10.15.4) and Anaconda Python 3.7.7. Everything else is up to date.

I followed the guide on the [adafruit MCP2221 website](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/mac-osx) to the word. I am sure my hidapi and adafruit-blinka packages are up to date (tried this many times over the past few weeks).

Problem is that I cannot pass the "import board" command. I am getting an error related to the hid library. It is throwing out an error about unknown symbols: Symbol not found: ___CFConstantStringClassReference. I also tried the system python3 and it is the same response. I am really lost. Probably I missed something rather simple. The board is working on my Windows and Linux machines.

More info regarding the libraries on my system:

Python 3.7.7 (default, Mar 26 2020, 10:32:53)
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin

hidapi                    0.9.0.post2              pypi_0    pypi
adafruit-blinka           4.3.0                    pypi_0    pypi
adafruit-platformdetect   2.5.0                    pypi_0    pypi
adafruit-pureio           1.1.0                    pypi_0    pypi

Here is the error message.

`
% export BLINKA_MCP2221="1"
% python
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/user/Programs/Anaconda/lib/python3.7/site-packages/board.py", line 32, in <module>
    from adafruit_blinka.agnostic import board_id, detector
  File "/Users/user/Programs/Anaconda/lib/python3.7/site-packages/adafruit_blinka/agnostic/__init__.py", line 18, in <module>
    chip_id = detector.chip.id
  File "/Users/user/Programs/Anaconda/lib/python3.7/site-packages/adafruit_platformdetect/chip.py", line 34, in id
    import hid  # pylint: disable=import-error
ImportError: dlopen(/Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so, 2): Symbol not found: ___CFConstantStringClassReference
  Referenced from: /Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so
  Expected in: flat namespace
 in /Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so
>>>
`

yeyeto2788, 2020-04-14:
Can you try on a python REPL the following code:

```python
import hid
for dev in hid.enumerate()
  print(dev['vendor_id'], dev['product_id'])
```

You should get this on the loop:
`dev['vendor_id'] == 0x04D8 and dev['product_id'] == 0x00DD`

If you're not able to import the `hid` module that might be error being raised. All this is coming from this conditional branch https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/34697fef8b2cce5652b354797d856cab106ca413/adafruit_platformdetect/chip.py#L33

If you still get the error try reading [this](https://github.com/trezor/cython-hidapi/issues/65) issue which seems to have a possible fix for the installation.

GoJian, 2020-04-14:
Thanks. I run the "import hid" command and indeed it failed with the same error. This should be a problem with the hidapi library on my setup.

```
% python
Python 3.7.7 (default, Mar 26 2020, 10:32:53)
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so, 2): Symbol not found: ___CFConstantStringClassReference
  Referenced from: /Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so
  Expected in: flat namespace
 in /Users/user/Programs/Anaconda/lib/python3.7/site-packages/hid.cpython-37m-darwin.so
>>>
```

However, going into [the issue](https://github.com/trezor/cython-hidapi/issues/65), it appears that by the version 0.9.0.post2 (the latest version I have), this issue should have been patched already?

hidapi 0.9.0.post2 pypi_0 pypi

Should I recompile the hidapi library? I am a little lost here. I am also surprised that I am the only one with this issue...

yeyeto2788, 2020-04-14:
You may want to try the [build from source](https://github.com/trezor/cython-hidapi#build-from-source) part and see if that solve your issue.

GoJian, 2020-04-15:
Thanks for your help and pointers. Indeed I tried to build hidapi on my system, which produced the exact error. By now I think the issue is with the hidapi library, so I went ahead and opened another issue [there](https://github.com/trezor/cython-hidapi/issues/82). I will close this issue here.

closed 2020-04-15 by GoJian