adafruit/Adafruit_Blinka #272: Chip specifications and board settings are different

hhk7734, 2020-04-17:
ODROID-C4 will be released on April 23.

Each I2C, SPI, and UART bus number will be set identically according to the pin location., to provide the same usage and code, no matter what the board user purchases(ODROID-C2, C4, N2, and XU4).

* I2C-0 : Pin 3(SDA), 5(SCL)
* I2C-1 : Pin 27(SDA), 28(SCL)
* SPI-0 : Pin 19(MOSI), 21(MISO), 23(SCLK), 24(CS0)
* UART-0 : UART(debug port)
* UART-1 : Pin 8(TX1), 10(RX1)

For example, s905x3 I2C-2 is marked I2C-0 at C4 and s905x3 I2C-3 is marked I2C-1.

Received the wrong information, I had to write I2C-0,1, but I worked with I2C-1,2. :(
So I'm going to change 
https://github.com/adafruit/Adafruit_Blinka/blob/c7e3b1b74da78bd07997e104727fad1ee204ebf0/src/adafruit_blinka/board/hardkernel/odroidc4.py#L5-L11
https://github.com/adafruit/Adafruit_Blinka/blob/c7e3b1b74da78bd07997e104727fad1ee204ebf0/src/adafruit_blinka/microcontroller/amlogic/s905x3/pin.py#L73-L76
https://github.com/adafruit/Adafruit_Blinka/blob/c7e3b1b74da78bd07997e104727fad1ee204ebf0/src/adafruit_blinka/microcontroller/amlogic/s905x3/pin.py#L86

In this case, how would it be best to fix it?

And these changes will be reflected in the OS deployed on Hardkernel and other OS based on the kernel of Hardkernel.

What to do with N2 and C2?

PS. I am sorry that the correction was repeated after PR was reflected.

makermelissa, 2020-04-17:
Hi, let me see if I understand correctly. You need to make a change on an unreleased chip and that's fine. You also want to change N2 and C2 boards to use I2C-1 and I2C-2 instead of I2C-0 and I2C-1? What will those change in the OS as well?

Thanks,
Melissa

hhk7734, 2020-04-17:
First, C4 problem.

If using mainline kernel or amlogic BSP without customization, s905x3 chip specifications are:
```python
 I2C2_SDA = GPIOX_17 
 I2C2_SCL = GPIOX_18 
 I2C3_SDA = GPIOA_14 
 I2C3_SCL = GPIOA_15 
```

But, if adding Hardkernel board.dts like below
https://github.com/hardkernel/linux/blob/56cc8c912485366a7d4f8b58309b2a3822ae27a2/arch/arm64/boot/dts/amlogic/meson-g12b-odroid-n2.dts#L19-L26

the s905x3 specification will be shown below
```python
 I2C1_SDA = GPIOX_17 
 I2C1_SCL = GPIOX_18 
 I2C2_SDA = GPIOA_14 
 I2C2_SCL = GPIOA_15 
```

If I set the s905x3 to match the hardkernel way, I was worried that if other companies release another board using the same chip, they may have trouble adding the board to Blinka.

I hope that various boards will be added to Blinka for users.

Before next Thursday, below commit will be merged into N2 and C4 hardkernel's Kernel.
```
aliases {
		serial0 = &uart_AO;
		serial1 = &uart_A;
		i2c0 = &i2c0;
		i2c1 = &i2c2;
		i2c2 = &i2c3;
		spi0 = &spicc0;
	};
```

If it is okay to modify it, I will modify it to `i2cPorts = ((0, I2C0_SCL, I2C0_SDA), (1, I2C1_SCL, I2C1_SDA), ) `.

makermelissa, 2020-04-17:
Oh, I see. Well if another board is added that uses the I2C ports in another way, then a separate chip file could always be added. :)

hhk7734, 2020-04-17:
:+1:  :), If so, I'm going to submit the s905x3 as amended.

Second, on N2, giving up backward compatibility, the bus number will be changed.

Any suggestions on what to do for backward compatibility in this case?

hhk7734, 2020-04-17:
First, we will find a way to recognize before and after changes.

hhk7734, 2020-04-17:
If there are no problems, close this issue. Thank you. :)

makermelissa, 2020-04-17:
We may need to add something to PlatformDetect to check whether it has the hardkernel dts or not and count them as separate revisions like the Raspberry Pi board revisions. I think then we check against the revision and use the appropriate set of i2c.

makermelissa, 2020-04-17:
Let's leave this open until we have something working.

hhk7734, 2020-04-20:
The I2C bus number will be detected on import. - #274 
S922X(N2) is similar to S905X3(C4). So it seems to be able to update while maintaining backward compatibility.

hhk7734, 2020-04-22:
Ref: #275

closed 2020-04-22 by hhk7734