adafruit/Adafruit_Blinka #278: Board not supported exception

bawilless, 2020-04-28:
I'm using an Avnet IIoT Gateway based on the Pi and I see this exception:

Traceback (most recent call last):
  File "pumpjack.py", line 17, in <module>
    import board
  File "/usr/local/lib/python3.6/site-packages/board.py", line 155, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None

If I dump my cpuinfo I get the following data:

avnet@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 60a220b0
Serial		: 0000000025fe521e

Can you help?

Thanks

makermelissa, 2020-04-28:
This will most likely need to be added into Adafruit_Python_PlatformDetect and I'd imagine it would be fairly easy to get working. What I'd suggest doing is getting it detected as a new board along with the BCM2835 proc, which is what the Pis have, and then when the new board is detected, just load the board definition for the Pi and test to make sure it works for you.

makermelissa, 2020-04-28:
Also, if you're not sure how to do that, check out the guides at:
https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka
and
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

bawilless, 2020-05-05:
Hi Melissa,

Thanks for the help!  I have my board being detected now . . .   The Avnet board is based off the Pi 3B+.

Chip id:  BCM2XXX
Board id:  RASPBERRY_PI_3B_PLUS
Is this a DragonBoard 410c? False
Is this a Pi 3B+? True
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? True
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this an OS environment variable special case? False
Raspberry Pi detected.

However, something is still not correct.  My application is not finding the attributes I need.

If I dump the board attributes I get an empty list.  Any ideas?

$ python3
Python 3.6.3 (default, Apr 27 2020, 18:52:20) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> dir(board)
['Board', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__repo__', '__spec__', '__version__', 'boards', 'chips', 'os', 're']

makermelissa, 2020-05-06:
Hi, I understand that it is a derivative of the Raspberry Pi, but it should be detected as its own board since there may be differences and why it isn't working. Then in Blinka, it can just load the raspberry pi board and microcontroller files for the gateway. It's a little bit of a long way, but avoids confusion. If you can get it detected as its own board, I can help you with making the changes in Blinka.

makermelissa, 2020-05-08:
Since you have this being detected as a 40-pin Pi, there may not be any changed required for Blinka. Go ahead and test it.

bawilless, 2020-05-08:
That's what I thought too!  I tried it before committing my changes, it still does not work.  See output below.  

Traceback (most recent call last):
  File "i2c_issue.py", line 31, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
  File "/usr/local/lib/python3.6/site-packages/busio.py", line 20, in __init__
    self.init(scl, sda, frequency)
  File "/usr/local/lib/python3.6/site-packages/busio.py", line 55, in init
    (scl, sda), i2cPorts
ValueError: No Hardware I2C on (scl,sda)=(3, 2)
Valid I2C ports: ((3, 3, 2), (1, 3, 2), (0, 1, 0))

makermelissa, 2020-05-08:
Make sure you have i2c enabled. You should see some stuff if you type `ls /dev/i2c*`.

bawilless, 2020-05-08:
Thanks Melissa,

Nothing there, I've verified the /boot/config.txt file has i2c enabled.  Thanks for the pointer, I'll get this working next.

Brian

makermelissa, 2020-05-15:
@bawilless, were you able to get this working?

bawilless, 2020-05-15:
@makermelissa, yes I'm up and running.  Thank yo for all the support!

closed 2020-05-15 by bawilless

makermelissa, 2020-05-15:
Thank you. I can add it to circuitpython.org now.