adafruit/Adafruit_Blinka #283: Beaglebone Black not recognized

TASSDevon, 2020-05-03:
When not running as root I get:

`Traceback (most recent call last):
  File "ams_bbb.py", line 5, in <module>
    import busio
  File "/home/mqtt/.local/lib/python3.5/site-packages/busio.py", line 12, in <module>
    from adafruit_blinka import Enum, Lockable, agnostic
  File "/home/mqtt/.local/lib/python3.5/site-packages/adafruit_blinka/agnostic/__init__.py", line 19, in <module>
    board_id = detector.board.id
  File "/home/mqtt/.local/lib/python3.5/site-packages/adafruit_platformdetect/board.py", line 71, in id
    board_id = self._beaglebone_id()
  File "/home/mqtt/.local/lib/python3.5/site-packages/adafruit_platformdetect/board.py", line 182, in _beaglebone_id
    with open("/sys/bus/nvmem/devices/0-00500/nvmem", "rb") as eeprom:
PermissionError: [Errno 13] Permission denied: '/sys/bus/nvmem/devices/0-00500/nvmem'`

When I run as root to bypass the permission error I get:

`Traceback (most recent call last):
  File "ams_bbb.py", line 3, in <module>
    import board
  File "/usr/local/lib/python3.5/dist-packages/board.py", line 156, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None`

ladyada, 2020-05-03:
@pdp7 please look

pdp7, 2020-05-03:
@TASSDevon please run this script and paste the output in a reply:
```
sudo /opt/scripts/tools/version.sh
```
This will allow me to see more about the BeagleBone that you have and the software installed on it.

Also, please let me know which release of Adafruit_Blinka you running.

TASSDevon, 2020-05-03:
Output of python script is:

```
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT000B0514BBBK3455]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v10.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.107102] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.118967] gpio-of-helper ocp:cape-universal: ready
END
```

Versions of modules:

```
Adafruit-BBIO==1.0.10
Adafruit-Blinka==4.7.0
Adafruit-GPIO==1.0.3
Adafruit-PlatformDetect==2.8.0
Adafruit-PureIO==1.1.5
```

pdp7, 2020-05-05:
@TASSDevon It appears that the board ID in the eeprom does not match the list that is in Adafruit_Python_PlatformDetect.

Your eeprom:
```
eeprom:[A335BNLT000B0514BBBK3455]
```
I think this indicates a BeagleBone Black rev. B.

The existing list of IDs:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/constants/boards.py#L187

I think the fix would be to add that ID to the list.

However, I am going to first check with @RobertCNelson to see if `A335BNLT000B0514BBBK3455` is a valid ID.  The other possibility is that the eeprom value got corrupted.

RobertCNelson, 2020-05-05:
@pdp7 it's valid:

https://elinux.org/Beagleboard:BeagleBoneBlack#Revision_B

Just very Rare.. I've never seen one before..

@TASSDevon , do you have a story on how you got it? Ebay/etc or just bought it normally..

Regards,

closed 2020-05-05 by ladyada

pdp7, 2020-05-05:
I've created a pull request which should fix this issue:
https://github.com/pdp7/Adafruit_Python_PlatformDetect/commit/d5f3b987a4dceeb4f176e03a733c9a4d56d1a5f5

Note: it does need to run as root to read the eeprom contents:
`PermissionError: [Errno 13] Permission denied: '/sys/bus/nvmem/devices/0-00500/nvmem'`

pdp7, 2020-05-05:
@TASSDevon if you are running as the 'debian' user on a recent BeagleBoard.org Debian image (http://beagleboard.org/latest-images), then you do not need to be root.
```
debian@beaglebone:~$ ls -ltar /sys/bus/nvmem/devices/0-00500/nvmem 
-rw-rw---- 1 root gpio 0 Jan  1 00:13 /sys/bus/nvmem/devices/0-00500/nvmem
debian@beaglebone:~$ id debian
uid=1000(debian) gid=1000(debian) groups=1000(debian),4(adm),15(kmem),20(dialout),24(cdrom),25(floppy),29(audio),30(dip),44(video),46(plugdev),100(users),101(systemd-journal),108(i2c),111(bluetooth),112(netdev),999(gpio),998(pwm),997(eqep),996(remoteproc),994(admin),993(spi),992(tisdk),991(weston-launch),990(xenomai),989(cloud9ide)
debian@beaglebone:~$ hexdump -C /sys/bus/nvmem/devices/0-00500/nvmem |head
00000000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 30 43  |.U3.A335BNLT000C|
00000010  33 30 31 34 42 42 42 4b  31 33 31 36 ff ff ff ff  |3014BBBK1316....|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00001000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 30 43  |.U3.A335BNLT000C|
00001010  33 30 31 34 42 42 42 4b  31 33 31 36 ff ff ff ff  |3014BBBK1316....|
00001020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00002000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 30 43  |.U3.A335BNLT000C|
00002010  33 30 31 34 42 42 42 4b  31 33 31 36 ff ff ff ff  |3014BBBK1316....|

debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2019-12-01

```

makermelissa, 2020-05-05:
I just pushed a release of Platform Detect with the update in it.

TASSDevon, 2020-05-06:
@RobertCNelson Back in 2014 I got 2 Beaglebone Blacks, one as a present starting at my employer (no clue where they bought them) and one during a Linux Belgium Embedded SW course (also no idea where they got them). It's funny that exactly the BBB I've been using is so rare that it hadn't even been considered in the ID list :-)

I gave one away to a friend so not sure if that one also was a rev B, and one has been used as a zigbee to mqtt bridge for a long time.

@pdp7 That being said, I have no interest in updating the Debian image as I don't want to break things, it's running fine as it is (if it works, don't touch it ;-) )

I've updated the PlatformDetect module:
```
mqtt@beaglebone:~$ sudo pip3 install --upgrade Adafruit-PlatformDetect
[sudo] password for mqtt:
Collecting Adafruit-PlatformDetect
  Downloading https://files.pythonhosted.org/packages/7f/f0/866f38d3aa4eb763cdacf13aa0ea05d0eafe593f47819a582d95d2400dff/Adafruit-PlatformDetect-2.9.1.tar.gz
Building wheels for collected packages: Adafruit-PlatformDetect
  Running setup.py bdist_wheel for Adafruit-PlatformDetect ... done
  Stored in directory: /root/.cache/pip/wheels/b6/67/b1/5a381d9d3c4fc6726d09cae971b95cf02db8223c993987a467
Successfully built Adafruit-PlatformDetect
Installing collected packages: Adafruit-PlatformDetect
  Found existing installation: Adafruit-PlatformDetect 2.8.0
    Uninstalling Adafruit-PlatformDetect-2.8.0:
      Successfully uninstalled Adafruit-PlatformDetect-2.8.0
Successfully installed Adafruit-PlatformDetect-2.9.1
```

And it looks like it's at least doing something now :-)

```
mqtt@beaglebone:~/skyhome/python_scripts$ sudo python3 ams_bbb.py
hello blinka!
I2C devices found:  ['0x5a']
```

If you're feeling generous, could you assist in what it is I need to do to be able to fetch data from the AMS iAQ Core C sensor? Datasheet here: https://datasheet.octopart.com/IAQ-CORE-C-ams-datasheet-44382459.pdf

DBell33, 2022-03-21:
Hello, I have no idea if this thread is still monitored but I'm having a similar issue. 

My printout from the version.sh script is as follows:

 sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT000C1713BBBG0936]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot.dts]
UBOOT: Loaded Overlay:[AM335X-20-00A0]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[univ-bbb-EVA-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.24.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-wl18xx-firmware]:[1.20211222.2-0~buster+20211222]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai                           cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   53.899413] remoteproc remoteproc0: wkup_m3 is available
[   53.941140] remoteproc remoteproc0: powering up wkup_m3
[   53.941172] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   53.945465] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   56.720931] remoteproc remoteproc1: 4a334000.pru is available
[   56.732648] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   56.720931] remoteproc remoteproc1: 4a334000.pru is available
[   56.721090] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   56.732648] remoteproc remoteproc2: 4a338000.pru is available
[   56.732850] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.878725] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.886195] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

Would anyone be able to tell me how I can add this serial number to get the board module working?

makermelissa, 2022-03-21:
@DBell33 please create a new issue.