adafruit/Adafruit_Blinka #288: OneWire Support?

phorensic, 2020-05-08:
Will Blinka ever support OneWire like the main fork of Circuit Python? I have a DS18B20 temp sensor and many hours have gone into trying to get it to work. I'm down to writing lots of code (bitbang?) from scratch at this point. It would be nice if busio or bitbingio modules had the OneWire code in them.

ladyada, 2020-05-08:
not likely soon - there's kernel support for ds18b's for linux - we recommend that! are you using like an ft232h?

phorensic, 2020-05-08:
Yes, FT232H.

ladyada, 2020-05-08:
unlikely it will ever work - without a native 1wire peripheral you will be bit banging at 1khz, which is incredibly slow... you could try this UART-1Wire convsion method and library, then you're just using pyserial which works with the FT232H when its in uart mode
https://github.com/mcsakoff/pydigitemp/tree/master/digitemp

phorensic, 2020-05-08:
Well, I also have an I2C temp/hum sensor (Si7021) and wanted to control relays on the GPIO ports at the same time. A saltwater reef aquarium controller all in. So dropping down into UART mode was not an option. I had luck finally getting I2C to work with GPIO at the same time, if the DS18B20 worked I would have my waterproof tank temperature sensor and everything needed! Now I might be looking at a non-OneWire waterproof temp sensor or a Raspberry Pi. I was trying to avoid the Raspberry Pi as my computer with 16 USB ports is sitting one foot away from the aquarium!