adafruit/Adafruit_Blinka #292: Bitbangio: ModuleNotFoundError: No module named 'machine'

tinue, 2020-05-21:
I am trying to use BitBang on a Raspberry Pi 4. The module "bitbangio" references a module "machine" that I can't find. Is this an external dependency that needs to be installed first?

I see that there is a "# TODO untested" in the code, so maybe this does not work at the moment.

makermelissa, 2020-05-21:
Hi, please try using https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO. I'm going to leave this request open as a reminder to myself to update the message to hint to people to use that library.

tinue, 2020-05-21:
Hello, thank you for the quick reply! Initial tests were partially successful, I am now getting "RuntimeError: First call try_lock()", but I'll experiment some more. I'm not expecting an answer here; This is another library after all ;-)

PR #293 (2020-05-21): Display message to linux user to use new Bitbang Library for now
Fixes #292.
closed 2020-05-21 by makermelissa