adafruit/Adafruit_Blinka #295: mcp2221: GP2 always reads low when configured as input

bablokb, 2020-05-22:
I have pullups connected to GP2 and GP3 and two buttons pulling them to GND. When I run this code

    #!/usr/bin/python3

    import board
    import digitalio
    import time

    btn2 = digitalio.DigitalInOut(board.G2)
    btn2.direction = digitalio.Direction.INPUT

    btn3 = digitalio.DigitalInOut(board.G3)
    btn3.direction = digitalio.Direction.INPUT

    while True:
      if not btn2.value:
        print("button 2")
        time.sleep(0.200)

       if not btn3.value:
         print("button 3")
         time.sleep(0.200)

it turns out that GP3 works as expected, while reading GP2 always returns low. This is no issue with the board-hardware, since I can reproduce this also with a bare mcp2221. It seems that GP2 is either pulled to low (in contradiction to what the datasheet says), or there is an error in  the Blinka code. I did a quick check but did not notice anything, but that is not surprising since I'm not familiar with the code.

bablokb, 2020-05-22:
Reading your code I actually found a hidden setting:

    export BLINKA_MCP2221_HID_DELAY=0.01

solves the problem.

I leave this issue open as a reminder that you might want to update the user's guide.

caternuson, 2020-05-22:
Interesting. That parameter was added in the process of working this issue:
https://forums.adafruit.com/viewtopic.php?f=60&t=160414
which was related to initialization. And it turned out to be hopefully unneeded, which is why we haven't as yet documented it. There is an FAQ for the other parameter here:
<https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/faq-troubleshooting#faq-3055702>

Can you provide more details on your setup. What OS? And to clarify the behavior, when you ran your example above, it spammed prints of `button 2`?

bablokb, 2020-05-22:
I'm using Linux (OpenSuse Leap 15.1). And yes, the program just printed "button 2" over and over again. I actually noticed it when I tested my own pcb (https://github.com/bablokb/pcb-mcp2221) which has buttons attached to GP2 and GP3. First, I thought it was an error of my pcb, but it turns out I have the same problem with the adafruit-pcb and with a breadboard-only setup of the MCP2221.

If I find some time I will test this on Raspbian. This is also Linux, but maybe just enough slower to not cause any problems.

Maybe it is not wise to set a global hid-delay, because it delays all hid-calls?! Maybe only the configuration calls are critical. Interesting also that the same code GP3 does not cause problems.

I spent quite some time analyzing your code and I think it is correct.

bablokb, 2020-05-25:
Ok, with some more testing I have to revise my theory that BLINKA_MCP2221_HID_DELAY fixes the issue. And I now also think there is an error in your code.

Some more insights: if I define GP3 after GP2, then GP2-behavior is corrupted. If I define GP2 after GP3, then GP3 behavior is corrupted. So configuring the second GP overwrites something for the first GP.

Looking at the code, there is a method `gp_set_mode()` which uses the `x61`-command to read current settings from SRAM for all GPs before updating the settings in SRAM with the `x60`-command for one GP. This command alters the direction (bit 3 of the GPx settings bytes).

And then there is a method `gpio_set_direction()` which uses the `x50`-command to alter the direction. So there are two commands that alter the direction, but it is unclear from the datasheet if they alter the same registers within the chip. 

The definition of both buttons will fire a `x61`,  `x60` and `x50` command. And if `x50` does not update the registers read by `x61`, then the second `x60`-command will undo the changes made by the first `x50`-command.

caternuson, 2020-05-25:
Thanks for the additional information. Your comments above seem more likely. I have recreated the issue from your first post. Also, by simply reversing the order the pins are set up, it starts spamming the prints for the other button.

caternuson, 2020-05-27:
Just dumping some notes. Reference MCP2221 Datasheet.

Pin **direction** can be checked in two locations.
  * `0x51 GET GPIO VALUES`, bytes 3, 5, 7, 9
  * `0x61 GET SRAM SETTINGS`, bit 3 of bytes 22, 23, 24, 25

Using `0x61 GET SRAM SETTINGS` doesn't seem to return pin direction correctly. This example sets GP2 to INPUT and then checks that both ways:
```python
>>> mcp.gpio_set_direction(2, 1)
>>> mcp._hid_xfer(b"\x51")[7]
1
>>> mcp._hid_xfer(b"\x61")[24]
0
>>>
```
Pin **value** can also be checked in two locations.
  * `0x51 GET GPIO VALUES`, bytes 2, 4, 6, 8
  * `0x61 GET SRAM SETTINGS`, bit 4 of bytes 22, 23, 24, 25
Using `0x61 GET SRAM SETTINGS` doesn't seem to return pin value correctly either. This example sets GP1 to OUTPUT and HIGH then checks the value in both locations:
```python
>>> mcp.gpio_set_direction(1, 0)
>>> mcp.gpio_set_pin(1, 1)
>>> mcp._hid_xfer(b"\x51")[4]
1
>>> mcp._hid_xfer(b"\x61")[23]
0
```
It at least seems to work for pin **mode**
  * 0x61 GET SRAM SETTINGS, bits 2-0 of bytes 22, 23, 24, 25

This example sets all pins to GPIO and then checks. Then sets each to different alt function and checks again.
```python
>>> mcp.gp_set_mode(0, 0)
>>> mcp.gp_set_mode(1, 0)
>>> mcp.gp_set_mode(2, 0)
>>> mcp.gp_set_mode(3, 0)
>>> mcp._hid_xfer(b"\x61")[22:26]
[0, 0, 0, 0]
>>> mcp.gp_set_mode(0, 2)
>>> mcp.gp_set_mode(1, 3)
>>> mcp.gp_set_mode(2, 1)
>>> mcp._hid_xfer(b"\x61")[22:26]
[2, 3, 1, 0]
>>>
```

**TLDR** - the transpose approach being used here is not working as expected:
https://github.com/adafruit/Adafruit_Blinka/blob/5346e2eeeeb549add7b051f9d3dd8b11dfc29713/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L87-L91

closed 2020-05-27 by caternuson

caternuson, 2020-05-27:
Hopefully fixed. Please update to the 4.10.1 release when it becomes available:
https://github.com/adafruit/Adafruit_Blinka/releases/tag/4.10.1

bablokb, 2020-05-28:
I can confirm this works now. It does have a little side effect: I tested it with a different program doing I2C. Up to now, GP3 was at configured at it's default, i.e. as I2C traffic indicator. I think GP0/GP1 as UART-RX/TX indicator are also lost now. But this is a different story.

Thanks for your help and support, it is very appreciated.
Bernhard

caternuson, 2020-05-28:
Cool. Thanks for letting us know.

As part of the change, all of the GP pins are set to GPIO and INPUT at start up. So now they won't have the POR defaults as shown in the datasheet. Maybe not really necessary. Mainly being done to force a known initial state.

Hopefully you can just add some extra lines of code to explicitly set each pin to what you want? i.e. don't rely on defaults.

bablokb, 2020-05-28:
I think setting the board to a defined state is the best approach. And since the library provides all the ALT-modes, this should be no issue.

PR #299 (2020-05-27): Update MCP2221
This should fix #295. See discussion in issue thread for more info.

The main change is in `gp_set_mode()`. Using a read command (`0x61`) to get current GP settings has been replaced with storing the states locally in `._gp_config`. These are then used when sending GP settings (`0x60`). They are also updated as needed elsewhere in various functions.

Tested various ways, including with example from issue thread. This is the setup for that test:

[image]