adafruit/Adafruit_Blinka #297: CircuitPython HID Keyboard and Mouse (adafruit_hid) support

MSumulong, 2020-05-26:
I would like to be able to use CircuitPython's adafruit_hid module on a BeagleBone Black but from what I understand, it's pointless to try to install CircuitPython onto a BeagleBone because it already has a full python implementation. Does it make sense to implement a adafruit_hid module in this Blinka library?

makermelissa, 2020-05-26:
That's a really good question. @pdp7 do you have any more insight into this? My initial thoughts are that maybe some kind of BeagleBone only implementation like we have for neopixels on the Pi.

dhalbert, 2020-05-26:
I am not sure if the BeagleBone can act as a USB peripheral (OTG) or not.

MSumulong, 2020-05-26:
The BeagleBone Black that I have can definitely act as a USB peripheral (OTG). The default configuration (Debian image) has the BBB act as a USB storage device, Ethernet, and serial port (USB Multifunction Composite Gadget). I was able to configure it to act as a USB HID device and I'd like to use the adafruit_hid module but it requires the usb_hid module from CircuitPython.

MSumulong, 2020-05-26:
I also believe this might apply to the RPi Zero which has USB Gadget / OTG support (https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget).

makermelissa, 2020-05-26:
I'm not sure of the actual implementation and accessibility from Python, but if it's general enough it might be something that needs its own library like we did with bleio and displayio.

dhalbert, 2020-05-26:
`adafruit_hid` should be resuable as it stands, but what we need to do is to re-implement the native `usb_hid` module as part of Blinka or as a separate Blinka library.

GameDungeon, 2022-02-24:
So blinka still doesn't support usb_hid on the pi zero? That's sad as a project of mine requires it.

dhalbert, 2022-02-24:
> So blinka still doesn't support usb_hid on the pi zero? That's sad as a project of mine requires it.

If you would like to work on it your contributions would be welcome.

GameDungeon, 2022-02-24:
I would be happy to help, but I am quite new to micro-computers and circuit python. I would need a jumping-off point for me to be useful.

tannewt, 2022-02-25:
@GameDungeon Thanks for the offer! Unfortunately I don't know much about the starting point either. The main thing is to figure out the Linux USB Gadget mode that allows Linux to act as a USB device. I'm not sure if there is a way to do this directly from Python already. If there is, then you'd need to wrap the existing API in one that matches `usb_hid`.

fivetide, 2022-02-28:
i started implementing the gadget configuration according to this tutorial, maybe it helps...
https://www.rmedgar.com/blog/using-rpi-zero-as-keyboard-setup-and-device-definition/
you need to load the corresponding kernel modules first as described there.
```
import os
from pathlib import Path

config_root = '/sys/kernel/config/usb_gadget/adafruit-blinka'
report_desc = b'\\x05'#this was some wrong descriptor i copied once, didnt work. has to be a valid one we generate...
try:
    os.makedirs('%s/functions/hid.usb0/protocol' % config_root)
except FileExistsError:
    pass

#tear down if it was still running: set UDC file to empty, unlink feature dir from config
Path('%s/UDC' % config_root).write_text('')
try:
    Path('%s/configs/c.1/hid.usb0' % config_root).unlink()
except FileNotFoundError:
    pass

#set up actual device config
Path('%s/bcdDevice' % config_root).write_text('%s' % 1) # Version 1.0.0
Path('%s/bcdUSB' % config_root).write_text('%s' % 0x0200) # USB 2.0
Path('%s/bDeviceClass' % config_root).write_text('%s' % 0xEF)
Path('%s/bDeviceProtocol' % config_root).write_text('%s' % 0x01)
Path('%s/bDeviceSubClass' % config_root).write_text('%s' % 0x02)
Path('%s/bMaxPacketSize0' % config_root).write_text('%s' % 0x08)

Path('%s/idProduct' % config_root).write_text('%s' % 0x0104)# Multifunction Composite Gadget
Path('%s/idVendor' % config_root).write_text('%s' % 0x1d6b)# Linux Foundation

Path('%s/functions/hid.usb0/protocol' % config_root).write_text('%s' % 1)
Path('%s/functions/hid.usb0/report_length' % config_root).write_text('%s' % 8) # 8-byte reports
Path('%s/functions/hid.usb0/subclass' % config_root).write_text('%s' % 1)
#Path('%s/functions/hid.usb0/report_desc' % config_root).write_bytes(report_desc)
# temp workaround, has to be a proper descriptor
os.system('echo "05010906a101050719e029e71500250175019508810275089501810175019503050819012903910275019505910175089506150026ff00050719002aff008100c0" | xxd -r -ps > functions/hid.usb0/report_desc')
try:
    os.makedirs('%s/configs/c.1/strings/0x409' % config_root)
    #here we want to add human readable stuff later
except FileExistsError:
    pass

Path('%s/configs/c.1/bmAttributes' % config_root).write_text('%s' % 0x080)
Path('%s/configs/c.1/MaxPower' % config_root).write_text('%s' % 100)

#boot up the gadget:
Path('%s/configs/c.1/hid.usb0' % config_root).symlink_to('%s/functions/hid.usb0' % config_root)
os.system('ls /sys/class/udc > %s/UDC' % config_root)
```

tannewt, 2022-03-01:
Thanks @fivetide! I don't know Linux USB gadget but appreciate you sharing your work.

fivetide, 2022-03-15:
I believe this issue can be closed now, has been resolved with #560

makermelissa, 2022-03-15:
Oh right. Thanks for the reminder and thanks for adding support.

closed 2022-03-15 by makermelissa