adafruit/Adafruit_Blinka #301: Document setup for the GreatFET One and other compatible boards

siddacious, 2020-06-03:
After looking on circuitpython.org and learn.adafruit.com for instructions on how to set up the GreatFET One with blinka, I managed to figure it out after searching for "GreatFET one blinka" on google and finding [this blog post](https://blog.adafruit.com/2020/05/26/blinka-support-added-to-greatfet-one-circuitpython-adafruit-gsglabs/)

Once I figured it out it was super simple but finding the info wasn't quite as accessible.

for reference, this is all it took:
```
pip install --upgrade adafruit-blinka
pip install --upgrade greatfet
greatfet info
greatfet fw --auto
export BLINKA_GREATFET=1
```

makermelissa, 2020-06-03:
Yeah, the blog post was my better than nothing attempt for a quickstart guide at this point. What did you mean by "other compatible boards"?

siddacious, 2020-12-28:
This page is great:
https://circuitpython.org/blinka/

It's easy to link to and quickly shows which boards are supported.

It would be cool to have something like the text in my post above included on this page:
https://circuitpython.org/blinka/greatfet_one/

by "other compatible boards" I meant to do the same for any other supported boards that have special installation steps.

Let me know if you'd like me to split off the "other compatible boards" piece into its own long term ticket

makermelissa, 2021-01-04:
Sure, you can split it off into something in the circuitpython.org repo.

makermelissa, 2021-05-06:
Closing because this isn't really something for the Blinka Repo.

closed 2021-05-06 by makermelissa