adafruit/Adafruit_Blinka #302: Board not supported exception - Pi 4B 8GB

SmartCardMaster, 2020-06-04:
Hi,

i tried to use some Adafruit diplay drivers and tools, but on my new Pi4B 8GB i get some trouble. On my Pi4B 2GB all is working.

Is there some simple workaround?

***

pi@raspberrypi:~ $ python3 /home/pi/MicroXServer/start_MicroXServer.py
Traceback (most recent call last):
  File "/home/pi/MicroXServer/start_MicroXServer.py", line 1, in <module>
    import MicroPlayer
  File "/home/pi/MicroXServer/MicroPlayer.py", line 2, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 166, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None

***
output of detect.py

pi@raspberrypi:~ $ python3 /home/pi/MicroXServer/detect.py
Chip id:  BCM2XXX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an OS environment variable special case? False
Raspberry Pi detected.

***
output of cat /proc/cpuinfo

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : d03114
Serial          : 10000000543ea152
Model           : Raspberry Pi 4 Model B Rev 1.4

***
output of pip list

pi@raspberrypi:~ $ pip list
Package           Version
----------------- -----------
arandr            0.1.9
asn1crypto        0.24.0
automationhat     0.2.0
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2018.8.24
chardet           3.0.4
Click             7.0
colorama          0.3.7
colorzero         1.1
configparser      3.5.0b2
cookies           2.2.1
cryptography      2.6.1
dnspython         1.16.0
drumhat           0.1.0
entrypoints       0.3
enum34            1.1.6
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.0.2
fourletterphat    0.1.0
funcsigs          1.0.2
gpg               1.12.0
gpiozero          1.5.1
gyp               0.1
idna              2.6
ipaddress         1.0.17
itsdangerous      0.24
Jinja2            2.10
keyring           17.1.1
keyrings.alt      3.1.1
MarkupSafe        1.1.0
mcpi              0.1.1
microdotphat      0.2.1
mock              2.0.0
mote              0.0.4
motephat          0.0.3
numpy             1.16.2
oauthlib          2.1.0
olefile           0.46
pantilthat        0.0.7
pbr               4.2.0
phatbeat          0.1.1
pianohat          0.1.0
picamera          1.13
picraft           1.0
piglow            1.2.5
pigpio            1.44
Pillow            5.4.1
pip               18.1
pycairo           1.16.2
pycrypto          2.6.1
pyflakes          2.0.0
pygame            1.9.4.post1
PyGObject         3.30.4
pyinotify         0.9.6
PyJWT             1.7.0
pyOpenSSL         19.0.0
pyserial          3.4
pyxdg             0.25
rainbowhat        0.1.0
requests          2.21.0
requests-oauthlib 1.0.0
responses         0.9.0
RPi.GPIO          0.7.0
RTIMULib          7.2.1
scrollphat        0.0.7
scrollphathd      1.2.1
SecretStorage     2.3.1
sense-emu         1.1
sense-hat         2.2.0
setuptools        40.8.0
simplejson        3.16.0
six               1.12.0
skywriter         0.0.7
sn3218            1.2.7
spidev            3.4
touchphat         0.0.1
twython           3.7.0
unicornhathd      0.0.4
urllib3           1.24.1
Werkzeug          0.14.1
wheel             0.32.3

 sudo pip3 install adafruit-circuitpython-rgb-display gpiozero pigpio Adafruit-Blinka Adafruit-PlatformDetect Adafruit-PureIO
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: adafruit-circuitpython-rgb-display in /usr/local/lib/python3.7/dist-packages (3.9.3)
Requirement already satisfied: gpiozero in /usr/lib/python3/dist-packages (1.5.1)
Requirement already satisfied: pigpio in /usr/lib/python3/dist-packages (1.44)
Requirement already satisfied: Adafruit-Blinka in /usr/local/lib/python3.7/dist-packages (5.0.1)
Requirement already satisfied: Adafruit-PlatformDetect in /usr/local/lib/python3.7/dist-packages (2.13.2)
Requirement already satisfied: Adafruit-PureIO in /usr/local/lib/python3.7/dist-packages (1.1.5)
Requirement already satisfied: adafruit-circuitpython-busdevice in /usr/local/lib/python3.7/dist-packages (from adafruit-circuitpython-rgb-display) (4.3.1)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka) (0.7.0)
Requirement already satisfied: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (4.2.3)
Requirement already satisfied: sysv-ipc; platform_system != "Windows" and platform_machine != "mips" in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.0.1)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.51.2)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (3.4)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (1.0.2)

makermelissa, 2020-06-04:
Hi, this rev code was just added yesterday, but I just did a release that includes the change a couple minutes ago. It should be available in about 5-10 minutes in which you can type: `pip3 install --upgrade adafruit-platformdetect`.

SmartCardMaster, 2020-06-05:
Hi, thanks for the fast reply. Works great after update to the current release.

Greetings

closed 2020-06-05 by SmartCardMaster