adafruit/Adafruit_Blinka #303: module 'board' has no attribute 'TX'

indramal, 2020-06-07:
When I configure following line it gives me `module 'board' has no attribute 'TX'` error.

```
import board, busio
uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=10)
```

indramal, 2020-06-07:
This is the reply for `dir(board)`

`['CE0', 'CE0_1', 'CE1', 'CE1_1', 'D10', 'D11', 'D12', 'D13', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'MISO', 'MISO_1', 'MOSI', 'MOSI_1', 'SCK', 'SCK_1', 'SCL', 'SCLK', 'SCLK_1', 'SCL_1', 'SDA', 'SDA_1', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']`

jerryneedell, 2020-06-07:
What system are you running Blinka on?

indramal, 2020-06-07:
Nvidia Jetson Nano

jerryneedell, 2020-06-07:
hmmm -- sorry - I have no experience with that board. Hopefully someone else can help.

indramal, 2020-06-07:
I found the solution. Thank you very much for helping me. 
I can not use this,
`uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=10)`
Instead I can use,
```
import serial
uart = serial.Serial("/dev/ttyUSB0", baudrate=9600, timeout=3000)
```

closed 2020-06-07 by indramal

CaptainJackey, 2020-06-11:
this should be fixed if we use circuitpython!!!