adafruit/Adafruit_Blinka #305: Jetson nano module 'board' has no attribute 'TX' and 'RX'

CaptainJackey, 2020-06-11:
the problem with jetson nano's module 'board' has no attribute 'TX' and 'RX'! @cefn

ladyada, 2020-06-11:
they are not available for use, so there's no RX/TX attributes. 
you can use pyserial 
also, don't post in multiple issues.

closed 2020-06-11 by ladyada

CaptainJackey, 2020-06-11:
> they are not available for use, so there's no RX/TX attributes.
> you can use pyserial
> also, don't post in multiple issues.

thanks , maybe you can update the Tutorial:
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/uart-serial

ladyada, 2020-06-11:
read this section
```

We'll need to configure this code to work with our UART port name.

    If you're using a USB-to-serial converter, the device name is probably /dev/ttyUSB0 - but check dmesg to make sure.
    If you're using the built-in UART on the DragonBoard, the device name is /dev/ttyTHS1.

Comment out the lines that reference board.TX, board.RX and busio.uart and uncomment the lines that import serial and define the serial device, like so:
```