adafruit/Adafruit_Blinka #308: FT2232H support

fharding1, 2020-06-17:
Hi! I'm working on a board called [Tigard](https://github.com/tigard-tools/tigard) at the moment, which is based off the FTDI FT2232H. The FT2232H is similar to the FT232H, but with dual MPSSEs. It would be awesome if I could use Tigard (or some of the other FT2232H-based boards such as [timep](https://github.com/Matir/timep)) with CircuitPython/Blinka, and I see there is already support for the FT232H! I'd be happy to take this work on myself, but wanted to ask some questions before throwing a PR at you so my work can hopefully be merged upstream if this is something Blinka wants to support:

There are a few options for implementing AFAICT:

* Make a generic class that's used for both the FT232H and the FT2232H classes (similar to how the "tegra" folder is structured) . Since the two chips are really similar, I think this makes the most sense. I'm not sure about the name, maybe something along the lines of `ftdi`, `ftdi-mpsse`, or `ftxxxx`?
* Just copy the FT232H class and make a FT2232H class. This is easy and keeps the FT232H code as-is but super redundant.
* Simply modify the FT232H class to look for `ftdi:///1` instead of `ftdi://ftdi:ft232h/1`. This technically works (I have tested it), but you can only use the first MPSSE so it's really not ideal.

I think the first is the clear choice, but I'm not super familiar with this library, so let me know what you think, thanks!

mpratt14, 2020-07-28:
> * Simply modify the FT232H class to look for `ftdi:///1` instead of `ftdi://ftdi:ft232h/1`. This technically works (I have tested it), but you can only use the first MPSSE so it's really not ideal.

What if you set the url to what it actually is? Then you can use both with more than one instantiation of the board
`ftdi://ftdi:2232h/1`
`ftdi://ftdi:2232h/2`

Note that "support" for the board is already provided by the pyftdi library, not here. The only thing that this library does in terms of "adding support" is identifying the board by product ID and pin name mapping. This is what `import board` does as far as I know (along with set the default url). You can set the BLINKA_FT232H variable but the actual port and everything else can be changed with the .configure() function and redefining variables.

mpratt14, 2020-07-28:
If you want to make the support permanent (via an environment variable like BLINKA_FT232H) you would start here

https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/chip.py#L67

and then fill in the missing pieces in the chips and board and detect classes
and then here in this library, you can rename the Pin class I guess and have both boards refer to it

caternuson, 2020-07-28:
Here's a basic example of enabling FT232H and then showing the available pins:
```python
$ export BLINKA_FT232H=1
$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> dir(board)
['C0', 'C1', 'C2', 'C3', 'C4', 'C5', 'C6', 'C7', 'D4', 'D5', 'D6', 'D7', 'I2C', 'MISO', 'MOSI', 'SCK', 'SCL', 'SCLK', 'SDA', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
>>>  
```

What would the expected behavior of the above be for the case of multiple FT232H's?

fharding1, 2020-07-28:
@mpratt14 

I forked and updated https://github.com/tigard-tools/Adafruit_Python_PlatformDetect as well. I updated this library (see the fork I mentioned), so I know what support entails, but thank you for the explanation. I don't think it makes sense to instantiate two FT232H boards to represent one FT2232H board, and that's definitely not very user friendly.

mpratt14, 2020-07-28:
I wanted to make it clear to others who will see this as well and might be confused, thinking that this library does it all.

Sorry for the confusion. I meant it is possible to instantiate two FT232H, one with a unique Product ID OR to instantiate one of them as an FT2232H by changing its product ID to that, not to do both at the same time.

fharding1, 2020-07-28:
@mpratt14 yeah you can instantiate as an FT2232H, but you would still just have access to one channel. The point of the PR is to make it so that you can use one instantiated FT2232H to access both channels.

mpratt14, 2020-07-29:
I was confusing this with the other issue with multiple FT232H. I took a look at your branch and it looks done, are you going to start a PR?

fharding1, 2020-07-29:
@mpratt14 Yes, but I was hoping to get a response from a maintainer saying this is something they would want to merge before opening a PR. E.g. whether they think FT2232H support is valuable.

mpratt14, 2020-07-29:
> to get a response from a maintainer saying this is something they would want to merge

IMO this is one of the purposes of opening a PR. It's been silent here since you made this issue

fharding1, 2020-07-29:
@mpratt14 

As a maintainer of at least one relatively popular Github project, I really appreciate when people first open an issue to discuss a feature they want to add and wait until maintainers approve it to open a PR. Otherwise it can be awkward rejecting a feature that someone has spent a lot of time on. I am not going to open a PR to get someone's attention when they haven't even said whether they would consider the feature valuable or not.

I am going to wait for a response from a maintainer before opening a PR. Until then, I will be maintaining my own downstream fork. I am asking you to please refrain from commenting on this issue anymore unless you have feedback about the actual proposed feature. Thanks.

mpratt14, 2020-07-29:
That's understandable, but in this case it looks like you will be waiting forever. You did 95% of the work of making the PR. At this point, it is the same effort as writing an email, not a huge rejection or awkward failure. If you really want this to move forward you need to @ them or just make your case in a PR.

I would love to give your work actual review and feedback and make sure it is compatible with my FT232H, but without a PR it's not as straightforward and organized.

fharding1, 2020-07-29:
@mpratt14 you are not a maintainer. If the maintainers are too busy to respond to my issue, why would they have time to review a PR? I'm not going to bother them beyond maybe pinging them if this issue gets completely buried. And why would you want to spend time reviewing my PR if the maintainers haven't even said they would accept it? I am happy to wait until they're able to respond. I won't be continuing this conversation as I've already told you what I'm going to do with my issue/potential PR and asked you to drop it.

ladyada, 2020-07-29:
@fharding1 we're happy to support FT2232 thru a PR. its probably best to copy and paste FT232H folder, and extend it so that it seemlessly works with all the pins for the FT2232 (i've never used the FT2232 but i'm assuming its a single USB interface). once you have all the pins/peripherals working, we could refactor to make FT232H/FT2232 work as one code

caternuson, 2020-07-29:
Could the FT2232H be treated the same as "multiple FT232H"s as being discussed in #260 ?

In terms of the MPSSE, which is what provides GPIO/SPI/I2C, the FT2232H appears to be essentially 2 FT232H's shoved in one package. From the host's point of view - is there any difference between connecting a FT2232H and connecting 2 FT232H's?

ladyada, 2020-07-29:
kinda? but i think having it be 1 USB device vs 2 could make a fairly big difference in implementation

mpratt14, 2020-07-29:
@caternuson in order to do that I believe the user would be required to reprogram the product ID, or am I mistaken?

Much easier is the exact opposite (for #260 not this) as my suggestion for using 2 boards of FT232H at the same time by reprogramming one of them to be identified as an FT2232H instead. The environment variable BLINKA_FT232H decides how this library will handle both of them.

I like his implementation as it is. Maybe consider having two environment variables instead of one? BLINKA_FTDI and FTDI_232H vs FTDI_2232H, or maybe ID the boards the same way that the pyftdi library does.

caternuson, 2020-07-29:
In general, I'm not sure. I don't have any experience with the FT2232H. While it's a single USB device, it might show up as 2 FTDI URLs? And two FT232H's, which would be two USB devices, may show up the same?

mpratt14, 2020-07-29:
See this which I also linked in #260 

https://github.com/eblot/pyftdi/blob/master/pyftdi/doc/urlscheme.rst

FT2232H is 1 USB device which means 1 Product ID
but 2 interfaces which means 2 URLs

caternuson, 2020-07-29:
So 1xFT2232H and 2xFT232H both result in 2 URLs. The URLs are what matter to pyftdi. So maybe these two cases could be treated the same?

mpratt14, 2020-07-29:
Ah thats right, 2 interfaces means 2 urls and 2 scripts to use them both...

So the real question is, is there anything that a 2232H interface has that a 232H interface doesn't?

fharding1, 2020-07-30:
From before:

> The point of the PR is to make it so that you can use one instantiated FT2232H to access both channels.

It doesn't really make sense in a consumer's code to initialize two FT232H devices to represent an FT2232H in my opinion. It's awkward, and would probably result in them implementing their own wrapper class. I think creating a new FT2232H device with both channels exposed like I have setup in my fork is much cleaner than two individual FT232H's with the same pinouts, regardless of whether it's _possible_ to do so. For the same reason, I like having two different environment variables: FT232H and FT2232H.

Semantically an FT2232H is not two FT232H's. An FT232H is an FTDI chip with a single MPSSE, and an FT2232H is an FTDI chip with two MPSSEs. I might be okay with just a generic "MPSSE" class, but that would break existing functionality. I think considering the fact that we already have an FT232H, we should just add the FT2232H as a similar device.

caternuson, 2020-07-30:
A scheme similar to what you have here:
https://github.com/tigard-tools/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/ftdi_mpsse/ft2232h/pin.py
could possibly be generalized though, to any N number of MPSSE's. The library would determine the number of MPSSE's somehow, via available URLs maybe, and then dynamically create and setup the pins.

fharding1, 2020-07-30:
Yes, but that would be confusing to new users. Blinka provides support for specific devices such as an FT232H, not engines that might be implemented in any number of chips like a generic MPSSE.

securelyfitz, 2021-03-26:
Hey, just checking in on this. Tigard boards are just finally getting into people's hands, and I'm still pointing them to @fharding1 's fork of Blinka.

Is see both this and #260 are tagged as enhancements. Is there a catch-all solution in the works to support multiple FT232Hs and FT2232H (and probably the FT4232H) at the same time? 

If not, is there anything I can do to help get @fharding1's FT2232H support added?

ladyada, 2021-03-26:
yes! we'd look at a PR that adds FT2232H support without breaking FT232H support, on mac/win/linux... but none have been submitted yet :)

fharding1, 2021-03-26:
Great, I'll try to have a PR up this Tuesday or sometime next week (don't have access to an FT2232H device at the moment).

I should be able to test on Linux and Windows, but I might need some help with macOS.

ladyada, 2021-03-26:
ok we also dont have an FT2232 right now, but we'll find a dev board for one!

securelyfitz, 2021-03-26:
@ladyada let me know, If you'd like i'll send you a few Tigard boards: https://github.com/tigard-tools/tigard

PR #437 (2021-05-12): FT2232H support and FT232H/MPSSE refactor
This pull request adds support for the FT2232H without breaking any existing functionality with the FT232H, as outlined in #308. So far all the tests I've thrown at my Tigard FT2232H dev board (MCP9808 I2C temperature sensor, GPIO, etc) have worked out great, but if someone wants to verify with some other (SPI) devices (possibly on another OS besides Linux, too) that would be great.

https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/149 accompanies this pull request.

Closes #308
closed 2021-05-12 by caternuson