adafruit/Adafruit_Blinka #320: Docker + blinka + board on the Jetson Nano

lackdaz, 2020-07-15:
Hi! This is not so much of an issue per se but a feature for consideration. 

I had previously written some code for battery management for connecting with an i2c that does battery management (MAX17261) and all is well and fine because of the splendid guides that you've written: [link to learn.adafruit guide](https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano?view=all). **I tend to write my codes in modules, so I can import them and run them realtime.**

So I got my current readings, time-to-charge(TTC) etc for the battery management IC within my docker container running a [board hack](https://github.com/Heerpa/jetson-gpio-nanohard). It was a hack only because the nvidia Jetson.GPIO library also has some kind of platformdetect that doesn't work very well outside of a privileged container -- certain parts of a virtual filesystem like the DTS  cannot be mounted to a docker container. But this is another story for another time.

So with custom `i2c` completed, I set out to write some more new GPIO code, specifically implementing some a base class that just turns on/off, and perhaps some switch. 

`base.py`
```
__all__ = ["BaseGPIO"]

import RPi.GPIO as GPIO
import sys

if sys.version_info.major < 3:
    print("You need to run this on Python 3")
    sys.exit(-1)

class BaseGPIO(object):
    """Parent class for GPIO objects

    Args:
        object (Object): Default
    """

    __slots__ = ["pin", "gpio_state"]

    def __init__(self, pin: int, *args, **kwargs):
        """Creates an instance of BaseGPIO

        Args:
            pin (int): GPIO pin e.g. 13
        """
        self.pin = pin
        self.gpio_state = GPIO.LOW  # starts as off
        GPIO.setmode(GPIO.BOARD)
        GPIO.setup(self.pin, GPIO.OUT, initial=GPIO.LOW)
        GPIO.setwarnings(False)  # ssh!

    def on(self):
        """Changes GPIO pin state to HIGH"""
        GPIO.output(self.pin, GPIO.HIGH)

    def off(self):
        """Changes GPIO pin state to LOW"""
        GPIO.output(self.pin, GPIO.LOW)

    def switch(self):
        """switch GPIO pin state e.g. Off -> On -> Off"""
        self.gpio_state ^= GPIO.HIGH
        GPIO.output(self.pin, self.gpio_state)
```

When trying to import this library however, I'm getting errors about like these:
```
$ python3 -i capture.py 
Traceback (most recent call last):
  File "capture.py", line 21, in <module>
    main()
  File "capture.py", line 16, in main
    NIR = PWM(33)
  File "/home/seth/code/eyescan/api/app/gpio/led/pwm.py", line 40, in __init__
    super().__init__(*args, **kwargs)
  File "/home/seth/code/eyescan/api/app/gpio/base.py", line 28, in __init__
    GPIO.setmode(GPIO.BOARD)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 298, in setmode
    raise ValueError("A different mode has already been set!")
ValueError: A different mode has already been set!
```

The issue lies with blinka's [`GPIO.setmode(GPIO.TEGRA_SOC)`](https://github.com/adafruit/Adafruit_Blinka/blob/a3770186a94117be56a8d9a8038a2d540bd2884c/src/adafruit_blinka/microcontroller/tegra/t210/pin.py#L6). You can't run setmode twice in the same python runtime environment and is going to be stumble robotics projects that integrate multiple gpio functions like mine.

blinka works great per se, but shouldn't dictate the use of less idiomatic pin mappings in the runtime environment for other modules.

ladyada, 2020-07-15:
yes you'll need to port the 'gpio' code to circuitpythons - we have all the above functions already implemented :)

lackdaz, 2020-07-16:
I do want to retain some control over the pin functions, as I'm doing stuff like threaded software pwm in the child classes later. I tried using importing one of the board pins to the above class and this error then popped up

```
Traceback (most recent call last):
  File "capture.py", line 24, in <module>
    main()
  File "capture.py", line 20, in main
    foo = Fixation(D19)
  File "/app/app/gpio/led/fixation.py", line 56, in __init__
    super().__init__(pin, *args, **kwargs)
  File "/app/app/gpio/base.py", line 29, in __init__
    GPIO.setup(self.pin, GPIO.OUT, initial=GPIO.LOW)
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 326, in setup
    ch_infos = _channels_to_infos(channels, need_gpio=True)
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 121, in _channels_to_infos
    for c in _make_iterable(channels)]
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 121, in <listcomp>
    for c in _make_iterable(channels)]
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 103, in _channel_to_info_lookup
    if channel not in _channel_data:
TypeError: unhashable type: 'Pin'
```

lackdaz, 2020-07-16:
@ladyada I chased about the issue and realised there are pin errors in the [jetson_nano.py](https://github.com/adafruit/Adafruit_Blinka/blob/0162c2e0b57603929c8f5f09933ff63579e6a9aa/src/adafruit_blinka/board/nvidia/jetson_nano.py#L19) pins file. 

I'm using board pin 13, which would correspond to D13 which points to E06 in the t210 [pins.py file](https://github.com/adafruit/Adafruit_Blinka/blob/a3770186a94117be56a8d9a8038a2d540bd2884c/src/adafruit_blinka/microcontroller/tegra/t210/pin.py#L126) which is actually the PWM pin (33) as shown in the [GPIO schematic](https://nekodaemon.com/images/pasted-0.png)

Could someone verify this? I switched to B06 and it works on hardware but is listed as D27 in the [jetson_nano.py](https://github.com/adafruit/Adafruit_Blinka/blob/0162c2e0b57603929c8f5f09933ff63579e6a9aa/src/adafruit_blinka/board/nvidia/jetson_nano.py#L19) pins file.

This is for a jetson nano b01, with 32/33 enabled as pwm pins.

ladyada, 2020-07-16:
we dont own a nano so not sure - @swarren do you have any insight?

lackdaz, 2020-07-16:
I would be happy to submit a pull request for the required changes to get these pin mappings in good order, if someone could verify that there are indeed mistakes.

swarren, 2020-07-16:
I have confirmed that the pin numbering tables in Blinka are correct, for Nano at least. The confusion comes from what the number in e.g. "D13" means. Here, "13" does not refer to the physical pin number on the Jetson GPIO expansion connector, but rather the Broadcom GPIO number of the matching physical pin on a Raspberry Pi's GPIO expansion connector.

As background: When we added Jetson support to Blinka, we wanted to use the same pin numbering/naming scheme as the existing Blinka support for the Raspberry Pi, so that SW written for the Pi would work unmodified on Jetson. Thus, we used the Pi's Broadcom chip's GPIO numbering scheme. Sorry, this can be a bit confusing!

To find the physical pin that a Blinka pin name/number refers to on Jetson, you'll need to trace through a few mapping files (see below) or refer to any online Pi GPIO pinout diagram that shows the Broadcom GPIO numbers.

Start with Blinka's definition of the pin name/number:
https://github.com/adafruit/Adafruit_Blinka/blob/0162c2e0b57603929c8f5f09933ff63579e6a9aa/src/adafruit_blinka/board/nvidia/jetson_nano.py#L10
e.g.:
`D4 = pin.BB00`
(As an aside, note that the value assigned here is usually the Jetson SoC (Tegra) GPIO pin name/number)

Next, look up the assigned value in Blinka's definition of the SoC/chip that is on that Jetson board:
https://github.com/adafruit/Adafruit_Blinka/blob/a3770186a94117be56a8d9a8038a2d540bd2884c/src/adafruit_blinka/microcontroller/tegra/t210/pin.py#L120
i.e.:
`BB00 = Pin("AUD_MCLK")`

Next, look at the Jetson.GPIO pin data for that board and pin:
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L176
i.e.:
`(216, "/sys/devices/6000d000.gpio", 7, 4, 'GPIO9', 'AUD_MCLK', None, None),`

The fields in that file are mentioned near the top:
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L30
i.e.:
Linux GPIO, sysfs dir, connector pin number, Broadcom pin number, Jetson CVM pin name, Tegra pin name
The last of those matches the value from Blinka's pin.py, and the third of those is the physical pin on the connector. So, pin AUD_MCLK is BCM GPIO ID 4, and is routed to GPIO connector physical pin 7.

lackdaz, 2020-07-17:
Ok @swarren Ok its nice to finally see some documentation of the Jetson BCM/SOC mode mappings. The names seem to differ from the jetsonhacks/neko diagram. So a new diagram with BCM pin numbers might be required (unless you know of one).

It looks like the entire library is using BCM pins hence it was done with the lookup tables. Hooboy, Confusing indeed.

closed 2020-07-17 by lackdaz