adafruit/Adafruit_Blinka #323: Board Not Supported

bennyfuckinbo, 2020-07-28:
/proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02042
Serial		: 00000000e9cc5bf0
Model		: Raspberry Pi 2 Model B Rev 1.2

Detect.py

Chip id:  BCM2XXX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an OS environment variable special case? False
Raspberry Pi detected.
ASUS Tinker Board device detected. 

(I dont have a ASUS Tinker Board)

Adafruit_Platform Detect is 2.14.1

Im having no luck.

makermelissa, 2020-07-28:
Hi, this sounds like a bug in PlatformDetect. Is Raspberry Pi 2 Model B Rev 1.2 the correct board?

bennyfuckinbo, 2020-07-28:
I fixed it by adding my rev code to the boards.py file in the constants folder. Yes that is the version

makermelissa, 2020-07-28:
Cool. What is your Rev code? I'd like to add it so it doesn't happen to someone else?

makermelissa, 2020-07-28:
Nevermind XD I see it above.

closed 2020-07-28 by makermelissa