adafruit/Adafruit_Blinka #325: how to manage unit tests with board busio packages ?

MkLHX, 2020-07-29:
Hello, i'm working on a project who used adafruit-blinka library and i need to build unit tests on my project.
My device use i2c devices and i need to make unit test on my own code who use board, busio and i2c protocol.
I'm looking for the best way to manage this and make my project as robust as possible.

tannewt, 2020-07-29:
I don't know of anyone doing this yet. However it should be easy to pass in a mock i2c bus device or i2c bus to do your testing. Then, the mock can stub out the i2c data.

MkLHX, 2020-07-30:
it's what i expect to deal. But i found nothing  about the recommended way to do. so i'll going to try to mock i2c bus and deal my tests.

MkLHX, 2020-08-05:
ok so i manage mocking i2c by using MocMagic class from unittest module. So i close the issue.

closed 2020-08-05 by MkLHX