adafruit/Adafruit_Blinka #328: AttributeError: module 'gpiod' has no attribute 'Chip'

mateo, 2020-08-11:
Hi there,

Thanks for the amazing library, used it many times on a raspberry pi and love it.  I've been trying to get it to work on an Odroid XU4 though and I keep getting the titled error.  Libgpiod is present on the device, but it seems severely inconsistent with this library.  For starters, 'Chip' doesn't exist but 'chip' does.  I began rewriting the file at /microcontroller/generic_linux/libgpiod_pin.py and after fixing the first couple errors it's clear that my version of gpiod is wildly different than what this is written for.  I am running libgpiod directly from git.kernel.org and I've tested versions 1.4 and 1.1.1.  I tried using the Adafruit Raspberry Pi Scripts installer for libgpiod v1.5.(whatever the latest is) but the latest version throws an error if you don't have kernel 5.5.

So I guess the question, sorry for the slight rant: what version of libgpiod is this written for and is kernel v5.5 required?  (I imagine it has to be something wrong with my version of gpiod since Ubuntu doesn't even support 5.5 yet, 5.4 is the latest LTS)

makermelissa, 2020-08-11:
I'll have to take a look. I do have the XU4, so I can try it on that.

mateo, 2020-08-11:
Ok, thank you for the quick response!  If it's relevant, I'm running Kernel v4.14.180-176, Ubuntu 18.04.4 LTS for odroid

makermelissa, 2020-08-11:
Ok, cool.

makermelissa, 2020-08-11:
When it was added, it was likely using armbian, so it might be a contributing factor.

makermelissa, 2020-08-11:
Oh never mind. I might be wrong about that.

makermelissa, 2020-08-11:
Ok, I poked around and was able to get it to work. Here's what I did:

Start by running the Adafruit libgpiod script:
```
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/libgpiod.sh
chmod a+x libgpiod.sh
sudo ./libgpiod.sh
```

It will fail at this point with the linux headers message, but we at least have all the dependencies installed. Since you already did that, you probably don't need to do it again.

To get it working, I basically installed the 1.4.2 branch. Here's the steps:
```
git clone git://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git
cd libgpiod
git checkout v1.4.2 -b v1.4.2
./autogen.sh --enable-tools=yes --prefix=/usr/local/ --enable-bindings-python CFLAGS="-I/$include_path"
make
sudo make install
sudo ldconfig
sudo cp bindings/python/.libs/gpiod.so /usr/local/lib/python3.?/dist-packages/
sudo cp bindings/python/.libs/gpiod.la /usr/local/lib/python3.?/dist-packages/
sudo cp bindings/python/.libs/gpiod.a /usr/local/lib/python3.?/dist-packages/
```
Make sure Blinka is installed and you can check by opening Python and typing `import board`.

Let me know how it goes.

makermelissa, 2020-08-12:
I updated the Adafruit script now, so you can pass it a `--legacy` or `-l` flag for older systems, which automatically does what I pasted above.

closed 2020-08-12 by makermelissa

mateo, 2020-08-12:
It works!  Always a joy to be greeted with a new error after staring at the same one for multiple days.  Not Adafruit related though, so yeah this solved it on my system too.  Thanks for the quick fix!

makermelissa, 2020-08-12:
You're welcome. Thanks for letting us know.

hallskelet, 2021-01-05:
Hello. I am having this problem. If I execute the script through SUDO, then everything is fine. But the problem is that octoprint is not executed with superuser rights. and I do not know what needs to be done to make the octoprint "venv" work correctly.
If I use the command in the first part of the proposed solution, then it works but there is no effect.
And if from the second part, then an error appears that it is impossible to find python.h and the distant commands do not work.
Can you tell me how to fix this.
I am not a python programmer and am very poor at Armbian.

Here is the error log output with this library:```

2021-01-05 22:01:37,674 - octoprint.plugin.core - ERROR - Error loading plugin display_panel
Traceback (most recent call last):
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/orangepi/oprint/lib/python3.7/site-packages/octoprint_display_panel/__init__.py", line 16, in <module>
    from board import SCL, SDA
  File "/home/orangepi/oprint/lib/python3.7/board.py", line 76, in <module>
    from adafruit_blinka.board.orangepi.orangepipc import *
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/board/orangepi/orangepipc.py", line 3, in <module>
    from adafruit_blinka.microcontroller.allwinner.h3 import pin
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/microcontroller/allwinner/h3/pin.py", line 4, in <module>
    PA0 = Pin(0)
  File "/home/orangepi/oprint/lib/python3.7/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip("gpiochip0", gpiod.Chip.OPEN_BY_NAME)
AttributeError: module 'gpiod' has no attribute 'Chip'
```

makermelissa, 2021-01-05:
@hallskelet try running it with the `-l` or `--legacy` flag.

makermelissa, 2021-01-05:
I can't remember the exact cause of the problem, but it was fixed by installing an older version on libgpiod where they removed the Chip attribute. The legacy flag tells the installer script to use the older version with the Chip attribute. It seems to be enabled when libgpiod is installed as a package, but perhaps the file using it needs some updating.

makermelissa, 2021-01-05:
> And if from the second part, then an error appears that it is impossible to find python.h and the distant commands do not work.
@hallskelet I just noticed the part about python.h not being found in your comments, which is new to me.

Could you open a new issue in https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues? Please post the complete error message. Thanks.