adafruit/Adafruit_Blinka #329: SPI Port appears incorrect

gaynoraj19, 2020-08-13:
[https://learn.adafruit.com/adafruit-radio-bonnets/rfm9x-raspberry-pi-setup]
`CS = DigitalInOut(board.CE1)
RESET = DigitalInOut(board.D25)
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, 915.0)`

The link above recommends to set up the radio as a SPI device using CE1, This would suggest that the board should be connected reserving `/dev/spidev0.1` on raspberry pi. The source code attempts to use `/dev/spidev0.0` however.
This causes conflicts with other spi devices since the board is both reserving a the chip select from spi0.1 and the handle to spi0.0

gaynoraj19, 2020-08-19:
I experimented with adding additional SPI chip select lines to an overlay file. These chip select lines function and are able to transfer data from multiple MCP2515 CAN transceivers. The moment I initialize the radio, all SPI data transfers fail from the CAN transceivers...

makermelissa, 2021-05-06:
The reason it does this is because we don't are using the full SPI port and Blinka kind of ignores the CS pins (see https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/spi.py#L79). I'm closing this because there really isn't an easy way to change this since Linux twiddles with the CS pins a bit different than some of the hardware expects. My suggestion for additional devices would be to use SPI1.

closed 2021-05-06 by makermelissa