adafruit/Adafruit_Blinka #331: wrong pins for Pi Model A original

caternuson, 2020-08-17:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=60&t=168458

Board detect seems to be working OK. But logic in Blinka is grouping Model A's with Rev 1 Model B's. So SCL/SDA pins end up mapped incorrectly. I don't think there ever was a "Rev 1" Model A.

ladyada, 2020-08-17:
jeez yeah - this is older - do they have a fix?

caternuson, 2020-08-17:
Yep, it looks like it. They did a simple hack in the forum thread and got things working. Basically just move SCL/SDA to current D3/D2 pins. I *think* could maybe just move Model A down to match with PI B rev 2:
https://github.com/adafruit/Adafruit_Blinka/blob/ad7c0ada0067ee960ede0cb407ca75a76815a9d0/src/board.py#L53-L57

ladyada, 2020-08-17:
ok cool @makermelissa take a look some time in the next week or so?

PR #337 (2020-08-26): Adjusted Pi 1 Model A to use Pi 1 B Rev 2 board def
Fixes #331.
closed 2020-08-26 by makermelissa