adafruit/Adafruit_Blinka #332: UART missing several properties and methods

dhalbert, 2020-08-18:
`UART` is missing:
- `baudarte`
- `in_waiting`
- `timeout`
- `__enter__()`
- `__exit__()`
- `reset_input_buffer()`

I don't actually need these on Blinka right now, but because Blinka is used by pylint to check methods and properties, I have to disable checking for these.

makermelissa, 2020-08-18:
If you are referring to the UART class in busio, that is only used in MicroPython. We have been using PySerial for anything linux based.

dhalbert, 2020-08-18:
Ah, OK, though it's still true that pylint uses these definitions to vet calls to UART. Perhaps this is overloading the function of Blinka.

Could you actually wrap PySerial inside busio.UART?

makermelissa, 2020-08-18:
Hmm, that's not a bad idea. Then we wouldn't need people to change code for examples.