adafruit/Adafruit_Blinka #334: Possible to use this library on i.mx8m mini board?

arrow53, 2020-08-23:
I'm trying to use the qwiic version of the adafruit ssd1306 oled with a i.mx8m mini board

https://boundarydevices.com/product/nitrogen8m-mini/

I'm using yocto/dunfell. This part seems to show up fine
```
root@nitrogen8mm:~# i2cdetect 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- UU 3d -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
```

But, when I try to run the examples I can't get past the import board line

```
Python 3.8.2 (default, Feb 25 2020, 10:39:28) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/board.py", line 169, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None
>>> 
```

Should I continue to pursue this or is this just going to be a difficult path? Where should I start?

ladyada, 2020-08-23:
this board is not supported by Blinka - here is a guide on how you can do it, we dont have plans to add this board as we dont own one. https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

arrow53, 2020-08-24:
@ladyada that link was what I was after. thanks. 

I'm not sure it makes sense to add this board support as there is no standard header.

But, as I'd still like to use the oled I'm hoping I can still get some help.

My i2c bus is `/dev/i2c-2`. It's a standard i2c bus. My question do I add support in the `pin.py` file?

I have just this in my file
```
i2cPorts = (
  (0, I2C2_SCL, I2C2_SDA),
)
```

this throws an error as those are aliases that aren't define. But, I'm not sure what pin numbers to use as this is just a dedicated i2c bus?

ladyada, 2020-08-24:
how are you plugging in the sensor then, there must be *some* pins exposed

arrow53, 2020-08-24:
I am using i2c3, which uses pads E10 and F10 on the NXP i.m8m mini processor. I'm not sure this makes sense in terms of gpio as i'm not forming the i2c bus from gpio pins. If I just specify random numbers for Pin() then gpiod throws an error. Ideally, I'd just remove gpiod from this processor entirely.

ladyada, 2020-08-24:
if they are pads, they are GPIO

arrow53, 2020-08-24:
sorry, I don't know enough to interpret that. Do I need t figure out the "pin" number? How does the mapping in gpiod go from the pin to the /dev/i2c-2 device? I'm not really sure even what to google.

If I know the my device `/dev/i2c-2` is there a way to figure out the value to set I2C2_SCL/SDA to?

This is what I get if I use just 0 and 1 as the pins
```
Python 3.8.2 (default, Feb 25 2020, 10:39:28) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/board.py", line 121, in <module>
    from adafruit_blinka.board.nitrogen8mm import *
  File "/usr/lib/python3.8/site-packages/adafruit_blinka/board/nitrogen8mm.py", 
line 3, in <module>
    from adafruit_blinka.microcontroller.nxp_imx8mm import pin
  File "/usr/lib/python3.8/site-packages/adafruit_blinka/microcontroller/nxp_imx
8mm/pin.py", line 5, in <module>
    I2C2_SCL = Pin(0)
  File "/usr/lib/python3.8/site-packages/adafruit_blinka/microcontroller/generic
_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip("gpiochip0", gpiod.Chip.OPEN_BY_NAME)
AttributeError: module 'gpiod' has no attribute 'Chip'
>>> 
```

arrow53, 2020-08-24:
@ladyada I could also just use the ssd1306 linux driver. If I use that it shows up as `/dev/fb0`. Is there a way to use the display with python if it's just a linux framebuffer device?

```
root@nitrogen8mm:~# fbset 

mode "128x64-0"
        # D: 0.000 MHz, H: 0.000 kHz, V: 0.000 Hz
        geometry 128 64 128 64 1
        timings 0 0 0 0 0 0 0
        accel false
        rgba 1/0,1/0,1/0,0/0
endmode
```

makermelissa, 2020-08-25:
If you want to add the board to Blinka and just access the I2C port directly, you can use this library in conjunction with Blinka: https://github.com/adafruit/Adafruit_Python_Extended_Bus

arrow53, 2020-08-26:
@makermelissa this is perfect! works great!

closed 2020-08-26 by arrow53